TrunkRecorder/trunk-recorder #1: Display status of recorder on webpage.

robotastic, 2015-02-03:
The goal would be a way to display system information about the number of calls currently being recorded and information about the trunking system. This information would be sent out as JSON over a localhost port to a node.js server. Node.js would handle updating a dashboard and serve up the page locally. It would make it really easy to check on what is happening with the recorder while it is on your network.

amlethojalen, 2018-01-10:
I'm in the process of doing a dashboard at the moment, it uses nodejs and the status socket, i'm happy to add all the source into this project to keep it all together if you want.

amlethojalen, 2018-01-10:
If you send me an email, i can send you a link to the test page i'm working on (its public but hidden at the moment)

robotastic, 2018-01-10:
Awesome! I will send you an email. I am still getting everything documented around status socket. I have a sample server here: https://github.com/robotastic/trunk-recorder-status-server
it is ugly though.

amlethojalen, 2018-01-10:
I've got some pretty basic doco on the json messages that i implemented, they are in the PR, i'll clean up my implementation a bit before i make it public. There was only one breaking change which changing the rates to be an array of rates instead of key value set.

allonucleus, 2018-12-03:
Made a pull request that fixes it

robotastic, 2023-02-19:
This is largely handled through the Status Server and MQTT.

closed 2023-02-19 by robotastic