TrunkRecorder/trunk-recorder #6: Analog squelch

patrickfish, 2015-02-28:
Implement RF squelch for analog transmissions.

robotastic, 2015-02-28:
Do you know much about the GnuRadio Squelch blocks? There are a couple and I am not sure the best to use. In the original SmartNet program, they use Power Squelch, I could try that.

https://github.com/bistromath/gr-smartnet/blob/master/src/python/logging_receiver.py#L41

patrickfish, 2015-03-07:
I tried (unsuccessfully) to test it on my own branch. I'd be willing to test it if you implemented it.

robotastic, 2015-03-07:
I just added the code, but it looks like there is an error in Gnuradio where it is not installing a header file correctly. I am going to posted it to the mailing list to get fixed. I will see if I can install the header file locally and make it work.

robotastic, 2016-03-05:
Closing since there has been no update and the code has been updated to fix a lot of problems.

closed 2016-03-05 by robotastic