TrunkRecorder/trunk-recorder #7: Can't select RTL device for systems with multiple rtl dongles. (Includes fix)

zeronullity, 2015-03-24:
RTL device string is incomplete it always defaults to device index 0. Which means you can't use more than 1 rtl device at a time.. or select another device if device 0 is being used by another program.

Following is a fix to allow device selection and use multiple rtl devices.
You can either use  rtl="device index number" example: rtl=0,rtl=1,rtl=2,etc.. 
or you can use rtl="serial number"   example: rtl=00000001,rtl=00000002, etc.. or what ever you programmed them to using rtl_eeprom from the rtl-sdr package. There is a config.json example at the bottom.

Fix for source.cc: 

Source::Source(double c, double r, double e, std::string drv, std::string dev)

                if (driver == "osmosdr") {
                osmosdr::source::sptr osmo_src;
                osmo_src = osmosdr::source::make(dev);

                std::cout << "SOURCE TYPE OSMOSDR (osmosdr)" << std::endl;
                std::cout << "Setting sample rate to: " << rate << std::endl;
                osmo_src->set_sample_rate(rate);
                std::cout << "Tunning to " << center + error << "hz" << std::endl;
                osmo_src->set_center_freq(center + error,0);
                source_block = osmo_src;
        }

config.json example:

serial number
{"center": 859000000.0,
"rate":2400000.0,
"error":-43100,
"gain": 49,
"antenna": "TX/RX",
"digitalRecorders":4,
"analogRecorders":4,
"driver": "osmosdr",
"device": "rtl=00000004"}

or 

device index
{"center": 859000000.0,
"rate":2400000.0,
"error":-43100,
"gain": 49,
"antenna": "TX/RX",
"digitalRecorders":4,
"analogRecorders":4,
"driver": "osmosdr",
"device": "rtl=2"}

robotastic, 2015-03-26:
Awesome - just added this code, I still have to update the Readme

zeronullity, 2015-03-26:
Sounds good to me. I've been working on my own fork, I have a TODO.md list that I'm about to
get started working on. I've made a few rough changes to README.md already.

robotastic, 2016-03-05:
Closing since there has been no update and the code has been updated to fix a lot of problems.

closed 2016-03-05 by robotastic