TrunkRecorder/trunk-recorder #9: Does not seem to record on analog systems

blue1sparky, 2015-04-11:
Have it working with a local P25 system. Analog smartnet system seems to not be recording.  with the following messages:

 Not recording because of priority
[2015-04-10 22:21:11.938259] [0x8831b440] [error]   	Not recording call

[ osmosdr ] No Debug Recorders Available
[2015-04-10 22:21:17.212800] [0x8831b440] [error]   	Call created for: 33968	TDMA: 32767	Encrypted: true

	TG Priority: 3 Available Recorders: 0
 Not recording because of priority
[2015-04-10 22:21:17.212908] [0x8831b440] [error]   	Not recording call

Recorders SHOULD be available according to config:

{ "sources": [
{"center": 857000000.0,
"rate": 8000000.0,
"error": -4135,
"gain": 30,
"antenna": "TX/RX",
"analogRecorders": 5,
"driver": "osmosdr",
"ifGain": 50,
"bbGain": 50,
"device": ""}
],
"system": {
	"control_channels": [856062500],
	"type": "smartnet"
	},
"talkgroupsFile": "MSP_TrunkTGs.csv"

robotastic, 2015-04-11:
Did you setup a channel list to tell the program what is analog and what is digital? I will check this out. Is anything getting recorded it all?

blue1sparky, 2015-04-11:
The only time anything gets recorded is when I assign debug recorders. I am using the channel list generated by radioreference with an added column for priority which was added via hand edit.

- Michael Leger

[This email has been sent from a mobile device. Responses may be delayed.]

<div>-------- Original message --------</div><div>From: Luke Berndt <notifications@github.com> </div><div>Date:04/10/2015  10:32 PM  (GMT-05:00) </div><div>To: robotastic/trunk-recorder <trunk-recorder@noreply.github.com> </div><div>Cc: blue1sparky <mpleger@comcast.net> </div><div>Subject: Re: [trunk-recorder] Does not seem to record on analog systems (#9) </div><div>
</div>Did you setup a channel list to tell the program what is analog and what is digital? I will check this out. Is anything getting recorded it all?

patrickfish, 2015-04-11:
I haven't tried a recent build but around the same time as the debug
recorder I lost the ability to do analog records.

robotastic, 2015-04-11:
I will go check this out. Right now it has a pretty simple test to see if it should record analog, it just checks to see if the Talkgroup from ChanList.txt has an 'A' for mode.

patrickfish, 2015-04-16:
Output:

linux; GNU C++ version 4.7.2; Boost_105500; UHD_003.008.002-83-g09c9c6bb

[2015-04-16 00:29:32.488722] [0x000007fb21482940] [info]    Center: 8.53e+08

[2015-04-16 00:29:32.488899] [0x000007fb21482940] [info]    Rate: 8e+06

[2015-04-16 00:29:32.488930] [0x000007fb21482940] [info]    Error: -4255

[2015-04-16 00:29:32.488958] [0x000007fb21482940] [info]    Gain: 18

[2015-04-16 00:29:32.488981] [0x000007fb21482940] [info]    IF Gain: 18

[2015-04-16 00:29:32.489001] [0x000007fb21482940] [info]    BB Gain: 18

[2015-04-16 00:29:32.489018] [0x000007fb21482940] [info]    Digital
Recorders: 0

[2015-04-16 00:29:32.489042] [0x000007fb21482940] [info]    Debug
Recorders: 0

[2015-04-16 00:29:32.489060] [0x000007fb21482940] [info]    Analog
Recorders: 5

[2015-04-16 00:29:32.489079] [0x000007fb21482940] [info]    driver: osmosdr

gr-osmosdr v0.1.4-9-g48045b59 (0.1.5git) gnuradio v3.7.6.1-103-g8ecfd13a
built-in source types: file osmosdr fcd rtl rtl_tcp uhd hackrf rfspace
Using HackRF One with firmware 2014.08.1
SOURCE TYPE OSMOSDR (osmosdr)
Setting sample rate to: 8e+06
Tunning to 8.52996e+08hz
[2015-04-16 00:29:33.164786] [0x000007fb21482940] [info]    Max HZ: 8.57e+08

[2015-04-16 00:29:33.164892] [0x000007fb21482940] [info]    Min HZ: 8.49e+08

Using Volk machine: sse4_2_64
[2015-04-16 00:29:33.175046] [0x000007fb21482940] [info]    Control
Channels:
[2015-04-16 00:29:33.175104] [0x000007fb21482940] [info]    8.5115e+08
[2015-04-16 00:29:33.175118] [0x000007fb21482940] [info]

[2015-04-16 00:29:33.175135] [0x000007fb21482940] [info]    Talkgroups
File: ChanList.csv

[2015-04-16 00:29:33.175164] [0x000007fb21482940] [info]    Loading
Talkgroups...

Control channel offset: -1.84998e+06
Control channel: 8.5115e+08
Decim: 740
Samples per symbol: 3.003
[2015-04-16 00:29:34.506539] [0x000007fb21482940] [error]   Call created
for: 14800 TDMA: 0 Encrypted: true

TG Priority: 3 Available Recorders: 0
 Not recording because of priority
[2015-04-16 00:29:34.506615] [0x000007fb21482940] [error]   Not recording
call

[ osmosdr ] No Debug Recorders Available
[2015-04-16 00:29:38.962731] [0x000007fb21482940] [error]   Call created
for: 14832 TDMA: 0 Encrypted: true

TG Priority: 3 Available Recorders: 0
 Not recording because of priority
[2015-04-16 00:29:38.962834] [0x000007fb21482940] [error]   Not recording
call

[ osmosdr ] No Debug Recorders Available
[2015-04-16 00:29:40.471415] [0x000007fb21482940] [error]   Call created
for: 15856 TDMA: 0 Encrypted: true

TG Priority: 3 Available Recorders: 0
 Not recording because of priority
[2015-04-16 00:29:40.471475] [0x000007fb21482940] [error]   Not recording
call

[ osmosdr ] No Debug Recorders Available
^C
 Signal caught!
[2015-04-16 00:29:41.732868] [0x000007fb21482940] [info]    stopping flow
graph

robotastic, 2015-04-18:
Hmm, since it is printing the priority and num of available recorders, that means that it is trying to do an analog recording instead of a digital one:
https://github.com/robotastic/trunk-recorder/blob/master/source.cc#L157

Is it possible that TG 15856 & 14832 are not part of your ChanList? It defaults to a Priority 3 Digital recording if a TG can't be found in ChanList. If it is in ChanList is the Mode set to A?

If not, you could try messing with this section and force an analog recording:
https://github.com/robotastic/trunk-recorder/blob/master/main.cc#L218

I will see if I can monitor a system with analog in it, or just force an analog recording.

blue1sparky, 2015-04-18:
Would it help if I sent over some recorded raw data? I might be able to get that from my hackrf and Dropbox it over.

- Michael Leger

[This email has been sent from a mobile device. Responses may be delayed.]

<div>-------- Original message --------</div><div>From: Luke Berndt <notifications@github.com> </div><div>Date:04/18/2015  3:47 PM  (GMT-05:00) </div><div>To: robotastic/trunk-recorder <trunk-recorder@noreply.github.com> </div><div>Cc: blue1sparky <mpleger@comcast.net> </div><div>Subject: Re: [trunk-recorder] Does not seem to record on analog systems (#9) </div><div>
</div>Hmm, since it is printing the priority and num of available recorders, that means that it is trying to do an analog recording instead of a digital one:
https://github.com/robotastic/trunk-recorder/blob/master/source.cc#L157

Is it possible that TG 15856 & 14832 are not part of your ChanList? It defaults to a Priority 3 Digital recording if a TG can't be found in ChanList. If it is in ChanList is the Mode set to A?

If not, you could try messing with this section and force an analog recording:
https://github.com/robotastic/trunk-recorder/blob/master/main.cc#L218

I will see if I can monitor a system with analog in it, or just force an analog recording.

robotastic, 2015-04-18:
I think I can force it by switching ChanList. It should at least help me triage it. I will give it a try tonight.

robotastic, 2016-03-05:
Closing since there has been no update and the code has been updated to fix a lot of problems.

closed 2016-03-05 by robotastic