TrunkRecorder/trunk-recorder #10: Filename timereference appears to remain constant

blue1sparky, 2015-06-23:
While running the recorder with a small modification to run with analog systems it appeared that the filenames produced all had identical timestamps in their filenames. This was remedied by moving the time(NULL) statement to occur immediately before the filename generation (sprintf) commanding.

robotastic, 2015-06-24:
Hmm... Do you mean this line: https://github.com/robotastic/trunk-recorder/blob/master/analog_recorder.cc#L170
That line looks right, it sets the variable to the current time and then converts it into the local time for the system. I wonder if locality somehow messes up.

robotastic, 2016-03-05:
Closing since there has been no update and the code has been updated to fix a lot of problems.

closed 2016-03-05 by robotastic