TrunkRecorder/trunk-recorder #11: Error returned running cmake

handwiper1, 2015-07-18:
I have installed gnuradio+your forked version of DSD, OP25(I already had and have been running a week now) and when i clone trunk-recorder, i put it into the directory ~/pybombs/src/trunk-recorder.
So I cd there and attempt to run       "              cmake -DCMAKE_PREFIX_PATH=/home/jordan/target        "   no quotes. and it gives me the following error:
 checking for module 'gnuradio-uhd'
--   package 'gnuradio-uhd' not found
-- gnuradio-uhd not found.
-- Could NOT find GNURADIO_UHD (missing:  GNURADIO_UHD_INCLUDE_DIRS) 
-- Found UHD: /home/jordan/target/lib/libuhd.so  
-- Configuring Boost C++ Libraries...
-- Could NOT find Boost
CMake Error at CMakeLists.txt:116 (message):
  Boost required to build Smartnet-Recorder

I have the version of GNU Radio and im not actually sure if I have the latest version of OP25 or not, I installed it from 3rd party git repo because their main site is down. Here is the repo i installed op25 from: https://github.com/eddof13/op25
I am using Ubuntu 14.04. 

Any help on getting this to at least configure would be GREATLY appreciated, I will supply any additional info as needed in order to solve this issue.

oddballbutler, 2015-07-24:
I am having the same issue here. Let me know if you make any progress.

handwiper1, 2015-07-25:
Ok, same to you, let me know here if you fix it as well. I'm not too much of an expert in coding at all.

robotastic, 2015-07-26:
Hi - sorry for not responding quicker, things have been busy. It looks like the cmake script is unable to find UHD. UHD is the drivers for the Ettus SDRs. Do you know if you installed them when you built Gnuradio? If you used pybombs, I think you can just do: pybombs install uhd.

If you did install UHD, it could be that the script is just old and not looking in the right place. The locations it checks are in this file:
https://github.com/robotastic/trunk-recorder/blob/master/cmake/Modules/FindUHD.cmake

It looks like it will check: 
/usr/local/lib
 /usr/lib

It could be that I need to update this file to make it more flexible and make sure it checks custom install paths. I will try to check it out tonight.

oddballbutler, 2015-07-27:
It is true that I did not have UHD installed. I built and installed UHD from source, then rebuilt gnuradio from source. Now I'm getting this error when I run cmake.

```
-- checking for module 'gnuradio-uhd'
--   found gnuradio-uhd, version 3.7.8git
-- gnuradio-uhd not found.
-- Could NOT find GNURADIO_UHD (missing:  GNURADIO_UHD_INCLUDE_DIRS) 
-- Found UHD: /usr/local/lib/libuhd.so 
-- Configuring Boost C++ Libraries...
-- Could NOT find Boost
CMake Error at CMakeLists.txt:116 (message):
  Boost required to build Smartnet-Recorder

-- Configuring incomplete, errors occurred!
```
I'm running 64bit Ubuntu 12.04
I have boost version 1.48 installed on my system

oddballbutler, 2015-07-27:
I put set(Boost_DETAILED_FAILURE_MSG TRUE) into the top level CMakeLists.txt and ran cmake. I gave me some informative output.

```
-- Configuring Boost C++ Libraries...
-- Could NOT find Boost
Boost version: 1.48.0
Boost include path: /usr/include
The following Boost libraries could not be found:
        boost_log
Some (but not all) of the required Boost libraries were found. You may need to install these additional Boost libraries. Alternatively, set BOOST_LIBRARYDIR to the directory containing Boost libraries or BOOST_ROOT to the location of Boost.
CMake Error at CMakeLists.txt:118 (message):
  Boost required to build Smartnet-Recorder

-- Configuring incomplete, errors occurred!
```
It seems that it is finding boost, but that the version of boost I am using doesn't have boost_log. From what I can tell, I cannot get libboost-log-dev from the ubuntu 12.04 repositories. Perhaps I'll try to get a more recent version of boost from source or something.

handwiper1, 2015-07-27:
I finally got this working, here is what I did. Try this and get back to me.

cd pybombs

./pybombs uninstall boost

(yes, uninstall EVERYTHING, you'll see why in a bit)

then,

sudo apt-get uninstall libboost*
then
sudo apt-get uninstall libboost*-dev
now that boost is uninstalled... reinstall everything again...

sudo apt-get install libboost1.55
HERE COMES THE TRICKY PART!  PAY ATTENTION NOW
open ~/pybombs/recipes folder
open boost.lwr using gedit
now you can either copy and paste the following in there or replace every single 1.53 in the file to 1.55 yourself:

****BEGIN COPY********

category: baseline
depends: python libbzip wget
source: wget://http://downloads.sourceforge.net/project/boost/boost/1.55.0/boost_1_55_0.tar.bz2
satisfy_deb:  (libboost1.55-dev && libboost-date-time1.55-dev && libboost-filesystem1.55-dev && libboost-program-options1.55-dev && libboost-regex1.55-dev && libboost-thread1.55-dev && libboost-test1.55-dev) || ((libboost-dev >= 1.55) && (libboost-date-time-dev >= 1.55) && (libboost-filesystem-dev >= 1.55) && (libboost-program-options-dev >= 1.55) && (libboost-regex-dev >= 1.55) && (libboost-thread-dev >= 1.55) && (libboost-test-dev >=1.55)) 
satisfy_rpm: boost-devel >= 1.55
configure {
    ./bootstrap.sh --prefix=$prefix --libdir=$prefix/lib/
}
install_static {
    ./bjam cxxflags='-fPIC' --threading=multi --layout=tagged link=static -j$makewidth install
}
install {
    ./bjam --threading=multi --layout=tagged link=shared -j$makewidth install
}

**********END COPY*******

now save that file and go back into terminal
cd pybombs
./pybombs install gr-op25
or 
./pybombs install gnuradio

once this is done continue on as you were before, this time it will find boost_log and make your makefiles right(hopefully). and you should be able to make the recorder executable...

If you have the filepatch issue as luke mentioned above,(nolt sure if I did have this problem or not), but I went into trunk-recorder/cmake/Modules and I edited each and every one of the files to search in my /home/<user>/target directory (which is where i have gnuradio,boost installed.. etc via pybombs. and it worked for me. 

Plz let me know if it worked or not!! I know when I finally got it working I didn't want to hop back in here to share my success, but rather keep it to myself... so.. please let us know.

robotastic, 2015-07-28:
Nicely done!
I have it installed on Unbuntu 14.04. That might be a different way to fix it. The default boost version for it is  1.54: http://packages.ubuntu.com/trusty/libboost-all-dev <http://packages.ubuntu.com/trusty/libboost-all-dev>
It is a little weird though because 1.53 should work with boost-log.

Boost_log isn’t critical for it, it just helps manage all of the logging. An option could be to fork the code and remove the functions and libraries. 

I just checked and I have Boost Version 1.54 installed. It looks like 
> On Jul 27, 2015, at 4:19 PM, handwiper1 <notifications@github.com> wrote:
> 
> boost_log

oddballbutler, 2015-07-29:
I finally got it to build. In the end it was the boost issue. It looks like boost_log was first introduced in version 1.54. I got version 1.58 installed using the method described by handwiper1. I was also having issues with not being able to use the ssh protocol of many of the git repositories for cloning. I'm pretty sure that ssh traffic is just being blocked at my work. Ended up having to change the pybombs recipe files for each one to get the source using http. I also had another version of gnuradio previously installed from source that I had to get rid of by building that version again ( gnuradio takes so long to build XD ).

Anyways, it has been an adventure, but I learned a lot on the way! Thanks for the help guys!

handwiper1, 2015-07-30:
No problem, ill close the issue now !

robotastic, 2015-07-30:
Great to hear everyone got it working! I will update the docs with a note
on the version of Boost.

Let me know how it is working!

closed 2016-03-05 by robotastic