TrunkRecorder/trunk-recorder #14: Is there a website that this program outputs?

handwiper1, 2015-07-31:
I felt like there was a website implemented by this as i was reading the intructions here, is this true? if so, how does it work?

and if not, is there another program you recomment i can use to set up a website?

robotastic, 2015-07-31:
Yep the source code for that website is here:
https://github.com/robotastic/smartnet-player
I have a script that encodes the recorded wav files to .mp4 and uploads
them to a Dir that the website is watching. I will add that script in a
little.

handwiper1, 2015-07-31:
ok, but i'm monitoring p25, does that make a difference?

robotastic, 2015-07-31:
It works fine with P25 - I just started out with Smartnet.

I have it up here: http://openmhz.com covering DC's P25 system

handwiper1, 2015-07-31:
yea, i'm not going to be able to do anything with that, I dont really have any coding experience and i'm not sure how that would work. I guess ill have to just keep recording things for myself. :/

closed 2016-03-05 by robotastic