TrunkRecorder/trunk-recorder #21: I can't get talkgroups to compile (or the software to actually start recording)

brendanvg, 2015-08-24:
My issue is that when i run `./recorder` it shows the block of text directly below.....

```
$ ./recorder
linux; GNU C++ version 4.9.2; Boost_105500; UHD_003.009.000-1-g71985969

[2015-08-24 01:25:39.437629] [0x000007f0fa7f1540] [info]    Center: 8.52288e+08

[2015-08-24 01:25:39.438157] [0x000007f0fa7f1540] [info]    Rate: 8e+06

[2015-08-24 01:25:39.438368] [0x000007f0fa7f1540] [info]    Error: 1400

[2015-08-24 01:25:39.438563] [0x000007f0fa7f1540] [info]    Gain: 12

[2015-08-24 01:25:39.438720] [0x000007f0fa7f1540] [info]    IF Gain: 0

[2015-08-24 01:25:39.438874] [0x000007f0fa7f1540] [info]    BB Gain: 0

[2015-08-24 01:25:39.439042] [0x000007f0fa7f1540] [info]    Digital Recorders: 4

[2015-08-24 01:25:39.439192] [0x000007f0fa7f1540] [info]    Debug Recorders: 0

[2015-08-24 01:25:39.439341] [0x000007f0fa7f1540] [info]    Analog Recorders: 0

[2015-08-24 01:25:39.439508] [0x000007f0fa7f1540] [info]    driver: usrp

[2015-08-24 01:25:39.884282] [0x000007f0fa7f1540] [error]   LookupError: KeyError: No devices found for ----->
Empty Device Address

[2015-08-24 01:25:39.884670] [0x000007f0fa7f1540] [info]    Loading Talkgroups...

Error Opening TG File: 
Segmentation fault (core dumped)
```

This is the raw text data of my ChanList.csv file.

```
31553,7b41 ,D ,OPD Patrol 1,Patrol 1 - West Dispatch ,Law Dispatch ,Oakland Police ,2
31554,7b42 ,D ,OPD Patrol 2,Patrol 2 - Central Dispatch ,Law Dispatch ,Oakland Police ,3
31555,7b43 ,D ,OPD Patrol 3,Patrol 3 - East Dispatch ,Law Dispatch ,Oakland Police ,4
31556,7b44 ,D ,OPD Patrol 4,Patrol 4 - Car-to-Car ,Law Tac ,Oakland Police ,1
31557,7b45 ,D ,OPD Patrol 5,Patrol 5 - Car-to-Car ,Law Tac ,Oakland Police ,5
31558,7b46 ,D ,OPD Serv 1,Service 1 ,Law Dispatch ,Oakland Police ,6
31559,7b47 ,D ,OPD Serv 2,Service 2 ,Law Dispatch ,Oakland Police ,10
31561,7b49 ,D ,OPD Yth Svcs,Youth Services ,Law Tac ,Oakland Police ,9
31569,7b51 ,D ,OPD Tac 1,Tac 1 ,Law Tac ,Oakland Police ,7
31570,7b52 ,D ,OPD Tac 2,Tac 2 ,Law Tac ,Oakland Police ,8
31571,7b53 ,D ,OPD Tac 3,Tac 3 ,Law Tac ,Oakland Police ,11
31572,7b54 ,D ,OPD Tac 4,Tac 4 ,Law Tac ,Oakland Police ,12
31573,7b55 ,D ,OPD Tac 5,Tac 5 ,Law Tac ,Oakland Police ,13
31577,7b59 ,D ,OPD Tac 6,Tac 6 ,Law Tac ,Oakland Police ,17
31578,7b5a ,D ,OPD Tac 7,Tac 7 ,Law Tac ,Oakland Police ,16
31579,7b5b ,D ,OPD Tac 8,Tac 8 ,Law Tac ,Oakland Police ,15
31580,7b5c ,D ,OPD Tac 9,Tac 9 ,Law Tac ,Oakland Police ,18
31581,7b5d ,D ,OPD Tac 10,Tac 10 ,Law Tac ,Oakland Police ,14
```

robotastic, 2015-08-24:
Are you using the latest code? Try doing a git pull. Someone just added code to do better parsing of the file.

robotastic, 2016-03-05:
Closing since there has been no update and the code has been updated to fix a lot of problems.

closed 2016-03-05 by robotastic