TrunkRecorder/trunk-recorder #22: UEO - Unidentified Error Object

brendanvg, 2015-08-29:
It seems to be recording but there is an error message I can't figure out and when i check the recorded files they are empty... I didn't do anything to setup my R820T dongle and antenna, just plugged it in... 

I've gotten the trunk-recorder configured thusly: 

`linux; GNU C++ version 4.9.2; Boost_105500; UHD_003.009.000-1-g71985969

[2015-08-28 20:07:55.883588] [0x000007feb9a51b40] [info]    Center: 8.53106e+08

[2015-08-28 20:07:55.884078] [0x000007feb9a51b40] [info]    Rate: 1e+06

[2015-08-28 20:07:55.884284] [0x000007feb9a51b40] [info]    Error: 0

[2015-08-28 20:07:55.884477] [0x000007feb9a51b40] [info]    Gain: 40

[2015-08-28 20:07:55.884633] [0x000007feb9a51b40] [info]    IF Gain: 0

[2015-08-28 20:07:55.884789] [0x000007feb9a51b40] [info]    BB Gain: 0

[2015-08-28 20:07:55.884991] [0x000007feb9a51b40] [info]    Digital Recorders: 1

[2015-08-28 20:07:55.885164] [0x000007feb9a51b40] [info]    Debug Recorders: 0

[2015-08-28 20:07:55.885323] [0x000007feb9a51b40] [info]    Analog Recorders: 0

[2015-08-28 20:07:55.885498] [0x000007feb9a51b40] [info]    driver: osmosdr

gr-osmosdr v0.1.4-48-g86ad5842 (0.1.5git) gnuradio 3.7.8
built-in source types: file osmosdr fcd rtl rtl_tcp uhd hackrf bladerf rfspace 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
SOURCE TYPE OSMOSDR (osmosdr)
Setting sample rate to: 1e+06
Exact sample rate is: 1000000.026491 Hz
[R82XX] PLL not locked!
Tunning to 8.53106e+08hz
[2015-08-28 20:07:57.265120] [0x000007feb9a51b40] [info]    Max HZ: 8.53606e+08

[2015-08-28 20:07:57.265350] [0x000007feb9a51b40] [info]    Min HZ: 8.52606e+08

Using Volk machine: ssse3_64
Decoding only P25 Phase 1 frames.
Enabling only GFSK modulation optimizations.
[2015-08-28 20:07:57.403611] [0x000007feb9a51b40] [info]    Control Channels: 
[2015-08-28 20:07:57.403911] [0x000007feb9a51b40] [info]    8.53106e+08 
[2015-08-28 20:07:57.404033] [0x000007feb9a51b40] [info]    

[2015-08-28 20:07:57.404190] [0x000007feb9a51b40] [info]    Talkgroups File: ChanList.csv

[2015-08-28 20:07:57.404380] [0x000007feb9a51b40] [info]    Loading Talkgroups...

Prechannel Decim: 20 Rate: 50000 system_channel_rate: 48000
After GCD - Prechannel Decim: 20 Rate: 25 system_channel_rate: 24
 FM Gain: 12 PI: 3.14159 Samples per sym: 10`

THIS IS THE ERROR MESSAGE: 

[2015-08-28 20:11:01.764587] [0x000007feb9a51b40] [error]   	Control Channel Message Decode Rate: 1/sec

how should i go about debugging this?? 
Thanks robotastic!

robotastic, 2015-09-03:
Hmm... good question. I am not sure if anything is necessarily wrong. Are you sure you have the error correction set? Try giving it a negative value instead or vis-versa. I will try it with my RTLSDR this weekend, I haven't done that in a while.

brendanvg, 2015-09-03:
something seems to be wrong because nothing is recorded to the audio files outputted and (as per my more recent issue) the void handle_message function never gets executed.... I'm not sure why this function wouldn't be getting called, but I;ll look into it... also, the next thing I'm going to do is try to set the hardware gain because in gqrx when the hardware gain control is not set to automatic the signal is much weaker.....(and perhaps i'm not getting enough message packets....)

robotastic, 2015-09-03:
Yep! Definitely set the gain. Try something high like 60 in the config file and then walk it down. Take the gain off auto in GQRX and look at what value works well there. it should be the same.

robotastic, 2016-03-05:
Closing since there has been no update and the code has been updated to fix a lot of problems.

closed 2016-03-05 by robotastic