TrunkRecorder/trunk-recorder #24: Build failure at linking step

Hainish, 2015-09-02:
I'm getting this at the end of the make step, with op25 and gr-dsd installed:

```
Linking CXX executable recorder
CMakeFiles/recorder.dir/dsd_recorder.cc.o: In function `dsd_recorder::dsd_recorder(double, double, long, long, int)':
dsd_recorder.cc:(.text+0xa3b): undefined reference to `dsd_make_block_ff(dsd_frame_mode, dsd_modulation_optimizations, int, bool, int, bool, int)'
CMakeFiles/recorder.dir/dsd_recorder.cc.o: In function `dsd_recorder::deactivate()':
dsd_recorder.cc:(.text+0x1f18): undefined reference to `dsd_block_ff::get_state()'
dsd_recorder.cc:(.text+0x2557): undefined reference to `dsd_block_ff::reset_state()'
CMakeFiles/recorder.dir/nonstop_wavfile_sink_impl.cc.o: In function `gr::blocks::nonstop_wavfile_sink_impl::open(char const*)':
nonstop_wavfile_sink_impl.cc:(.text+0x89c): undefined reference to `gr::blocks::wavheader_write(_IO_FILE*, unsigned int, int, int)'
CMakeFiles/recorder.dir/nonstop_wavfile_sink_impl.cc.o: In function `gr::blocks::nonstop_wavfile_sink_impl::close_wav()':
nonstop_wavfile_sink_impl.cc:(.text+0x9b7): undefined reference to `gr::blocks::wavheader_complete(_IO_FILE*, unsigned int)'
CMakeFiles/recorder.dir/nonstop_wavfile_sink_impl.cc.o: In function `gr::blocks::nonstop_wavfile_sink_impl::work(int, std::vector<void const*, std::allocator<void const*> >&, std::vector<void*, std::allocator<void*> >&)':
nonstop_wavfile_sink_impl.cc:(.text+0xd5f): undefined reference to `gr::blocks::wav_write_sample(_IO_FILE*, short, int)'
collect2: error: ld returned 1 exit status
make[2]: *** [recorder] Error 1
make[1]: *** [CMakeFiles/recorder.dir/all] Error 2
make: *** [all] Error 2
```

robotastic, 2015-09-03:
You have to use the version of gr-dsd I have modified. It is available here:
https://github.com/robotastic/gr-dsd

That should fix it! Good luck

closed 2015-09-03 by robotastic

Hainish, 2015-09-05:
Great, thanks, that cleared up that problem.  Now I have this one though:

```
Linking CXX executable recorder
CMakeFiles/recorder.dir/nonstop_wavfile_sink_impl.cc.o: In function `gr::blocks::nonstop_wavfile_sink_impl::open(char const*)':
nonstop_wavfile_sink_impl.cc:(.text+0x89c): undefined reference to `gr::blocks::wavheader_write(_IO_FILE*, unsigned int, int, int)'
CMakeFiles/recorder.dir/nonstop_wavfile_sink_impl.cc.o: In function `gr::blocks::nonstop_wavfile_sink_impl::close_wav()':
nonstop_wavfile_sink_impl.cc:(.text+0x9b7): undefined reference to `gr::blocks::wavheader_complete(_IO_FILE*, unsigned int)'
CMakeFiles/recorder.dir/nonstop_wavfile_sink_impl.cc.o: In function `gr::blocks::nonstop_wavfile_sink_impl::work(int, std::vector<void const*, std::allocator<void const*> >&, std::vector<void*, std::allocator<void*> >&)':
nonstop_wavfile_sink_impl.cc:(.text+0xd5f): undefined reference to `gr::blocks::wav_write_sample(_IO_FILE*, short, int)'
collect2: error: ld returned 1 exit status
make[2]: *** [recorder] Error 1
make[1]: *** [CMakeFiles/recorder.dir/all] Error 2
make: *** [all] Error 2
```

robotastic, 2015-09-05:
That is really weird. The problem is that it is failing during the Linking step. This means that GCC/G++ is not finding the libraries. It would try doing the cmake steps again. Make sure the  right gnuradio libraries are in your lib path. 

Did you do a make install for gnuradio? If you did pybombs and used a non-standard path, make sure you specify it when you run cmake for trunk recorder:
cmake -DCMAKE_PREFIX_PATH=/path/to/GR/install   .