TrunkRecorder/trunk-recorder #27: Build errors on Ubuntu 14.04

devnulling, 2015-10-19:
I'm running into this error when running the 'make' command.

Setup is Ubuntu 14.04, pybombs used to install gnruadio&friends. 

Any suggestions?

```
node@computer:~/sdr/trunk-recorder$ make
[  6%] Building CXX object CMakeFiles/recorder.dir/main.cc.o
/home/node/sdr/trunk-recorder/main.cc: In function ‘void monitor_messages()’:
/home/node/sdr/trunk-recorder/main.cc:481:15: error: reference to ‘queue’ is ambiguous
         msg = queue->delete_head();
               ^
/home/node/sdr/trunk-recorder/main.cc:76:21: note: candidates are: gr::msg_queue::sptr queue
 gr::msg_queue::sptr queue;
                     ^
In file included from /usr/include/c++/4.8/queue:64:0,
                 from /usr/include/boost/assign/std/queue.hpp:21,
                 from /usr/include/boost/assign/std.hpp:23,
                 from /usr/include/boost/assign.hpp:19,
                 from /home/node/sdr/target/include/uhd/usrp/gpio_defs.hpp:22,
                 from /home/node/sdr/target/include/uhd/usrp/dboard_iface.hpp:25,
                 from /home/node/sdr/target/include/uhd/usrp/multi_usrp.hpp:45,
                 from /home/node/sdr/target/include/gnuradio/uhd/usrp_block.h:28,
                 from /home/node/sdr/target/include/gnuradio/uhd/usrp_source.h:26,
                 from /home/node/sdr/trunk-recorder/source.h:7,
                 from /home/node/sdr/trunk-recorder/main.cc:46:
/usr/include/c++/4.8/bits/stl_queue.h:93:11: note:                 template<class _Tp, class _Sequence> class std::queue
     class queue
           ^
/home/node/sdr/trunk-recorder/main.cc:471:12: warning: unused variable ‘lastUnitCheckTime’ [-Wunused-variable]
     time_t lastUnitCheckTime = time(NULL);
            ^
/home/node/sdr/trunk-recorder/main.cc: In function ‘bool monitor_system()’:
/home/node/sdr/trunk-recorder/main.cc:543:121: error: reference to ‘queue’ is ambiguous
             smartnet_trunking = make_smartnet_trunking(control_channel_freq, source->get_center(), source->get_rate(),  queue);
                                                                                                                         ^
/home/node/sdr/trunk-recorder/main.cc:76:21: note: candidates are: gr::msg_queue::sptr queue
 gr::msg_queue::sptr queue;
                     ^
In file included from /usr/include/c++/4.8/queue:64:0,
                 from /usr/include/boost/assign/std/queue.hpp:21,
                 from /usr/include/boost/assign/std.hpp:23,
                 from /usr/include/boost/assign.hpp:19,
                 from /home/node/sdr/target/include/uhd/usrp/gpio_defs.hpp:22,
                 from /home/node/sdr/target/include/uhd/usrp/dboard_iface.hpp:25,
                 from /home/node/sdr/target/include/uhd/usrp/multi_usrp.hpp:45,
                 from /home/node/sdr/target/include/gnuradio/uhd/usrp_block.h:28,
                 from /home/node/sdr/target/include/gnuradio/uhd/usrp_source.h:26,
                 from /home/node/sdr/trunk-recorder/source.h:7,
                 from /home/node/sdr/trunk-recorder/main.cc:46:
/usr/include/c++/4.8/bits/stl_queue.h:93:11: note:                 template<class _Tp, class _Sequence> class std::queue
     class queue
           ^
/home/node/sdr/trunk-recorder/main.cc:549:111: error: reference to ‘queue’ is ambiguous
             p25_trunking = make_p25_trunking(control_channel_freq, source->get_center(), source->get_rate(),  queue);
                                                                                                               ^
/home/node/sdr/trunk-recorder/main.cc:76:21: note: candidates are: gr::msg_queue::sptr queue
 gr::msg_queue::sptr queue;
                     ^
In file included from /usr/include/c++/4.8/queue:64:0,
                 from /usr/include/boost/assign/std/queue.hpp:21,
                 from /usr/include/boost/assign/std.hpp:23,
                 from /usr/include/boost/assign.hpp:19,
                 from /home/node/sdr/target/include/uhd/usrp/gpio_defs.hpp:22,
                 from /home/node/sdr/target/include/uhd/usrp/dboard_iface.hpp:25,
                 from /home/node/sdr/target/include/uhd/usrp/multi_usrp.hpp:45,
                 from /home/node/sdr/target/include/gnuradio/uhd/usrp_block.h:28,
                 from /home/node/sdr/target/include/gnuradio/uhd/usrp_source.h:26,
                 from /home/node/sdr/trunk-recorder/source.h:7,
                 from /home/node/sdr/trunk-recorder/main.cc:46:
/usr/include/c++/4.8/bits/stl_queue.h:93:11: note:                 template<class _Tp, class _Sequence> class std::queue
     class queue
           ^
/home/node/sdr/trunk-recorder/main.cc: In function ‘int main()’:
/home/node/sdr/trunk-recorder/main.cc:565:5: error: reference to ‘queue’ is ambiguous
     queue = gr::msg_queue::make(100);
     ^
/home/node/sdr/trunk-recorder/main.cc:76:21: note: candidates are: gr::msg_queue::sptr queue
 gr::msg_queue::sptr queue;
                     ^
In file included from /usr/include/c++/4.8/queue:64:0,
                 from /usr/include/boost/assign/std/queue.hpp:21,
                 from /usr/include/boost/assign/std.hpp:23,
                 from /usr/include/boost/assign.hpp:19,
                 from /home/node/sdr/target/include/uhd/usrp/gpio_defs.hpp:22,
                 from /home/node/sdr/target/include/uhd/usrp/dboard_iface.hpp:25,
                 from /home/node/sdr/target/include/uhd/usrp/multi_usrp.hpp:45,
                 from /home/node/sdr/target/include/gnuradio/uhd/usrp_block.h:28,
                 from /home/node/sdr/target/include/gnuradio/uhd/usrp_source.h:26,
                 from /home/node/sdr/trunk-recorder/source.h:7,
                 from /home/node/sdr/trunk-recorder/main.cc:46:
/usr/include/c++/4.8/bits/stl_queue.h:93:11: note:                 template<class _Tp, class _Sequence> class std::queue
     class queue
           ^
make[2]: *** [CMakeFiles/recorder.dir/main.cc.o] Error 1
make[1]: *** [CMakeFiles/recorder.dir/all] Error 2
make: *** [all] Error 2
node@computer:~/sdr/trunk-recorder$

```

devnulling, 2015-10-20:
I was able to resolve this by renaming the 'queue' variable.  I also had to modify the paths to find OP25 - https://github.com/devnulling/trunk-recorder/commit/f58ae86c26d2e7060695c8a500535b2886348eb9

robotastic, 2015-10-20:
> I was able to resolve this by renaming the 'queue' variable. I also had to modify the paths to find OP25 - devnulling@f58ae86 <https://github.com/devnulling/trunk-recorder/commit/f58ae86c26d2e7060695c8a500535b2886348eb9>

Weird! That sounds like the write thing to do. I don’t know why that happened. Maybe you have some additional Boost libraries installed. I will try to recreate it and put in a fix.

 - Luke

brendanvg, 2015-10-30:
Check it out y'all, I wrote a compiling script for a vagrant virtual box (ubuntu 14.04) if y'all are interested (you won't have to worry about building to your specific machine anymore): https://github.com/brendanvg/vagrantScanner

blue1sparky, 2015-10-31:
I've heard mixed reviews with regard to using anything with a high data rate through USB on a virtual machine. I'm definitely going to give this a try this weekend. Have you tried it at all with non-P25 systems?

brendanvg, 2015-10-31:
No I haven't tried it with non-P25 systems.... Please let me know how your
use with it goes...I'm very interested.
Also, don't hesitate to ask any questions if you have them for me...
-Brendan

robotastic, 2016-03-05:
Closing since there has been no update and the code has been updated to fix a lot of problems.

closed 2016-03-05 by robotastic

Rylvir, 2016-04-05:
I ran into this same error during build in Linux Mint 17.3. followed instructions above and only renamed the queue > new_queue in main.cc and build finished successfully with the exception of these two errors:

-/main.cc: In function ‘void unit_check()’:
-/main.cc:507:30: warning: ignoring return value of ‘int system(const char*)’, declared with attribute warn_unused_result [-Wunused-result]
         system(shell_command);
                              ^
-/call.cc: In member function ‘void Call::end_call()’:
-/call.cc:75:38: warning: ignoring return value of ‘int system(const char*)’, declared with attribute warn_unused_result [-Wunused-result]
                 system(shell_command);
                                      ^