TrunkRecorder/trunk-recorder #28: Empty .wav files..... acts like its recording, but it's not working

brendanvg, 2015-10-30:
I've spent the past month working with this software and I'm fairly sure I've configured everything correctly, however the audio files it outputs are empty.... help?

The dependencies/instance of trunk recorder I'm using comes from this repository:
https://github.com/brendanvg/vagrantScanner
This is using Vagrant...In order to address problems I came across while compiling/building trunk-recorder I wrote this VagrantFile and bootsrap.sh (which is a script which compiles the software to a virtual machine so that people can share/collaborate on software without troubleshooting each computer's dependencies/config..)... 
My virtualbox settings is set to Enable USB Controller USB 2.0 (EHCI) Controller, which points to Realtek RTL2838UHIDIR [0100].... I don't think this would be the problem...

The ChanList.csv I used is located in:
https://github.com/brendanvg/Oakland

My config.json file looks like: 
{
    "sources": [{
        "center": 853266800,
        "rate": 8000000.0,
        "error": 0,
        "gain": 60,
        "antenna": "TX/RX",
        "digitalRecorders": 1,
        "driver": "osmosdr",
        "device": ""
    }],
    "system": {
        "control_channels": [853266800],
        "type": "p25"
    },
    "talkgroupsFile": "ChanList.csv"
}

This is the Oakland specific data I'm working with: http://www.radioreference.com/apps/db/?sid=6077
but when looking at the spectrum in GQRX I found that 853266800 and 853105700 are a strong (seemingly digital signal) that are always on.... and I figured they might be "close enough" to be the two control channels specified in http://www.radioreference.com/apps/db/?sid=6077... -> 853.33750c and 853.46250c.....

After all this, it does write .wav files, but when I inspect them, they are empty (they're all 44bytes no matter the time I keep it on)
I keep the recording on between 30-60 sec....

Some things I can think of: 
in GQRX it's important for my hardware gain to be set, but it doesn't seem like I have this option in this software...is there anyway it can be added as a config attribute? I tried setting gain to 60 and even 80 with still no result... what type of gain is this???

Am I on the wrong frequency? I've tried many, so I wouldn't say this is a for sure problem...but would it be possible for you to take a gqrx screenshot of what a control channel should look like (its frequency, amplitude....the info shown in the waterfall display in gqrx)....

Thanks for all the help! Let me know what you think of the Vagrant solution and if I can be of help in any other way.
Sincerely,
Brendanvg

robotastic, 2015-10-30:
Good work in getting everything up and running! That is half the battle. I
am going to go checkout the vagrant stuff this weekend.

 So the problem is that RTL-SDR will also have a tuning error. The amount
of the error will also change over time because it is influenced by heat.
Once the RTLSDR has been on for a while, the error should be sort of
stable. The delta between what the control channel freq should be and what
you are finding in GQRX is the amount of error. It looks like the error
might be 7070. I am never sure if that should be 7070 or -7070. Try adding
one and then the other to the config file under error. Also set the control
channel to the correct freq listed on Radio Ref. The software will just
take the posted freq, add the error and then tune to the actual freq.

You also want the center freq to be about 50k or more away from any of the
freq you are going to be looking at. There is a spike at the center.

The Rate is also wrong. RTLSDR have a max of 2000000 I think. It should
just set it at the max I think.

brendanvg, 2015-11-01:
By the way.... the username:password is vagrant:vagrant

brendanvg, 2015-11-03:
I explored the spectrum in GQRX and it is -7070 .... so now my control channel is tuned to what is specified in Radio Ref and with the -7070 error, it lands right on top of what seems to be a digital control channel.... good tip! thanks.

It's still not working though....
Perhaps I have the center off.... If my control channel is set to 8533750, then should my center be at 8538750 (+50000)??

Also, setting DigitalRecorders to 1 is the safest when trying to get it to run at first, correct? (and then once it runs we can up this to a better number according to CPU capacity)?  

Thanks!

robotastic, 2015-11-03:
Cool!

Since it is decoding the control channel, that means the error is correct and that also means that the audio should be tuning to the correct frequency.

If you change the 0’s on this line to 1 and recompile, it should turn on some better debug messages for DSD.
https://github.com/robotastic/trunk-recorder/blob/master/dsd_recorder.cc#L60 <https://github.com/robotastic/trunk-recorder/blob/master/dsd_recorder.cc#L60>

robotastic, 2016-03-05:
Closing since there has been no update and the code has been updated to fix a lot of problems.

closed 2016-03-05 by robotastic