TrunkRecorder/trunk-recorder #43: lots of 0 byte and 44 byte wav files, no actual recordings

thefinn93, 2016-05-02:
I'm trying to get this working in my local area (Seattle, WA), which, according to [this](http://www.radioreference.com/apps/db/?sid=604), uses a "Motorola Type II SmartZone". A little Googling tells me that SmartZone and SmartNet are effectively the same thing, so this *should* work? But I wouldn't be surprised if I got that wrong. 

Anyway, what I do know: I can tune my RTLSDRs using GGRX and similar to the frequencies listed on [the radioreference.com page](http://www.radioreference.com/apps/db/?sid=604) and can clearly see something on the control channel, and hear people talking on some of the other frequencies. I set up trunk recorder with two of the SDRs using the config below. I have also tried setting the errors to positive instead of negative, it doesn't seem to seem to affect much.

```
{
  "sources": [
    {
      "center": 852177000,
      "rate": 2048000,
      "error": -10725,
      "gain": 50,
      "digitalRecorders": 2,
      "analogRecorders": 2,
      "driver": "osmosdr",
      "device": "4",
      "serialNumber": "00000004"
    },
    {
      "center": 854685500,
      "rate": 2048000,
      "error": -25747,
      "gain": 50,
      "digitalRecorders": 2,
      "analogRecorders": 2,
      "driver": "osmosdr",
      "device": "0",
      "serialNumber": "00000000"
    }
  ],
  "system": {
    "control_channels": [851412500],
    "type": "smartnet"
  },
  "talkgroupsFile": "talkgroups.csv"
}
```

Any help/ideas/things to try?

thefinn93, 2016-05-03:
I should mention that I also get a lot of this sort of thing in stdout:

```
[2016-05-03 11:37:01.819434] [0x000007f8a1b69b78] [error]       Control Channel Message Decode Rate: 0.142857/sec
[2016-05-03 11:38:07.831536] [0x000007f8a1b69b78] [error]       Control Channel Message Decode Rate: 0.0151515/sec
```

zeryl, 2016-05-03:
When I see this, it's due to a lack of "lock" on that channel. Normally I'm
either too low on gain, too far off on the frequency/correction, or trying
to watch too much bandwidth.

thefinn93, 2016-05-03:
hrm alright. Well I've amped up the gain pretty high, and I think the error is correct, so maybe I'm trying to watch too much bandwidth? How would I set that? I have RTL2838s, which can (supposedly) do ~2MHz each, but I'm also not seeing anywhere that I can actually set that, unless it's the `rate` value, but I thought that was the number of samples per second that it takes?

dreinhold, 2016-05-03:
If you are seeing the 
Control Channel Message Decode Rate: 0.142857/sec
error then you are not getting anything. That message come up for values < 10.0

You should be seeing these for voice calls: 
[2016-05-02 20:18:19.912545] [0x00007f1c7c23e780] [error]       Trying to record TDMA: 8.57713e+08 For TG: 4144

For the bandwidth, yes change the rate, go down to like 1M until you can lock onto the control channel.

robotastic, 2016-05-04:
Yep - folks are right. If the decode rate message is showing up, that means that the signal is not being received correctly. It sounds like you have things configured correctly. I haven't had a problem running the RTLSDR with 2MHz. The rate value is where you set that. Are there any other systems nearby you could try instead?

thefinn93, 2016-05-04:
Any other trunked radio systems? I'm not aware of them. I have a web SDR thing I could setup if people want to check it out...

robotastic, 2016-05-04:
Sorry - I meant are there any other Smartnet radio systems nearby that you could try listening to instead?

thefinn93, 2016-05-04:
yeah none that I know of. I'll look around

thefinn93, 2016-05-22:
So, after much time wasting and frustration I have determined it to be some voodoo with the computer I was using. I'm not sure if it was the OS (Ubuntu 15.04) or the hardware, but I tried again in a different box (a Mac Mini running Debian Sid) and it worked! So I'm gonna close this.

closed 2016-05-22 by thefinn93