TrunkRecorder/trunk-recorder #45: Unable to setup Control Channel Monitor

jonathanendersby, 2016-05-07:
Hi

I'm attempting to get this up and running with an rtlsdr dongle but am getting the aforementioned issue.

My config is as such:

<pre>{
	"sources": [{
		"center": 857800000.0,
		"rate": 2048000.0,
		"error": 1800,
		"ppm": 0,
		"gain": 42,
		"digitalRecorders": 4,
		"driver": "osmosdr",
		"device": ""
	}],
	"system": {
		"control_channels": [855462500],
		"type": "smartnet"
	},
	"talkgroupsFile": "ChanList.csv"
}</pre>

And when I try and run I get the following:

<pre>jonathan@tesla:~/trunk-recorder$ ./recorder 
linux; GNU C++ version 4.8.4; Boost_105400; UHD_003.010.git-202-g9e0861e1

[2016-05-07 17:26:56.251417] [0xbb603780] [info]    Control Channels: 
[2016-05-07 17:26:56.251560] [0xbb603780] [info]    8.55462e+08 
[2016-05-07 17:26:56.251583] [0xbb603780] [info]    
[2016-05-07 17:26:56.251605] [0xbb603780] [info]    Talkgroups File: ChanList.csv
[2016-05-07 17:26:56.251665] [0xbb603780] [info]    Center: 8.578e+08
[2016-05-07 17:26:56.251695] [0xbb603780] [info]    Rate: 2.048e+06
[2016-05-07 17:26:56.251733] [0xbb603780] [info]    Error: 1800
[2016-05-07 17:26:56.251760] [0xbb603780] [info]    PPM Error: 0
[2016-05-07 17:26:56.251785] [0xbb603780] [info]    Gain: 42
[2016-05-07 17:26:56.251806] [0xbb603780] [info]    IF Gain: 0
[2016-05-07 17:26:56.251825] [0xbb603780] [info]    BB Gain: 0
[2016-05-07 17:26:56.251845] [0xbb603780] [info]    Digital Recorders: 4
[2016-05-07 17:26:56.251865] [0xbb603780] [info]    Debug Recorders: 0
[2016-05-07 17:26:56.251884] [0xbb603780] [info]    Analog Recorders: 0
[2016-05-07 17:26:56.251902] [0xbb603780] [info]    driver: osmosdr
[2016-05-07 17:26:56.251921] [0xbb603780] [info]    Source Device not specified
gr-osmosdr v0.1.4-72-g164a09fc (0.1.5git) gnuradio 3.7.9.2
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy redpitaya 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
[2016-05-07 17:26:57.542556] [0xbb603780] [info]    SOURCE TYPE OSMOSDR (osmosdr)
[2016-05-07 17:26:57.542609] [0xbb603780] [info]    Setting sample rate to: 2.048e+06
[R82XX] PLL not locked!
[2016-05-07 17:26:57.660386] [0xbb603780] [info]    Actual sample rate: 2.048e+06
[2016-05-07 17:26:57.660433] [0xbb603780] [info]    Tunning to 8.57802e+08hz
[2016-05-07 17:26:57.711636] [0xbb603780] [info]    Max HZ: 8.58824e+08
[2016-05-07 17:26:57.711683] [0xbb603780] [info]    Min HZ: 8.56776e+08
Using Volk machine: avx_64_mmx_orc
Project 25 IMBE Encoder/Decoder Fixed-Point implementation
Developed by Pavel Yazev E-mail: pyazev@gmail.com
Version 1.0 (c) Copyright 2009
This program comes with ABSOLUTELY NO WARRANTY.
This is free software, and you are welcome to redistribute it
under certain conditions; see the file ``LICENSE'' for details.
[2016-05-07 17:26:57.745396] [0xbb603780] [info]    Loading Talkgroups...

[2016-05-07 17:26:57.746614] [0xbb603780] [info]    Read 135 talkgroups.
[2016-05-07 17:26:57.746656] [0xbb603780] [info]    Unable to setup Control Channel Monitor</pre>

Have you got any ideas of what I might be doing wrong?

dreinhold, 2016-05-07:
You need to move your center frequency closer to your control channel.
With your current center frequency you can only receive between the two listed frequencies.
[2016-05-07 17:26:57.711636] [0xbb603780] [info]    Max HZ: 8.58824e+08
[2016-05-07 17:26:57.711683] [0xbb603780] [info]    Min HZ: 8.56776e+08

robotastic, 2016-05-08:
I think Dylan is right. It looks like the center frequency is too far from the control channel. It should be at most only 1Mhz away since you have the rate set to 2MHz.

jonathanendersby, 2016-05-09:
Ah, that makes complete sense. I'm going to try again later and PR a single RTL-SDR config file (since I think that's where most people will start.) Perhaps there is room to detect this scenario and alert the user?

robotastic, 2016-05-09:
Good call! I am didn't think there would actually be that many people using
the code. I will try to make a pass and add in more helpful error messages
and checks.

closed 2017-03-10 by robotastic