TrunkRecorder/trunk-recorder #46: Any squelch control

dreinhold, 2016-05-07:
I see some commented out code with a squelch block.
I get a good 3 seconds of static after each transmission (smartnet analog), looking for options

robotastic, 2016-05-08:
I think uncommenting that should make it work. The settings may need to be adjusted. I had it commented out because there was an error in gnuradio and the Power Squelch block was not being built right. I think it has been fixed. I don’t have an analog system to test on. If you can try it out, that would be great!

dreinhold, 2016-05-08:
Okay was going to go that route, but was not sure why it was commented out..

robotastic, 2016-05-09:
I actually got a random, anonymous web submission from someone who fixed the squelch. I just created a Squelch branch. Give it a try and let me know how it works. I don't have a nearby analog system, but I at least got it to compile

dreinhold, 2016-05-09:
That did it. Sounds great now...

closed 2016-05-15 by robotastic