TrunkRecorder/trunk-recorder #50: Control Channel Message Decode Rate:

ralph-shepard, 2016-06-05:
I am for ever getting this error
`[0xb404f700] [error]   	Control Channel Message Decode Rate: 1/sec`

I know this has to do with tuning but I'm not clever enough to work out what the correct tuning is.

This is my config.json
`{
    "sources": [{
	"center": 85150000.0,
        "rate": 2399999.0,
        "squelch": -25,
        "error":  -850,
        "gain": 45,
        "digitalRecorders": 2,
        "driver": "osmosdr"
            }],
    "system": {
        "control_channels": [85132500],
        "type": "p25",
        "modulation": "QPSK"
    },
    "talkgroupsFile": "ChanList.csv"
}`

And this is what works with op25: 
`./scope.py --args rtl=0 -N LNA:46 -f 851.38100e6 -S 2400000 -q -1 -v 0 -o 5000 -V -T marrs-trunk.tsv`

With the above command line I can pretty much fire up op25 on a warm rtl-sdr and start decoding with no adjustments.    Not sure how to translate that to the config.json?

dreinhold, 2016-06-06:
Is 851.381 the true frequency for a voice channel on that system? If not you will need to get the real freq to figure out the error between the two, I ass you have -850
Also in scope you are passing a FREQ_CORR of -1, to do the same in config.json add a "ppm" : -1
Should probably stick with fixing the error via PPM or error.

ralph-shepard, 2016-06-06:
Smack my forehead...  that is not a voice freq on the system....  the control channel is 851.32500

Not sure how 851.381 snuck in there....   But I think you've told me what I need to know...  the error I'm getting is a tuning error that I just have to fiddle with and get the tuning correct?

dreinhold, 2016-06-06:
It looks like you have the control channel in the right spot in the config. I was just asking about the freq in scope to see if you were correcting for error offset by setting the frequency.
Right now my best guess is your error correction/ppm offsets is what you need to work with.

robotastic, 2016-06-07:
I would try setting the error at 850 instead of -850. Also try setting the sampling rate to 2048000. Having a gain of 45 also seems a bit high. I would try backing that down a little.

My guess is that it is the tuning though. Try using GQRX to see how far off it is.

ralph-shepard, 2016-06-08:
So in GQRX a -2 ppm seems to get me right on the money...  but when I add "PPM" :  -2, and remove the "error": entry I still have the same problem....  
according to GQRX I have to adjust 800hz to get spot on. so to tune to the center of 851.325.000 tune to 851.325.800 and I'm spot on.  that's with a zero PPM adjustment

I'm rather lost as to where to go from here

robotastic, 2016-06-08:
Did you try setting the error in the config file to be 800?
Do you have a link to the system in Radio Reference that you could share?
Also try FSK4 for the modulation instead of QPSK.

ralph-shepard, 2016-06-08:
The system is here
http://www.radioreference.com/apps/db/?sid=6916
more specifically
http://www.radioreference.com/apps/db/?siteId=18377

I've tried both setting the error in config to 800 and setting the modulation to FSK4 same result

robotastic, 2016-06-09:
The only other thing I can think of is trying the other control channels. Those are the one that are red.

It is sort of weird that it is not working. The code is pretty similar to OP25. There is no thing going on that is that different.

normFranklin, 2016-06-12:
It looks like you left a zero out of your center frequency and control frequency.  Set to 85.1 MHz not 851 MHz.  Unfortunately, two (consistent) wrongs make a valid setup.

robotastic, 2016-06-14:
It looks like @normFranklin has it! You have set it to 85.1MHz for the channels. Try adding another 0

ralph-shepard, 2016-06-22:
Well, I finally pulled my face outta my hands and tried to this, still no joy current config looks like this:

`{
    "sources": [{
	"center": 851500000.0,	
	"rate": 2048000,
        "error": 800,
        "gain": 30,
        "digitalRecorders": 2,
        "driver": "osmosdr"
            }],
    "system": {
        "control_channels": [851325000],
        "type": "p25",
        "modulation": "QPSK"
    },
    "talkgroupsFile": "ChanList.csv"
}

`

normFranklin, 2016-06-23:
Here's a question. When you measured your tuning error as 800 Hz, was the PPM correction of the tuner set to 0?  

If not, you should either:
1) determine the frequency correction needed in Hz to properly tune to near the center of the control channel when the PPM correction of the tuner is set to zero, or,
2) determine what PPM correction of the tuner gets the frequency close to the  control channel, then in your config.json file set "ppm":  to the PPM correction and set "error": 0,  
 
My (cheap) rtl-sdr's have PPM corrections of between 50 to 57 ppm.  Your config file implies that the frequency correction of your tuner is less than 1 ppm.

Good luck

robotastic, 2017-02-27:
Please reopen if this is still an issue

closed 2017-02-27 by robotastic

shwetapalande, 2017-08-15:
I am stuck up with a similar error. I am trying to run the trunk-recorder with a LimeSDR. All the packages and dependencies seem to be properly installed but the trunk-recorder is not able to scan the channels and it always gives a decode rate of 0/sec. I am not able to figure out what is going wrong here. 
Following is the output after running the recorder

`linux; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_003.010.002.000-release
[2017-08-15 00:55:52.554619] (info)   Using Config file: ./config.json
[2017-08-15 00:55:52.557723] (info)   System Type: p25
[2017-08-15 00:55:52.558005] (info)   Control Channels: 
[2017-08-15 00:55:52.558638] (info)   7.73469e+08 
[2017-08-15 00:55:52.559486] (info)   7.73744e+08 
[2017-08-15 00:55:52.560032] (info)   7.74219e+08 
[2017-08-15 00:55:52.560377] (info)   7.74494e+08 
[2017-08-15 00:55:52.560828] (info)   
-------------------------------------
System Number: 1
[2017-08-15 00:55:52.561288] (info)   Short Name: ebrcs_alameda
[2017-08-15 00:55:52.561484] (info)   API Key: 
[2017-08-15 00:55:52.561627] (info)   Upload Script: 
[2017-08-15 00:55:52.561859] (info)   Call Log: true
[2017-08-15 00:55:52.562058] (info)   Audio Archive: true
[2017-08-15 00:55:52.562204] (info)   Loading Talkgroups...
[2017-08-15 00:55:52.564223] (info)   Read 19 talkgroups.
[2017-08-15 00:55:52.564469] (info)   Talkgroups File: ebrcs_alameda.csv
[2017-08-15 00:55:52.564621] (info)   Record Unkown Talkgroups: true
[2017-08-15 00:55:52.564789] (info)   
[2017-08-15 00:55:52.565187] (info)   Capture Directory: /home/wrat/trunk-recorder/trunk-build
[2017-08-15 00:55:52.565374] (info)   Upload Server: encode-upload.sh
[2017-08-15 00:55:52.565550] (info)   Default Mode: digital
[2017-08-15 00:55:52.565728] (info)   Call Timeout (seconds): 3
[2017-08-15 00:55:52.565909] (info)   Log to File: false
[2017-08-15 00:55:52.566356] (info)   Center: 7.71794e+08
[2017-08-15 00:55:52.566594] (info)   Rate: 8e+06
[2017-08-15 00:55:52.566828] (info)   Error: 0
[2017-08-15 00:55:52.567023] (info)   PPM Error: 0
[2017-08-15 00:55:52.567287] (info)   Gain: 30
[2017-08-15 00:55:52.567514] (info)   IF Gain: 0
[2017-08-15 00:55:52.567700] (info)   BB Gain: 0
[2017-08-15 00:55:52.567887] (info)   LNA Gain: 0
[2017-08-15 00:55:52.568078] (info)   MIX Gain: 0
[2017-08-15 00:55:52.568270] (info)   Squelch: 0
[2017-08-15 00:55:52.568464] (info)   Idle Silence: false
[2017-08-15 00:55:52.568632] (info)   Digital Recorders: 10
[2017-08-15 00:55:52.568835] (info)   Debug Recorders: 0
[2017-08-15 00:55:52.569009] (info)   Analog Recorders: 0
[2017-08-15 00:55:52.569194] (info)   Driver: osmosdr
[2017-08-15 00:55:52.569434] (info)   Modulation: qpsk
[2017-08-15 00:55:52.569632] (info)   Source Device: soapy=0,driver=lime
gr-osmosdr v0.1.4-98-gc653754d (0.1.5git) gnuradio 3.7.10
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy soapy redpitaya 
[INFO] Make connection: 'LimeSDR-USB [USB 2.0] 9062A00CE0D1A'
[INFO] Estimated reference clock 30.7191 MHz
[INFO] Selected reference clock 30.720 MHz
[INFO] Device name: LimeSDR-USB
[INFO] Reference: 30.72 MHz
[INFO] Init LMS7002M(0)
[INFO] Ver=7, Rev=1, Mask=1
[INFO] LMS7002M calibration values caching Enable
[INFO] LMS7002M cache /home/wrat/.limesuite/LMS7002M_cache_values.db
[INFO] Rx Filter calibrated from cache
[INFO] Tx Filter calibrated from cache
[INFO] Rx Filter calibrated from cache
[INFO] Tx Filter calibrated from cache
[2017-08-15 00:56:05.207768] (info)   SOURCE TYPE OSMOSDR (osmosdr)
[2017-08-15 00:56:05.208022] (info)   Setting sample rate to: 8e+06
[2017-08-15 00:56:07.150319] (info)   Actual sample rate: 8e+06
[2017-08-15 00:56:07.150635] (info)   Tunning to 7.71794e+08hz
SetFrequency using cache values vco:1, csw:193
[2017-08-15 00:56:07.192633] (info)   Gain Stage: LNA supported values: 0 1 2 3 4 5 6 7 8 9 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 26 27 28 29 30 
[2017-08-15 00:56:07.193424] (info)   Gain Stage: TIA supported values: 0 1 2 3 4 5 6 7 8 9 10 11 12 
[2017-08-15 00:56:07.194585] (info)   Gain Stage: PGA supported values: -12 -11 -10 -9 -8 -7 -6 -5 -4 -3 -2 -1 0 1 2 3 4 5 6 7 8 9 10 11 12 13 14 15 16 17 18 19 
[2017-08-15 00:56:07.194800] (info)   Max HZ: 7.75794e+08
[... 38 lines cut ...]
I: |   0 | 2047 | 0
Q: |   0 | 2047 |
############################################################
[INFO] L
[2017-08-15 00:56:14.020105] (error)   [ebrcs_alameda] Retuning to Control Channel: 7.73744e+08
[2017-08-15 00:56:14.020480] (info)      - System Source - Min Freq: 7.67794e+08 Max Freq: 7.75794e+08
[2017-08-15 00:56:14.022823] (error)   [ebrcs_alameda]   Control Channel Message Decode Rate: 0/sec, count:  0
[2017-08-15 00:56:17.027509] (error)   [ebrcs_alameda] Retuning to Control Channel: 7.74219e+08
[2017-08-15 00:56:17.029048] (info)      - System Source - Min Freq: 7.67794e+08 Max Freq: 7.75794e+08
[2017-08-15 00:56:17.033554] (error)   [ebrcs_alameda]   Control Channel Message Decode Rate: 0/sec, count:  0
[2017-08-15 00:56:20.024493] (error)   [ebrcs_alameda] Retuning to Control Channel: 7.74494e+08
[2017-08-15 00:56:20.025498] (info)      - System Source - Min Freq: 7.67794e+08 Max Freq: 7.75794e+08
[2017-08-15 00:56:20.028135] (error)   [ebrcs_alameda]   Control Channel Message Decode Rate: 0/sec, count:  0
[2017-08-15 00:56:23.023827] (error)   [ebrcs_alameda] Retuning to Control Channel: 7.73469e+08
[... cut ...]
[2017-08-15 00:56:29.023451] (info)      - System Source - Min Freq: 7.67794e+08 Max Freq: 7.75794e+08
[2017-08-15 00:56:29.026178] (error)   [ebrcs_alameda]   Control Channel Message Decode Rate: 0/sec, count:  0
[2017-08-15 00:56:32.027347] (error)   [ebrcs_alameda] Retuning to Control Channel: 7.74494e+08
[2017-08-15 00:56:32.027833] (info)      - System Source - Min Freq: 7.67794e+08 Max Freq: 7.75794e+08
[2017-08-15 00:56:32.030098] (error)   [ebrcs_alameda]   Control Channel Message Decode Rate: 0/sec, count:  0`