TrunkRecorder/trunk-recorder #58: Sample Rate Cannot Be a Multiple of 24k

normFranklin, 2016-06-16:
When the sample rate is a multiple of 24k the program exits with: 
"Something is probably wrong! Resampling rate too low in p25_trunking."

In p25_recorder.cc and p25_trunking.cc the software requires (arb_rate  <  1) to avoid an exit(0).

When the sample rate is a multiple of 24k, arb_rate is one.

The program is working great for me after a change to  (arb_rate <= 1).

robotastic, 2016-06-17:
Good call!  I had been meaning to add in a fix. I will go make this change. Open this back up if it doesn't work.

Thanks!

closed 2016-06-17 by robotastic

normFranklin, 2016-06-17:
I see that you updated p25_recorder.  p25_trunking.cc has the same issue.

robotastic, 2016-06-18:
Doh - yep, meant to change that too. got distracted getting kids ready for school