TrunkRecorder/trunk-recorder #60: Analog calls are not clearing from the Recorder array when complete using multiple RTL-SDRs

dreinhold, 2016-07-07:
I was noticing traffic that should have been recorded but it was not. This is on a second site which is using 2 RTL-SDR sticks to monitor. I do not see this issue with the same code on the main site using a single AirSpy.

Looking at the logs it was showing "No Analog Recorders Available", so I increased the available recorders to more than the voice channels (8).
And still was getting the error. After adding some debugging I can see the array is full of old calls that still show active. I am adding more debugging now to nail it down.

```
[2016-07-06 17:00:58.242711] [0x00007fb2ecf49780] [error]       Call created for: 720   TDMA: 0 Encrypted: false        Freq: 8.68375e+08
[2016-07-06 17:00:58.242757] [0x00007fb2ecf49780] [info]    [ osmosdr ] No Analog Recorders Available
[2016-07-06 17:00:58.242786] [0x00007fb2ecf49780] [info]    Recoder For TG 7024 On Freq 867.375Mhz is active true
[2016-07-06 17:00:58.242811] [0x00007fb2ecf49780] [info]    Recoder For TG 2224 On Freq 867.625Mhz is active true
[2016-07-06 17:00:58.242835] [0x00007fb2ecf49780] [info]    Recoder For TG 2224 On Freq 867.375Mhz is active true
[2016-07-06 17:00:58.242859] [0x00007fb2ecf49780] [info]    Recoder For TG 2224 On Freq 867.625Mhz is active true
[2016-07-06 17:00:58.242883] [0x00007fb2ecf49780] [info]    Recoder For TG 1712 On Freq 867.375Mhz is active true
[2016-07-06 17:00:58.242910] [0x00007fb2ecf49780] [info]    Recoder For TG 720 On Freq 867.375Mhz is active true
[2016-07-06 17:00:58.242933] [0x00007fb2ecf49780] [info]    Recoder For TG 2224 On Freq 867.625Mhz is active true
[2016-07-06 17:00:58.242957] [0x00007fb2ecf49780] [info]    Recoder For TG 7024 On Freq 867.375Mhz is active true
[2016-07-06 17:00:58.242976] [0x00007fb2ecf49780] [error]       Not recording call
[2016-07-06 17:01:00.018298] [0x00007fb2ecf49780] [error]       Control Channel Message Decode Rate: 39.6667/sec

```

robotastic, 2016-07-07:
Hmm... that is weird. I wonder why those old calls are staying around. If a call comes up on a freq that is already being used, it is supposed to kill the existing recording. Most weird! I will take a look at the code.

robotastic, 2016-07-10:
I got a weird one, I had this crop up when I was just running with 1 RTLSDR which does not cover all of the freq. Not sure if it is related:
````
[2016-07-09 23:20:50.284728] [0x00007fff74234000] [info]    p25_recorder.cc: Activating Logger   	[ 0 ] - freq[ 8.56188e+08] 	 talkgroup[ 149 ]
[2016-07-09 23:20:52.608693] [0x00007fff74234000] [error]   	Call created for: 11027	TDMA: 1	Encrypted: true	Freq: 7.70469e+08
[2016-07-09 23:20:53.060681] [0x00007fff74234000] [error]   	Call created for: 11031	TDMA: 0	Encrypted: true	Freq: 7.74031e+08
[2016-07-09 23:20:53.436429] [0x00007fff74234000] [error]   	Call created for: 709	TDMA: 0	Encrypted: true	Freq: 8.56988e+08
[2016-07-09 23:20:53.660796] [0x00007fff74234000] [error]   	Call created for: 11037	TDMA: 1	Encrypted: true	Freq: 7.74031e+08
[2016-07-09 23:20:58.532986] [0x00007fff74234000] [error]   	Call created for: 11300	TDMA: 0	Encrypted: true	Freq: 7.69969e+08
[2016-07-09 23:20:59.733079] [0x00007fff74234000] [error]   	Call created for: 2145	TDMA: 0	Encrypted: false	Freq: 8.54862e+08
[2016-07-09 23:20:59.733280] [0x00007fff74234000] [error]   	Not recording call
[2016-07-09 23:21:03.860302] [0x00007fff74234000] [error]   	Call created for: 11037	TDMA: 0	Encrypted: true	Freq: 7.69969e+08
[2016-07-09 23:21:08.884388] [0x00007fff74234000] [error]   	Call created for: 11049	TDMA: 0	Encrypted: true	Freq: 7.69906e+08
[2016-07-09 23:21:09.937122] [0x00007fff74234000] [error]   	Call created for: 11057	TDMA: 1	Encrypted: true	Freq: 7.73344e+08
[2016-07-09 23:21:10.984189] [0x00007fff74234000] [error]   	Call created for: 102	TDMA: 0	Encrypted: false	Freq: 8.57188e+08
[2016-07-09 23:21:10.984223] [0x00007fff74234000] [error]   	Recording not started because there was no source covering: 8.57188e+08 For TG: 102
[2016-07-09 23:21:29.584576] [0x00007fff74234000] [info]    	Update Retune - Elapsed: 39s 	Since update: 1s 	Talkgroup: 149	Old Freq: 8.56188e+08	New Freq: 8.54862e+08
[2016-07-09 23:21:44.509009] [0x00007fff74234000] [error]   	Call created for: 11029	TDMA: 1	Encrypted: true	Freq: 7.70719e+08
[2016-07-09 23:21:46.532710] [0x00007fff74234000] [error]   	Call created for: 102	TDMA: 0	Encrypted: false	Freq: 8.57188e+08
[2016-07-09 23:21:46.532755] [0x00007fff74234000] [error]   	Recording not started because there was no source covering: 8.57188e+08 For TG: 102
[2016-07-09 23:21:50.284297] [0x00007fff74234000] [error]   	Call created for: 11033	TDMA: 0	Encrypted: true	Freq: 7.70406e+08
[2016-07-09 23:21:52.908172] [0x00007fff74234000] [error]   	Call created for: 2145	TDMA: 0	Encrypted: false	Freq: 8.56188e+08
[2016-07-09 23:21:52.908189] [0x00007fff74234000] [error]   	Not recording call
[2016-07-09 23:21:55.008716] [0x00007fff74234000] [info]    p25_recorder.cc: Deactivating Logger 	[ 0 ] - freq[ 8.54862e+08] 	 talkgroup[ 149 ]
recorder(39930,0x7fff74234000) malloc: *** error for object 0x10e82f: pointer being freed was not allocated
*** set a breakpoint in malloc_error_break to debug
Abort trap: 6
````

dreinhold, 2016-07-12:
Working on setting up a second PC so I can re-test this without messing with my main setup.  
While setting up the second system I saw GNU Radio was pretty current in the Ubuntu repository, so I went for a fresh install using apt-get, it was so much faster to get going ;).  
I started a WIKI with my notes, check it out https://github.com/robotastic/trunk-recorder/wiki

robotastic, 2016-07-12:
Awesome! If I remember right it was also pretty easy doing an apt-get
install on the RasPi too. I will try to document that, it should be the
same.

Does it make sense to point to this from the Readme? Not sure what proper
github form is....

robotastic, 2016-08-05:
This should be fixed in the latest version of the multi-system branch

dreinhold, 2016-08-05:
Cool, I will check it out.

closed 2017-03-10 by robotastic