TrunkRecorder/trunk-recorder #64: Adding unit-to-unit calls

rosecitytransit, 2016-08-12:
First of all, I would like to thank you for creating this app.  There's a Windows program like it ([Trunking Recorder](http://www.scannerbox.us/TrunkingRecorder/)) but it'll be nice running everything on one box and one Web site.  Also, if anyone is interested, I documented what I had to do to get it working with a minimal install of GNU Radio that doesn't include GUI elements or UHD/USRP support [on the GNU Radio mailing list](http://lists.gnu.org/archive/html/discuss-gnuradio/2016-08/msg00106.html).

As for adding unit-to-unit calls, I did record a bit of the control channel using DSDPlus on Windows and also save the log file, which shows a unit-to-unit call between 104 and 9341, and between 101 and 9866.
http://www.rosecitytransit.org/files/DSDPlus-Raw-Input_2014-06-08@152857.wav
http://www.rosecitytransit.org/files/DSDPlus-Raw-Input_2016-08-11@151038.txt

Lastly, the captureDir argument is a global option and doesn't work inside a system object like the Readme describes.  And are you still running this off a laptop, because the two times I've tried using an old laptop as a server (including once for Trunking Recorder) they didn't last.

robotastic, 2016-08-13:
It is great to hear that people are using it! Iw as following along the discussion on the mailing list and Reddit. I am going to try and create a Docker image based on it.

I have the standards docs for P25. That, plus your captures should let me add in support for unit-to-unit calls. I will put together a fork and let you know.

I have it running on a pretty powerful I7 desktop. The multi-sytem branch is a bit more efficient. There is a config change in it. Good catch on the captureDir arg. I will fix that.

robotastic, 2016-08-14:
Ok - I think I may have added in unit-to-unit support. Give the gr-latency branch a try. The conf file format has been updated slightly. System is now Systems and needs to be specified as an array.

rosecitytransit, 2016-08-15:
> It is great to hear that people are using it!

Recording scanner audio can be useful and interesting, and I don't know of any other Linux program that can do it.

It is recording unit-to-unit calls, however every time one starts I get "[error]  Talkgroup not found: <frequency> For TG: <unit>" in the output, and the srcList in the JSON file never gets populated.

Also:
-Instead of creating the JSON files, can the file names be written as "HHMMSS-talkgroup-unit-unit..."?  I've noticed that the JSON files occasionally weren't being created (may be a resource issue), and I think I'm going to look for an app that can index calls by file name instead of JSON.  If the files are created at the start of calls the unit numbers could be added in the post-call encoding script.  I don't really care what frequency the system picked for a particular call.

-So far, the gr-latency branch is creating MUCH less system load than the master branch, 0.25-0.5 vs up to 1.50.

-Does the voice channel monitoring process consider non-voice data blocks as activity?  The system I am monitoring doesn't properly put unit-unit call updates on the control channel (I'm told there's a level 4 support issue open with the vendor regarding this) so the process needs to watch the channel until it is finally released in order to capture the entire call.  Note that UniTrunker (and Trunking Recorder) rely on call updates to know that a call is still going on so can't handle the unit-unit calls.

rosecitytransit, 2016-08-15:
I realized the "talkgroup not found" error was because the unit numbers weren't in the Channel List file and got that fixed, but it is not recording properly in that it will activate a new recorder every Channel Grant even if there is already one started for the frequency/units combo.
http://www.rosecitytransit.org/files/trunk1.log

In addition, the longer unit calls are running into the 45 second time out in main.cc

dreinhold, 2016-08-15:
> -Instead of creating the JSON files, can the file names be written as "HHMMSS-talkgroup-unit-unit..."?

Adding the option to customize the file name is a good idea, definitely should not be in-place of the json, just an addition. I wanted to do this when I first started using trunk-recoder. But once I started feeding a web interface I no longer even look at the file names so it fell off my radar.

robotastic, 2016-08-16:
On custom file names: Cool - I like this idea. Right now the best approach is to probably edit the source code. It is all in Call.cc. Adding a custom templating system for the filename would probably add a lot of complexity, if someone has a good approach though I am open to it. I can probably add a pointer in the Docs on editing the filename. I will create a new Issue for this.

robotastic, 2016-08-16:
On Unit-to-Unit - great to hear it is atleast recording something. Not sending Update messages for the Unit-to-unit calls does make things a bit more tricky. For all the other types of calls, I go by update messages. What is happening is that a call will time out after ~3 seconds of not receiving an update message and go into a "stopping" state where is will start recording until it doesn't hear anything more on the freq. During this time period it is like the call doesn't exist, so a duplicate could be created. This is the correct behavior when there are update messages. Hmm, I will have to think about the best way to handle this. The auto kill after 45 seconds is sort of an insurance policy I had against some weird corner cases. I will check and see if it is still needed. You can delete this check if you want, just search the code for 45.

rosecitytransit, 2016-08-16:
I interpreted the Readme as that the monitoring process watched the voice frequency and not control channel update messages.  I am getting much more of the unit-unit calls than I did with Unitrunker/Trunking Recorder which cut them off after a few seconds (upon not seeing an update message).

For now, I think there just needs to be a check to see if the call is already being recorded.  Recorders are duplicating calls, apparently every time there's another channel grant (and maybe when there hasn't been enough of a pause since the last person spoke to allow the timeout).

`[2016-08-14 23:12:05.289934] [0x00007fb19dab3780] [info]    tsbk04	Unit to Unit Chan Grant	Channel ID:   441	Freq: 771.756	Target ID:     101	TDMA 0	Source ID: 2646
[2016-08-14 23:12:05.290006] [0x00007fb19dab3780] [info]    Activating rec on src: rtl=0
[2016-08-14 23:12:05.290016] [0x00007fb19dab3780] [info]    p25_recorder.cc: Activating Logger   	[ 0 ] - freq[ 7.71756e+08] 	 talkgroup[ 101 ]
[2016-08-14 23:12:05.290459] [0x00007fb19dab3780] [info]    tsbk04	Unit to Unit Chan Grant	Channel ID:   441	Freq: 771.756	Target ID:     101	TDMA 0	Source ID: 2646
[2016-08-14 23:12:07.202913] [0x00007fb19dab3780] [info]    p25_recorder.cc: Stopping Logger 	[ 1 ] - freq[ 7.71206e+08] 	 talkgroup[ 1761 ]
[2016-08-14 23:12:10.101023] [0x00007fb19dab3780] [info]    Removing Recorded Call 	TG: 1761	Last Update: 8 Call Elapsed: 37 Stopping Elapsed: 3
[2016-08-14 23:12:10.101215] [0x00007fb19dab3780] [info]    p25_recorder.cc: Closing Logger 	[ 1 ] - freq[ 7.71206e+08] 	 talkgroup[ 1761 ]
[2016-08-14 23:12:10.105420] [0x00007fb19dab3780] [info]    p25_recorder.cc: Stopping Logger 	[ 0 ] - freq[ 7.71756e+08] 	 talkgroup[ 101 ]
[2016-08-14 23:12:17.592975] [0x00007fb19dab3780] [info]    tsbk04	Unit to Unit Chan Grant	Channel ID:   441	Freq: 771.756	Target ID:     101	TDMA 0	Source ID: 2646
[2016-08-14 23:12:17.593065] [0x00007fb19dab3780] [info]    Activating rec on src: rtl=0
[2016-08-14 23:12:17.593082] [0x00007fb19dab3780] [info]    p25_recorder.cc: Activating Logger   	[ 1 ] - freq[ 7.71756e+08] 	 talkgroup[ 101 ]
[2016-08-14 23:12:17.593743] [0x00007fb19dab3780] [info]    tsbk04	Unit to Unit Chan Grant	Channel ID:   441	Freq: 771.756	Target ID:     101	TDMA 0	Source ID: 2646
[2016-08-14 23:12:22.173602] [0x00007fb19dab3780] [info]    p25_recorder.cc: Stopping Logger 	[ 1 ] - freq[ 7.71756e+08] 	 talkgroup[ 101 ]
[2016-08-14 23:12:29.222622] [0x00007fb19dab3780] [info]    tsbk04	Unit to Unit Chan Grant	Channel ID:   441	Freq: 771.756	Target ID:     101	TDMA 0	Source ID: 2646
[2016-08-14 23:12:29.222695] [0x00007fb19dab3780] [info]    Activating rec on src: rtl=0
[2016-08-14 23:12:29.222714] [0x00007fb19dab3780] [info]    p25_recorder.cc: Activating Logger   	[ 2 ] - freq[ 7.71756e+08] 	 talkgroup[ 101 ]
[2016-08-14 23:12:29.223421] [0x00007fb19dab3780] [info]    tsbk04	Unit to Unit Chan Grant	Channel ID:   441	Freq: 771.756	Target ID:     101	TDMA 0	Source ID: 2646
[2016-08-14 23:12:34.020641] [0x00007fb19dab3780] [info]    p25_recorder.cc: Stopping Logger 	[ 2 ] - freq[ 7.71756e+08] 	 talkgroup[ 101 ]
[2016-08-14 23:12:42.981242] [0x00007fb19dab3780] [info]    tsbk04	Unit to Unit Chan Grant	Channel ID:   441	Freq: 771.756	Target ID:     101	TDMA 0	Source ID: 2646
[2016-08-14 23:12:42.981310] [0x00007fb19dab3780] [info]    Activating rec on src: rtl=0
[2016-08-14 23:12:42.981328] [0x00007fb19dab3780] [info]    p25_recorder.cc: Activating Logger   	[ 3 ] - freq[ 7.71756e+08] 	 talkgroup[ 101 ]
[2016-08-14 23:12:42.981793] [0x00007fb19dab3780] [info]    tsbk04	Unit to Unit Chan Grant	Channel ID:   441	Freq: 771.756	Target ID:     101	TDMA 0	Source ID: 2646
[2016-08-14 23:12:47.126422] [0x00007fb19dab3780] [info]    p25_recorder.cc: Stopping Logger 	[ 3 ] - freq[ 7.71756e+08] 	 talkgroup[ 101 ]
[2016-08-14 23:12:50.024372] [0x00007fb19dab3780] [info]    tsbk04	Unit to Unit Chan Grant	Channel ID:   441	Freq: 771.756	Target ID:     101	TDMA 0	Source ID: 2646
[2016-08-14 23:12:50.024435] [0x00007fb19dab3780] [info]    [ rtl=0 ] No Digital Recorders Available
[2016-08-14 23:12:50.024444] [0x00007fb19dab3780] [info]    [ 0 ] State: 4 Has stopped true Freq: 7.71756e+08
[2016-08-14 23:12:50.024453] [0x00007fb19dab3780] [info]    [ 1 ] State: 4 Has stopped true Freq: 7.71756e+08
[2016-08-14 23:12:50.024460] [0x00007fb19dab3780] [info]    [ 2 ] State: 4 Has stopped true Freq: 7.71756e+08
[2016-08-14 23:12:50.024468] [0x00007fb19dab3780] [info]    [ 3 ] State: 4 Has stopped true Freq: 7.71756e+08
[2016-08-14 23:12:50.024483] [0x00007fb19dab3780] [info]    tsbk04	Unit to Unit Chan Grant	Channel ID:   441	Freq: 771.756	Target ID:     101	TDMA 0	Source ID: 2646
[2016-08-14 23:12:53.137121] [0x00007fb19dab3780] [info]    Removing Recorded Call 	TG: 101	Last Update: 11 Call Elapsed: 11 Stopping Elapsed: 6
[2016-08-14 23:12:53.137305] [0x00007fb19dab3780] [info]    p25_recorder.cc: Closing Logger 	[ 3 ] - freq[ 7.71756e+08] 	 talkgroup[ 101 ]
[2016-08-14 23:12:55.047277] [0x00007fb19dab3780] [info]    Removing Recorded Call 	TG: 101	Last Update: 26 Call Elapsed: 26 Stopping Elapsed: 21
[2016-08-14 23:12:55.047471] [0x00007fb19dab3780] [info]    p25_recorder.cc: Closing Logger 	[ 2 ] - freq[ 7.71756e+08] 	 talkgroup[ 101 ]
[2016-08-14 23:12:56.085806] [0x00007fb19dab3780] [info]    Removing Recorded Call 	TG: 101	Last Update: 51 Call Elapsed: 51 Stopping Elapsed: 46
[... cut ...]
[2016-08-14 23:12:57.128836] [0x00007fb19dab3780] [info]    [ 3 ] State: 2 Has stopped true`

I'm not sure that will solve the problem when there's a long pause between the dispatcher and the remote person speaking that allows the recording to time out (besides no unit-unit update messages the system also never tells when the remote side keys up on one) or keep the entire call in one file when there's a long pause, but it would be a big help.

As for the auto-kill, I did find where it is and have changed it.

rosecitytransit, 2016-08-19:
Actually, what would be the harm in removing `(call->get_state() != stopping)` from line 483 of main.cc, increasing the 2-second wait on line 381 and removing the if statement on line 523?

robotastic, 2016-08-20:
That could work… but it will probably lead to some regular calls from getting cut off at the end and new group call not getting started on new frequency when they should. 

The problem I am running into with this is that, keeping calls going without update messages breaks the way things should normally work in a trunked system. To make this work, I would need to add a completely separate code path or lots of weird if statements. 

I will look to see if I can do this, but it may add a lot of cruft.

rosecitytransit, 2016-08-20:
I'm not seeing why calls would get cut off.  We're talking about extending recordings.  I'd leave in the stuff inside the if statement on line 523; just allowing it to run on stopping calls.  I think I'm going to try doing my idea; since the "talkgroups" of unit-unit calls are in a certain range, the timeout can be increased only for them.  I'd like to get this working by tomorrow, and after all, this is not normal correct behavior that's having to be dealt with.

Overall, it seems that if the squelch on the voice frequency is reliable (which it always is for digital), you could stop depending on the update messages for timing, and base the timeout on recording stopping.  Treat update messages the same as grant ones--if talkgroup is already being recorded, and frequency is the same, ignore them because we already know about the call.

robotastic, 2016-08-20:
Definitely give it a try! I think it could work. It can be difficult to debug this stuff when it doesn't work quite right. 

One of the problems is that because of all of the buffering with decoding audio, the audio that is getting save to the wav file can be up 3-8 seconds behind the status messages coming off the control channel. And it varies so it there is no way to have an easy fixed delay. I added a Stopping state so a Recording would keep going while it is still recvieing audio if there is some even if it stops getting update messages. There is a problem though when a channel has two talk groups back to back on it without a pause.

rosecitytransit, 2016-08-20:
> There is a problem though when a channel has two talk groups back to back on it without a pause.

If I'm reading things correctly, when it gets the message about the new call on the same frequency, it will immediately close and delete the existing call.

I have it going now and it seems to be working so far (the "Closing a non-closing logger' messages are no longer true): http://www.rosecitytransit.org/files/trunk3.log  My bigger problem is that I'm not familiar with the language (or the code).

The other issue is that the source of the unit-unit calls isn't getting saved.  I don't know how you want to get it into the JSON file, but since I'm not using them, it appears I can get access to message.source when the filename is created.

rosecitytransit, 2016-08-29:
The only thing I could use is getting the source radio ID of the existing call into the if statement on 483 of main.cc so that a new call from a different radio ID to the same destination radio isn't conjoined with the current one.  Increasing the stopping time creates the potential for it to happen.

As for saving the source radio ID into the JSON file, I'm pretty sure call.cc could be modified to use message.source if wav_src_list is empty, something like:

    if (wav_src_list[0]) {
      for (int i = 0; i < wav_src_count; i++) {
        if (i != 0) {
          myfile << ", " <<  wav_src_list[i].source;
        } else {
          myfile << wav_src_list[i].source;
        }
      } else {
          myfile << message.source
    }

With that fix, this issue could be closed.  The code still wouldn't keep different parts of the same unit-unit call together. but on correctly functioning systems it should get them recorded.

robotastic, 2016-08-30:
I am not sure if that would work. Right now it naturally groups together multiple sources talking on the same talkgroup during a short span. If I am understanding it right, this would create a new recording for each transmission for a different source. This would result in a lot of really small files for normal talkgroup calls. The other challenge is that with the buffering in the P25 audio decoder, it is tough to synchronize between what is happening on the control channel and what is getting recorded. 

Can you give it a try and see how it works?

rosecitytransit, 2016-08-30:
> Right now it naturally groups together multiple sources talking on the same talkgroup during a short span.

And for group calls, it should definitely do that.  I was going to have the statement be something like: `if ((call->get_talkgroup() == message.talkgroup) && ((message.talkgroup > 199) || (message.source == call->get_source_list())))`  (Only radio IDs < 200 can do unit-to-unit calls on this system and are numbered separate from actual talkgroups).  The problem is that I'm pretty sure get_source_list() doesn't get the source ID for unit-to-unit calls, or at least the routine that puts source IDs in the JSON file doesn't get them,

> The other challenge is that with the buffering in the P25 audio decoder

There's generally a gap between different unit-to-unit calls; this would just be to counteract having the stopping time for them be very high to accommodate pauses in a single call.

closed 2017-03-10 by robotastic