TrunkRecorder/trunk-recorder #65: Not able to build multi-system branch

dreinhold, 2016-08-16:
Finaly trying multi-system, but it wont cleanly build on my system.

Any ideas..

`
[ 40%] Building CXX object op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/p25_frame_assembler_impl.cc.o
/home/dylan/radio/trunk-recorder/op25_repeater/lib/p25_frame_assembler_impl.cc:60:5: error: prototype for ‘gr::op25_repeater::p25_frame_assembler::sptr gr::op25_repeater::p25_frame_assembler::make(int, const char*, int, int, bool, bool, bool, gr::msg_queue::sptr, bool, bool)’ does not match any in class ‘gr::op25_repeater::p25_frame_assembler’
     p25_frame_assembler::make(int sys_id, const char* udp_host, int port, int debug, bool do_imbe, bool do_output, bool do_msgq, gr::msg_queue::sptr 
     ^
In file included from /home/dylan/radio/trunk-recorder/op25_repeater/lib/p25_frame_assembler_impl.h:24:0,
                 from /home/dylan/radio/trunk-recorder/op25_repeater/lib/p25_frame_assembler_impl.cc:26:
/home/dylan/target/include/op25_repeater/p25_frame_assembler.h:50:19: error: candidate is: static gr::op25_repeater::p25_frame_assembler::sptr gr::op25_repeater::p25_frame_assembler::make(const char*, int, int, bool, bool, bool, gr::msg_queue::sptr, bool, bool)
       static sptr make(const char* udp_host, int port, int debug, bool do_imbe, bool do_output, bool do_msgq, gr::msg_queue::sptr queue, bool do_audi
                   ^
op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/build.make:100: recipe for target 'op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/p25_frame_assembler_impl.cc.o' failed
make[2]: *** [op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/p25_frame_assembler_impl.cc.o] Error 1
`

robotastic, 2016-08-16:
I would try a "make clean" - there were some edits to the op25 library. Do you have OP25 installed on your system outside of trunk-recorder? It maybe finding that and getting confused. I may have to mess around with the Cmake stuff if that is the case.

dreinhold, 2016-08-16:
> Do you have OP25 installed on your system outside of trunk-recorder?`

Yeah I did, and had this issue before. It was weird as master was building fine (even after a make clean) so I was no thinking it was that. But it was..  
It builds now thanks.

closed 2016-08-16 by dreinhold