TrunkRecorder/trunk-recorder #66: Custom filenames

robotastic, 2016-08-16:
-Instead of creating the JSON files, can the file names be written as "HHMMSS-talkgroup-unit-unit..."? I've noticed that the JSON files occasionally weren't being created (may be a resource issue), and I think I'm going to look for an app that can index calls by file name instead of JSON. If the files are created at the start of calls the unit numbers could be added in the post-call encoding script. I don't really care what frequency the system picked for a particular call. #64

rosecitytransit, 2016-08-29:
Not sure about others, but what I could use is, in nonstop_wavfile_sink_impl.cc, having the radio IDs be put into a non-duplicated dash-separated string instead the array, and have that string be a parameter that is passed to the encoding script in call.cc.  Something like:

    if (call_source not in sourcestring)
      sourcestring = sourcestring + "-" + call_source

robotastic, 2016-08-30:
OK - I think that could work. It would just be a '-' string with the numbers, that would be passed to the encoding shell script.

closed 2017-03-10 by robotastic