TrunkRecorder/trunk-recorder #67: Branch multi-system is not decoding smartnet control channel

dreinhold, 2016-08-16:
I just cant get the multi-system branch to decode my smart-net control channel.

First issue it was segfaulting, @` trunk-recorder/systems/smartnet_parser.cc line 59:`
`int command = atoi( x[2].c_str() );`
The x vector did not always contain more that one array element.

I went back where he message is built and found in main.cc
`trunk_messages = smartnet_parser->parse_message(msg->to_string().erase(0, 10));`
Which was this in master
`trunk_messages = smartnet_parser->parse_message(msg->to_string());`
If I dropped the erase the segfaults went away, but I was still only decoding like 4 messages p/ sec.

Using the same config, just the new system layout.

robotastic, 2016-08-17:
Sorry about that, I caught the .erase() bug it the slight later gr-latency branch. That is the branch I will probably merge in.

That is weird that it is getting that low decode rate. I did change the type of xlating filter that is used. Try switching out the FFT filter for the FIR filter. in trunk-recorder/systems/smartnet-trunking.cc, comment out lines 40-44 and try lines 45-48. You will also have to make the switch in smartnet-trunking.h

Let me know if that helps... That is pretty weird. Didn't seem to impact P25.

dreinhold, 2016-08-17:
Yes after going back to FIR filter it's able to decode the control channel at full ~43 messages p/second.

robotastic, 2016-08-17:
Interesting! 
Are the voice channels digital or analog? Are they coming through ok?

dreinhold, 2016-08-17:
The ones I listen to are analog so did not check that, but plenty of digital traffic. I'll check that later tonight. This is just on my test box, so i did not run it that long last night.

dreinhold, 2016-08-20:
I was not able to decode digital channels, but I was also having trouble getting digital to decode when I went back to master. So i'm not sure it's the new code yet. I will try again in the next couple days,

robotastic, 2016-08-20:
Hmm - I wonder if it could be the switch from DSD to OP25 to decode the audio. Maybe I will try to dig out a really old branch.

dreinhold, 2016-08-20:
Master is working for digital after you went to op25. It was just not
working on my test system when I reverted back. I think it's the system not
the code.

dreinhold, 2016-10-17:
Closing this one out. Seems like smartnet-fun is the branch moving forward. I have a new machine coming this week so I can have a test box ;) I'll try that branch out here on my local system and open any new issues on that branch.

closed 2016-10-17 by dreinhold

robotastic, 2016-10-17:
Great! I think I should have improved CPU usage for SmartNet. I am also
working on cleaning up the versioning/branching for the repo and adding in
Docker Releases