TrunkRecorder/trunk-recorder #68: error: narrowing conversion on make

ericbond007, 2016-09-01:
During the make, I get these errors as output. I installed the prereqs and followed the install guide in the readme. I am running on arch linux so I may be missing a package or library. This may be a local issue for me, not sure tho.

$ make
[  1%] Building CXX object op25_repeater/lib/imbe_vocoder/CMakeFiles/imbe_vocoder.dir/pitch_est.cc.o
/home/eric/code/trunk-recorder/op25_repeater/lib/imbe_vocoder/pitch_est.cc:60:1: error: narrowing conversion of ‘32970’ from ‘int’ to ‘Word16 {aka short int}’ inside { } [-Wnarrowing]
 };
 ^
/home/eric/code/trunk-recorder/op25_repeater/lib/imbe_vocoder/pitch_est.cc:60:1: error: narrowing conversion of ‘33226’ from ‘int’ to ‘Word16 {aka short int}’ inside { } [-Wnarrowing]
.......... [more almost identical lines]
/home/eric/code/trunk-recorder/op25_repeater/lib/imbe_vocoder/pitch_est.cc:60:1: error: narrowing conversion of ‘39626’ from ‘int’ to ‘Word16 {aka short int}’ inside { } [-Wnarrowing]
make[2]: *** [op25_repeater/lib/imbe_vocoder/CMakeFiles/imbe_vocoder.dir/build.make:327: op25_repeater/lib/imbe_vocoder/CMakeFiles/imbe_vocoder.dir/pitch_est.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:220: op25_repeater/lib/imbe_vocoder/CMakeFiles/imbe_vocoder.dir/all] Error 2
make: *** [Makefile:128: all] Error 2

robotastic, 2016-09-01:
Hmm… That is interesting. I haven’t tried building on Arch Linux before. What processor type are you building for? Is it a 64-bit arch?

ericbond007, 2016-09-01:
I'm running on a macbook pro, so standard x86 on Intel. 64-bit OS and kernel as well.

robotastic, 2016-09-21:
Sorry - I sort of ran out of ideas on this... did you ever figure it out?

ericbond007, 2016-09-21:
I think it might have to do with my version of GCC (gcc version 6.1.1 20160802). It might be causing these issues, but I'm not sure. What version of GCC is working for you?

Dark-Sky, 2016-09-26:
I have the exact some problem trying to compile op25 on arch linux from their git and the AUR op25 snapshot does the same and has not been updated in a while.

This is what I know so far:

There is no problem compiling it with ubuntu 14.04

I ran into trying to compile with gcc 6.1 w/arch a while back on another git source and found on a forum that 6.1 would not work on it but a lower version of gcc would. It appears to me that the lower version of gcc would print it out as warnings but the higher version would treat it as errors. They had a fix changing the code a little to get that code to compile.

So I am wondering also if you came up with a solution or my last ditch effort would be trying to get a lower version of gcc installed and hoping my cmake version will be ok.

Dark-Sky, 2016-09-27:
I compiled and installed gcc-5.6-alternate from AUR and op25 compiled with out any errors but then I ran into wx3 errors when it runs the scope.py script. I am guessing arch linux has moved on with their version of wx and it is not compatable. So I am about at the end of the road and will continue running op25 on ubuntu 14.04 as I do not know anything about coding to make it work:

`Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
gain: name: LNA range: start 0 stop 0 step 0
setting gain LNA to 46
supported sample rates 250000-2560000 step 24000
Traceback (most recent call last):
  File "./scope.py", line 2751, in <module>
    app = stdgui2.stdapp(p25_rx_block, "APCO P25 Receiver", 3)
  File "/usr/lib/python2.7/site-packages/gnuradio/wxgui/stdgui2.py", line 46, in __init__
    wx.App.__init__ (self, redirect=False)
  File "/usr/lib/python2.7/site-packages/wx-3.0-gtk2/wx/_core.py", line 8628, in __init__
    self._BootstrapApp()
  File "/usr/lib/python2.7/site-packages/wx-3.0-gtk2/wx/_core.py", line 8196, in _BootstrapApp
    return _core_.PyApp__BootstrapApp(*args, **kwargs)
  File "/usr/lib/python2.7/site-packages/gnuradio/wxgui/stdgui2.py", line 49, in OnInit
    frame = stdframe (self.top_block_maker, self.title, self._nstatus)
  File "/usr/lib/python2.7/site-packages/gnuradio/wxgui/stdgui2.py", line 76, in __init__
    self.panel = stdpanel (self, self, top_block_maker)
  File "/usr/lib/python2.7/site-packages/gnuradio/wxgui/stdgui2.py", line 98, in __init__
    self.top_block = top_block_maker (frame, self, vbox, sys.argv)
  File "./scope.py", line 212, in __init__
    self.__init_gui(frame, panel, vbox)
  File "./scope.py", line 466, in __init_gui
    self.frame.CreateStatusBar()
  File "/usr/lib/python2.7/site-packages/wx-3.0-gtk2/wx/_windows.py", line 615, in CreateStatusBar
    return _windows_.Frame_CreateStatusBar(*args, **kwargs)
wx._core.PyAssertionError: C++ assertion "!m_frameStatusBar" failed at ./src/common/framecmn.cpp(381) in CreateStatusBar(): recreating status bar in wxFrame
`

b1tninja, 2016-11-12:
make CXX_FLAGS="-Wno-narrowing" 

Apparently something to do with the default c standard, and what was a warning becoming an error.

ericbond007, 2016-11-14:
I've about given up on this myself. I found a way to build and run trunk recorder using docker w/ an ubuntu image. If anyone wants the dockerfile, I can post it later.

robotastic, 2016-11-14:
That would be great! I was thinking of having a supported Docker image to
make life easier for folks. I would be curious how you set it up. I was
thinking of mounting volumes inside the image for the recordings and the
config file.

ericbond007, 2016-11-14:
Full disclosure: I found this in a comment on [Reddit](https://www.reddit.com/r/RTLSDR/comments/53ez4q/getting_op25_to_work/d7slecy/)

FROM ubuntu:xenial

RUN apt-get update
RUN apt-get install -y gnuradio gr-osmosdr libhackrf-dev libuhd-dev
RUN apt-get install -y git cmake build-essential libboost-all-dev libusb-1.0-0.dev

RUN mkdir /src && cd /src && git clone https://github.com/robotastic/trunk-recorder.git
RUN cd /src/trunk-recorder && cmake . && make

ADD ChannelList.csv /src/trunk-recorder
ADD config.json /src/trunk-recorder

put your ChannelList.csv and config.json in the same directory as the Dockerfile, then run 'docker build .'. You'll see 'Successfully built 6bafb286a476' with some other hash for your image.
Run 'docker run -it --privileged -v /dev/bus/usb:/dev/bus/usb 6bafb286a476 /bin/bash' with the hash from the last step replaced, you are then in a container and can cd to /src/trunk-recorder and run ./recorder

Dark-Sky, 2016-11-17:
My solution to get op25 going on my arch install was to install bare minimum ubuntu 14.04 to run op25 and mount it in a chroot jail with arch-chroot in my arch linux and use xhosts.

One good thing with linux there is usually more than one way to skin a cat.

dStruct, 2017-01-05:
FWIW, I attempted to install from GNU Radio's pybombs on the latest version of Kali Linux (essentially Debian Jessie), and got the same build error.

```
/src/gr-op25/op25/gr-op25_repeater/lib/imbe_vocoder/pitch_est.cc:60:1: error: narrowing conversion of ‘32970’ from ‘int’ to ‘Word16 {aka short int}’ inside { } [-Wnarrowing] };
```
GNU Make 4.1
gcc version 6.2.1 20161124 (Debian 6.2.1-5)

closed 2017-03-10 by robotastic