TrunkRecorder/trunk-recorder #69: Control Channel Decode Error and No Recording

RachelOlivero, 2016-09-21:
Hello:

I am receiving a mix of control channel error messages along with what appear to be comments about starting recordings. I am running on a Raspberry pi 3 with two RTL dongles.

I'm blind, and so haven't been able to use a graphical tool like gqrx to determine the correct offset. I have successfully used fmp24 and DSDPlus on Windows and used the -375 offset value I believe DSDPlus is providing to FMP24 for fine tuning, but whether I enter it as 375 or -375 I get the error. I also tried using kal to calibrate the RTL dongles, but that didn't seem to help either. These are Newelec dongles that have a 0.5PPM TCXO.

Some output from the program:

robotastic, 2016-09-21:
Wow!! congrats on getting it up and running on a Raspberry Pi 3!
Did you use the apt-get version of gnuradio or did you build it from scratch?
I had some performance problems running on the Raspberry Pi 3. I was using the apt-get version of GR, which may not be tuned for the Raspberry Pi 3 processor. 

I think I spotted the problem. The RTL-SDRs do not have their gain set. In the config file, in the section for source add:

"gain": 32

You may need change that number depending on how good the signal is.
Your error values sound about right. The TXCO based dongles are generally pretty accurate, and that sounds about right. 

Let me know if this helps. If it is working, you should stop having the control channel error message.

Are you using a screen reader? Let me know if there are formatting things for the output that would make life easier.

RachelOlivero, 2016-09-21:
Thanks. I did try with the apt-get version of GR. Maybe that's part of the issue?

I did have gain set in config.json. Changed it from 49 to 32 to see if it would make a difference, but it seems the same.

i never did get op25 to work on the pi. I hoped trunk-recorder might work better since it doesn't use such a graphic intensive front end. I had originally tried trunk-recorder on an Odroid XU4, but switched when I ran into the issue with squelch_base_cc.h. Maybe I'll go back to the other board. I'll have to figure out how to compile GR properly; there's no gr-osmosdr in apt-get for 16.04 for the XU4, and building it against the apt-get GR casues a versoin mismatch. Ah, fun times. 

Yes, I'm using a screen reader. I SSH into the pi and run everything from the shell. It doesn't reflect in the log file I sent, but when it's not throwing errors, I get constant strings of 0s. Is that expected?

robotastic, 2016-09-21:
It should work fine with the apt-get version of GR, the performance may just lack. When GnuRadio outputs a lot of 0s, that is a sign that the CPU is being overloaded.

You are right! You did have the gain specified in the config file you pasted. Oddly though, it is not showing up in the output when trunk-recorder starts up after it parses the config file. 

The issues with squelch_base_cc.h is fixable. The problem is that the header file was left out in one version of gnuradio. If you manually copy it over from another build, into the gnuradio include directories, it should work:

cp squelch_base_cc.h /usr/include/gnuradio/analog/

I think the Odroid XU4, might have a little extra horsepower, but it still might be lacking.

I do think that the gain related error is also part of the problem and that is why you are getting the control channel errors. I will try to think through that and see if I spot anything.

closed 2017-03-10 by robotastic