TrunkRecorder/trunk-recorder #70: Separating Sequential Transmissions

kkotowick, 2016-10-11:
Hi there,

This is an awesome piece of software! Very easy to get set up, and it works perfectly right out of the box. I have two questions:

1) When several units on a talkgroup are chatting with each other and the transmissions are close to each other, multiple transmissions will get lumped into one recorded file. Is there any way to change this, so that each transmission is in its own file?

2) The WAV files are quite large. Is there any way to add an optional compression stage, so they get converted to MP3?

Thanks!

Dark-Sky, 2016-10-11:
I have no idea how I got put on this list. I am not using robastic. I use
op25 with an APCO P25 Receiver GUI. It has a hold where I can lock on a
particular conversation and I have recorded conversations using audacity.

It requires ubuntu 14.04

dreinhold, 2016-10-11:
@Dark-Sky if you ever clicked watch in github on the repo? Well does not muchy matter how. If you go to the project in GitHub https://github.com/robotastic/trunk-recorder top right click on the "Unwatch" button and select Not-Watching you wont get any more notices for this project (unless someone calls your user name out in a comment)

dreinhold, 2016-10-12:
@kkotowick For point 2 each time a transmission completes the program will attempt to call the script encode-upload.sh which gets the wav file name passed in. In that script you could call something like lame to encode it into an mp3 and delete the wav.

closed 2017-03-10 by robotastic