TrunkRecorder/trunk-recorder #73: Empty wav files, OS X, Airspy

ddb, 2016-11-30:
I'm using trunk-recorder on OS X Sierra. I'm using it with an Airspy. I'm using it to listen to the King County (WA) Pubilc Safety trunking system, which is a Motorola SmartNet system.

I am successfully decoding the control channel, I can see calls starting and stopping.

It is also creating wav and son files in the expected place, but none of the WAV files actually contain anything (I once -- ONCE -- got a WAV file that had a transmission in it).

I am unsure what to do except to start exploring the code and adding debug stuff to it.

Any ideas what to look at first?

dreinhold, 2016-11-30:
It's probably the channel number to frequency lookup area. I see this system is re banded.
I know I have have to edit it before for part of my system. I am moble now but the code is in smartnet_parser.cc

ddb, 2016-11-30:
Hmmm. I suppose that should become an option when configuring a SmartNet system.

Thanks for the pointer!

ddb, 2016-12-01:
Would it be as simple as changing smartness_parser.cc:25 from:

`freq = float(cmd * 0.025 + 851.0125 - 10.9875);`

to:

`freq = float(cmd * 0.0125 + 851.0125 - 10.9875);` ?

ddb, 2016-12-01:
This was absolutely the issue.  TYVM.

closed 2016-12-01 by ddb

robotastic, 2016-12-01:
Awesome! Thanks @dreinhold 
I keep meaning to put together a Wiki article on all of this. SmartNet doesn't always follow a standard channel plan and needs tweaking.