TrunkRecorder/trunk-recorder #74: Add "rebanded" option for SmartNet

ddb, 2016-12-01:
The current smartness parsing code is incorrect for rebanded systems -- the channel spacing needs to be changed from 25KHz to 12.5KHz in those cases.

It would be awesome if there was a way to configure the recorder to use the rebanded spacing.

robotastic, 2016-12-01:
Good call - I can make that switch. I originally had the code from an old SmartNet system that was 12.5KHz yet. This is a code block that could require tweaking though based upon the system so I will add better docs for it.

timothyclemansinsea, 2017-01-12:
I wonder if this is why I can't get the recorder working http://www.radioreference.com/apps/db/?sid=604&opt=all_tg#tgs

robotastic, 2017-01-12:
That could be - you might want to try messing around with the code here: https://github.com/robotastic/trunk-recorder/blob/master/trunk-recorder/systems/smartnet_parser.cc#L54 <https://github.com/robotastic/trunk-recorder/blob/master/trunk-recorder/systems/smartnet_parser.cc#L54>
basically it calculates the frequency to tune to based upon a number. Try adding in some print statements and messing around with the numbers. Definitely let me know if you figure it out.

robotastic, 2017-03-10:
Added in better support SmartNet Bandplans

closed 2017-03-10 by robotastic