TrunkRecorder/trunk-recorder #75: Retune when channel is out of band

ddb, 2016-12-02:
The particular trunked system I like to follow uses channels that cove more than 10MHz of spectrum. Regardless of the center frequency I choose, I will always run into a channel that will get skipped because it is outside of the 10MHz I can get from my Airspy.  I suppose I could use multiple RTL-SDR's, but even they would have to be carefully allocated in order to cover all possible channels.

I've been thinking that being able to dynamically retune the SDR to cover all channels being actively recorded would be very nice to have, and would also eliminate the need to specify the center frequency of the SDR.

When I try to figure out what would be hard to do, it seems to me it boils down to how disruptive it is to do the retune, and how much traffic would be missed as everything settles back down.

I also see that it would complicate the multiple SDR scenario, although it seems to me it would make figuring out where each should be centered easier, if cleverly done.

robotastic, 2016-12-03:
Something like this could work... I am just not sure how disruptive it would be to anything else that is being recorded. All of the trunk channel monitors and recorders would have to also be retuned because they are simply an offset from the center freq. SDRs also have different levels of buffers, so that could also scramble things a little. 

I am not sure if the benefit would be worth the hassle, especially if the edge channels are used alot. I think adding an RTL-SDR is probably a much better approach.

Auto-tuning the center freq is probably a good idea. Right now though, you don't need to enter the freq for the different channels, so that might be more annoying than just doing the math out by hand.

rosecitytransit, 2016-12-27:
UniTrunker does this, and whether it was because of this, streams being opened/closed or what, I was getting blips in the audio.  The laptop I was using failed before I could play with DSD+ and/or UniTrunker to try and mitigate things.

I know you said you're using an Airspy, but it seems that the way to do things would be to use multiple SDRs and assign channels for each of them to follow.

ddb, 2016-12-27:
I'm sure that's why there are blips in your audio stream.

This could easily be a feature that you could enable/disable depending on your preferences, and would provide an option for those that don't have multiple SDR (regardless of how cheap they are), or have run out of ports, or many other reasons.

It doesn't have to be a feature that is turned on for everyone.

vabiro, 2017-08-22:
I was looking around on this and apparently op25 automatically does the retuning when a center frequency is not specified. Has this been tried by anyone in trunk-recorder? I believe the presence of a center frequency is required in the config file or it will error out.

closed 2020-08-04 by robotastic