TrunkRecorder/trunk-recorder #77: Record only what is in ChanList.csv

ke5eua, 2016-12-06:
I have setup the software and have it monitioring my local system but only need to record one or two id's. How would I accomplish doing this? So far I only placed the one talkgroup in my chanlist file but I see alot of wav recording and json files. The other thing is there are a lot of wav files that are essentialy empty, how do I fix that also, have it only record actual traffic? One last thing is I get an error, see attached screenshots.

This is my config.json file

> { "sources": [
{"center": 770033000,
"rate": 2048000.0,
"error": 0,
"ppm": 0,
"gain": 20,
"digitalRecorders": 4,
"driver": "osmosdr",
"device": "rtl=11"},
{"center": 771181250,
"rate": 2048000.0,
"error": 0,
"ppm": 0,
"gain": 20,
"digitalRecorders": 4,
"driver": "osmosdr",
"device": "rtl=12"},
{"center": 772462500,
"rate": 2048000.0,
"error": 0,
"ppm": 0,
"gain": 20,
"digitalRecorders": 4,
"driver": "osmosdr",
"device": "rtl=13"},
{"center": 773968750,
"rate": 2048000.0,
"error": 0,
"ppm": 0,
"gain": 20,
"digitalRecorders": 4,
"driver": "osmosdr",
"device": "rtl=14"}
],
"system": {
	"control_channels": [770968750],
	"type": "p25"
	},
"talkgroupsFile": "ChanList.csv"
}

[image]
[image]

I only need to record talkgroup 18000, but I have a lot of json and blank wav files for other talkgroups

[image]

robotastic, 2016-12-19:
It is weird that you are getting all of those messages that the control channel rate is less that 1/s. Also, all those 000000 mean that your system maybe overloaded. What type of system are you running it on?

If you create a Channel List and set the priorities for those talkgroups you don't want to be greater than 100, it won't ever record those talkgroups.

Have you taken a look at the signal in a program like GQRX? Is the tuning dead on, or do you need to add some error correction to the frequency? Is the amount of gain correct?

rosecitytransit, 2016-12-27:
I don't see any error or PPM parameters set, if you're not using the newer RTL SDRs.

closed 2017-03-10 by robotastic