TrunkRecorder/trunk-recorder #79: Include a sample output Web page

rosecitytransit, 2016-12-27:
I should have done this before, but I am attaching a stripped down version of the Web page that I created that acts as an interface to the recorded files.  I know there's https://github.com/robotastic/smartnet-player but I didn't want to deal with a database so instead created a PHP script and HTML that can read through all the audio files in a directory and then play them sequentially using the AUDIO tag.

Note that it will need further customizing to handle details from the JSON files that trunk-recorder is designed to create (I modified it so everything is put in the filenames, using HHMMSS-talkgroup-radio-radio.mp3, a better option may be to have it create per-day delimited log files)

[audioplayer.php](https://github.com/robotastic/trunk-recorder/files/674706/basic.php.txt)
[The live version that I am using is here](http://www.rosecitytransit.org:90/radio/test2.php)

robotastic, 2016-12-28:
This is great! Thanks for sharing back.

dpknoll, 2017-01-14:
This webpage is outstanding. Looking forward to using this. I'm trying to get this to work but am receiving a 500 error, more specifically apache tail shows - PHP Parse error:  syntax error, unexpected '<' in /var/www/html/basic.php on line 103. 

I'm new to .php so I'm likely missing something simple here. Any ideas on what I can do to resolve?

robotastic, 2017-01-14:
I setup a portal that should make it easier to view recordings. You need to be running the latest version of trunk-recorder, because it automatically uploads files.

Goto: https://admin.openmhz.com/ and sign up for an account. After you do that, create a new system and it will give you the code you need to add to a config file.

porcej, 2017-01-14:
Submitting the signup form (https://admin.openmhz.com/signup) gives a 502 bad gateway error.

robotastic, 2017-01-14:
Doh! This is still a work in progress. I will figure it out and post back when it is working.

rosecitytransit, 2017-01-15:
OK, I have gone back and finished the page.  It appears that line 103 was incomplete.

[audioplayer.php.txt](https://github.com/robotastic/trunk-recorder/files/706321/audioplayer.php.txt)

You'll want to put it in the captureDir, where the year folders are or customize the $dir parameter.  You'll also want to fill up the talkgroups array on line 12 replacing the Example1 and Example2 ones with the ones you're recording.

It still does not handle the details in the JSON files (e.g. radio numbers) since I got rid of them and it takes resources to read 100s of little files (as I said, a solution would be a have a per-day log file instead/also).

This was based off of
http://devblog.lastrose.com/html5-audio-video-playlist/ 
http://www.terrillthompson.com/tests/html5-playlist-simple.html
http://www.w3schools.com/tags/tag_audio.asp

rosecitytransit, 2017-01-15:
Forgot to mention: The captureDir parameter in config.json would also need to be set to something in /var/www/html/ or otherwise made visible to Web site viewers.

The portal idea is interesting, but this is nice if you just want to store the files locally.  And even compressed as MP3s, I'm getting 400 MB+ of recordings a day.

robotastic, 2017-01-15:
@porcej OK - got that fixed. I had a silly error in my error checking. Time to add some unit testing.

@rosecitytransit  - Definitely! Do you have a GitHub repo up for your player? I know there a lot of folks that are interested in an easy way to do local viewing/listening.

rosecitytransit, 2017-01-15:
> Definitely! Do you have a GitHub repo up for your player?

No, I don't, but you're welcome to add it to the project.

porcej, 2017-01-15:
The signup pages seems to be working now! 

- the talkgroup upload seems to fail.  Uploading a talkgroups file displays "unknown" for all of the values in the table once uploaded.  The number of entries appears to be correct so maybe my file format is wrong? [https://dev.porcej.com/trunker/alex.csv](https://dev.porcej.com/trunker/alex.csv)
- There appears to be an error on the "profile" page.  When trying to access said page the response is "/home/app/trunk-server-admin/views/profile.ejs:76 74| </td> 75| <td> >> 76| <% if (stats[systems[i].shortName].usageMb) { %> 77| <%= stats[systems[i].shortName].usageMb.toFixed(2) %> Mb 78| <% } %> 79| </td> Cannot read property 'usageMb' of undefined"
-  When attempting to upload to openmhz, I'm receiving an error "SSL: Response returned with status code 500
[2017-01-15 14:39:33.219856] [0x00007faba4740700] [info]    HTTPS Upload result: 1"  Perhaps I have an error in my config file [https://dev.porcej.com/trunker/config.json](https://dev.porcej.com/trunker/config.json)

Also, thank you for doing this.  Is there anything you would like some help with?

dpknoll, 2017-01-16:
@rosecitytransit Thanks for updating the code for audioplayer.php and for explaining the steps needed to get things working.  I'm now able to start testing this out. This is a great way to view recordings.

@robotastic Thanks for all your hard work on Trunk Recorder and now the possibility of hosting recordings using a service. Loving this!

robotastic, 2017-01-16:
@porcej Awesome! It was cool to see a new system pop up.

- With the talkgroups, the format is slightly different for OpenMHz... which I am realizing now is sort of dumb. All you have to do is delete the Talkgroup column formatted as a Hex number. 

- Fixed the profile.djs error. Stats hadn't been generated for the system yet but it try to display them.

- There could be a couple things causing this. I just released that there is one undocumented requirement is that ffmpeg is installed... and that it has support for libfdk-acc, which has support for High-Efficiency AAC. There is info here: https://trac.ffmpeg.org/wiki/CompilationGuide/Ubuntu
The following command is run to convert the wav file to an M4A:
`ffmpeg -y -i %s  -c:a libfdk_aac -b:a 32k -cutoff 18000 -hide_banner -loglevel panic`

The reason it is problem getting a 500 error on uploads is that it is probably trying to also upload files for the Fairfax and Arlington, and since those aren't systems are in the portal and don't have an API key, it is probably generating an error. I would try it without those systems and see if it is still generating 500s. 

Testing this stuff out is a huge help! Thanks

porcej, 2017-01-16:
@robotastic 

- Talk groups are uploaded! (I should have RTFM)
- The profile is working well, thanks for the fix!
- Recompiling FFmpeg to support HEAAC fixed the SSL error 500 error  (also, it seems to be preventing the periodic seg faults that were occurring)
- I am now getting an 'SSL Exception: short read' on upload, however, the upload seems to be working just fine.
- Do you plan to support radio aliases?
- On openMhz, is there any plan to make complex filtering? (select multiple talk groups, or say 'Police' and 'Fire' groups?
- On open MHz, is there a plan to allow someone to listen to multiple radio systems at once (IE, DC Fire, Alex, Fairfax, and Arlington) without having multiple sessions?
- Is there a repo for the web player?

This is all great stuff! Thanks for all your work!

porcej, 2017-01-16:
Aahh! Getting a 504 error on upload attempt and no content on the audio player pages (regardless of the system)

imagesafari, 2017-01-16:
You guys should check out Dylan's Trunk Player too -- https://github.com/ScanOC/trunk-player

robotastic, 2017-01-16:
@porcej Thats awesome! I just saw that it was up. I am seeing a bunch of 0 sec recordings. Do you know that is causing them? I work in Courthouse, so I might try to do some debug work and record Arlington's system to see if I get 0's too. I will start a new thread on OpenMHz and answer questions.

robotastic, 2017-01-16:
@imagesafari agreed! I will setup a wiki page going over all of the options for playing back files.

porcej, 2017-01-16:
I believe the zero second recordings are due to poor reception. They started popping up yesterday when I changed antennas (brought the box to my desk).  Once the box is squared away I'll reconnect the 'good' antenna and those 0 second recordings should go away.

robotastic, 2017-01-16:
Cool! Let me know if it doesn’t and I can hunt down if there is a bug in the code.

I think the 504 errors you were getting were because I have my system setup to generate statistics at the top of every hour. It looks like it maybe swamping the CPU. If you see it again, check if it is near the top of the hour. I will keep an eye on it too.

porcej, 2017-01-16:
> I think the 504 errors you were getting were because I have my system setup to generate statistics at the top of every hour. It looks like it maybe swamping the CPU. If you see it again, check if it is near the top of the hour. I will keep an eye on it too.

Yes. It's happening again now. (~1602HR by my watch)

porcej, 2017-01-16:
Also, the the website seems to be a tad bit wonky about the same time frame.  Screen shot at [https://dev.porcej.com/trunker/web_wonkyness.png](https://dev.porcej.com/trunker/web_wonkyness.png)

robotastic, 2017-01-17:
@porcej Cool - looks like it is probably the stats builder. I just disabled it. It was doing a MongoDB Map-Reduce job. Using the aggregation framework supposedly improves performance, so I will look into that to generate the pretty graphs.

dreinhold, 2017-01-18:
You guys are killing me with all the good new features that i dont have
time to play with!!  Keep it up..

Dylan

kg6uyz, 2017-08-02:
Does audioplayer.php still work with the current version of trunk-recorder? i set it up and made the changes for example1/2, moved my audio files to the web directory and placed audioplayer.php into the year folder like some of the posts stated but i am getting this output on the webpage: 

"; echo date("H:i:s",$exploded[1]).""; if (isset($talkgroups[$exploded[0]])) echo $talkgroups[$exploded[0]]; else echo $exploded[0]; echo ""; echo sprintf($exploded[2]/1000000)."".round(filesize($file) / 1024)."k
"; } } } else echo "Pick a different date"; ?> 

Any help would be greatly appreciated.
[audioplayer.php.txt](https://github.com/robotastic/trunk-recorder/files/1193018/audioplayer.php.txt)

rosecitytransit, 2017-10-16:
I'm testing it now on a fresh, unmodified install.  Note that the Web page needs to be within the system folder, as in along with the 2017 (year) folder, though this could easily be changed and made able to support multiple systems.

closed 2017-12-12 by rosecitytransit