TrunkRecorder/trunk-recorder #81: Cannot build on latest Kali Linux (Debian)

dStruct, 2016-12-28:
Hello, I was interested in setting this up on Kali Linux, I'm running the latest version, however I'm having build errors during the make portion of the install.  Everything is pretty standard, and I followed all the build/install instructions.

```
user@laptop:~/Documents/code/trunk-recorder$ cmake .
-- The CXX compiler identification is GNU 6.2.1
-- The C compiler identification is GNU 6.2.1
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Build type not specified: defaulting to release.
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29") 
-- Checking for module 'gnuradio-runtime'
--   Found gnuradio-runtime, version 3.7.10
-- Found GNURADIO_RUNTIME: /usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so  
-- Checking for module 'cppunit'
--   Found cppunit, version 1.13.2
-- Found CPPUNIT: /usr/lib/x86_64-linux-gnu/libcppunit.so;dl  
Checking for GNU Radio Module: RUNTIME
 * INCLUDES=/usr/include
 * LIBS=/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so
-- Found GNURADIO_RUNTIME: /usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so  
GNURADIO_RUNTIME_FOUND = TRUE
Checking for GNU Radio Module: ANALOG
-- Checking for module 'gnuradio-analog'
--   Found gnuradio-analog, version 3.7.10
 * INCLUDES=/usr/include
 * LIBS=/usr/lib/x86_64-linux-gnu/libgnuradio-analog.so;/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so;/usr/lib/x86_64-linux-gnu/libvolk.so
-- Found GNURADIO_ANALOG: /usr/lib/x86_64-linux-gnu/libgnuradio-analog.so;/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so;/usr/lib/x86_64-linux-gnu/libvolk.so  
GNURADIO_ANALOG_FOUND = TRUE
Checking for GNU Radio Module: BLOCKS
-- Checking for module 'gnuradio-blocks'
--   Found gnuradio-blocks, version 3.7.10
 * INCLUDES=/usr/include
 * LIBS=/usr/lib/x86_64-linux-gnu/libgnuradio-blocks.so;/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so
-- Found GNURADIO_BLOCKS: /usr/lib/x86_64-linux-gnu/libgnuradio-blocks.so;/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so  
GNURADIO_BLOCKS_FOUND = TRUE
Checking for GNU Radio Module: DIGITAL
-- Checking for module 'gnuradio-digital'
--   Found gnuradio-digital, version 3.7.10
 * INCLUDES=/usr/include
 * LIBS=/usr/lib/x86_64-linux-gnu/libgnuradio-digital.so;/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so
-- Found GNURADIO_DIGITAL: /usr/lib/x86_64-linux-gnu/libgnuradio-digital.so;/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so  
GNURADIO_DIGITAL_FOUND = TRUE
Checking for GNU Radio Module: FILTER
-- Checking for module 'gnuradio-filter'
--   Found gnuradio-filter, version 3.7.10
 * INCLUDES=/usr/include
 * LIBS=/usr/lib/x86_64-linux-gnu/libgnuradio-filter.so;/usr/lib/x86_64-linux-gnu/libgnuradio-fft.so;/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so
-- Found GNURADIO_FILTER: /usr/lib/x86_64-linux-gnu/libgnuradio-filter.so;/usr/lib/x86_64-linux-gnu/libgnuradio-fft.so;/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so  
GNURADIO_FILTER_FOUND = TRUE
Checking for GNU Radio Module: PMT
-- Checking for module 'gnuradio-runtime'
--   Found gnuradio-runtime, version 3.7.10
 * INCLUDES=/usr/include
 * LIBS=/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so
-- Found GNURADIO_PMT: /usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so  
GNURADIO_PMT_FOUND = TRUE
-- Found GNURADIO_RUNTIME: /usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so  
-- Found CPPUNIT: /usr/lib/x86_64-linux-gnu/libcppunit.so;dl;dl  
-- Checking for module 'gnuradio-uhd'
--   Found gnuradio-uhd, version 3.7.10
[... 77 lines cut ...]
[ 50%] Building CXX object op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/p25p2_isch.cc.o
[ 52%] Building CXX object op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/p25p2_duid.cc.o
[ 53%] Building CXX object op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/p25p2_sync.cc.o
[ 54%] Building CXX object op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/p25p2_tdma.cc.o
[ 56%] Building CXX object op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/p25p2_vf.cc.o
[ 57%] Building CXX object op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/imbe_decoder.cc.o
[ 59%] Building CXX object op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/software_imbe_decoder.cc.o
[ 60%] Building C object op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/ambe.c.o
[ 61%] Building C object op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/mbelib.c.o
[ 63%] Linking CXX shared library libgnuradio-op25_repeater.so
[ 63%] Built target gnuradio-op25_repeater
Scanning dependencies of target recorder
[ 64%] Building CXX object CMakeFiles/recorder.dir/trunk-recorder/main.cc.o
/home/user/Documents/code/trunk-recorder/trunk-recorder/main.cc: In function ‘int main()’:
/home/user/Documents/code/trunk-recorder/trunk-recorder/main.cc:912:29: error: expected ‘;’ before ‘__attribute__’
   BOOST_STATIC_ASSERT(true) __attribute__((unused));
                             ^~~~~~~~~~~~~
/home/user/Documents/code/trunk-recorder/trunk-recorder/main.cc:912:29: error: expected primary-expression before ‘__attribute__’
CMakeFiles/recorder.dir/build.make:62: recipe for target 'CMakeFiles/recorder.dir/trunk-recorder/main.cc.o' failed
make[2]: *** [CMakeFiles/recorder.dir/trunk-recorder/main.cc.o] Error 1
CMakeFiles/Makefile2:68: recipe for target 'CMakeFiles/recorder.dir/all' failed
make[1]: *** [CMakeFiles/recorder.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
```

dStruct, 2016-12-28:
I should also point out, I resolved the hackrf missing dep, just grabbed the wrong copy/paste.

robotastic, 2016-12-29:
Hmm... that is quite weird. It looks like it may have something to do with the version of GCC being used: http://stackoverflow.com/questions/13198272/error-expected-primary-expression-before-attribute

Could you try commenting out like 912 in main.cc?

Also check and see what version of GCC it is - `gcc -v`

dStruct, 2016-12-29:
Hi Robotastic,

I tried that, and it passes that build step to later completely fall on it's face.

I'm using the latest Kali Linux (Debian) which uses: gcc version 6.2.1 20161124 (Debian 6.2.1-5).

```
user@laptop:~/Documents/code/trunk-recorder$ make
[ 64%] Built target imbe_vocoder
[ 64%] Built target gnuradio-op25_repeater
[ 64%] Building CXX object CMakeFiles/recorder.dir/trunk-recorder/systems/p25_parser.cc.o
In file included from /usr/include/boost/log/sources/record_ostream.hpp:31:0,
                 from /usr/include/boost/log/trivial.hpp:23,
                 from /home/user/Documents/code/trunk-recorder/trunk-recorder/systems/p25_parser.h:9,
                 from /home/user/Documents/code/trunk-recorder/trunk-recorder/systems/p25_parser.cc:1:
/usr/include/boost/log/utility/formatting_ostream.hpp: In instantiation of ‘typename boost::log::v2_mt_posix::aux::enable_if_formatting_ostream<StreamT, StreamT&>::type boost::log::v2_mt_posix::operator<<(StreamT&, T&) [with StreamT = boost::log::v2_mt_posix::basic_formatting_ostream<char>; T = std::__cxx11::basic_ostringstream<char>; typename boost::log::v2_mt_posix::aux::enable_if_formatting_ostream<StreamT, StreamT&>::type = boost::log::v2_mt_posix::basic_formatting_ostream<char>&]’:
/usr/include/boost/log/sources/record_ostream.hpp:212:51:   required from ‘typename boost::log::v2_mt_posix::aux::enable_if_record_ostream<StreamT, StreamT&>::type boost::log::v2_mt_posix::operator<<(StreamT&, T&) [with StreamT = boost::log::v2_mt_posix::basic_record_ostream<char>; T = std::__cxx11::basic_ostringstream<char>; typename boost::log::v2_mt_posix::aux::enable_if_record_ostream<StreamT, StreamT&>::type = boost::log::v2_mt_posix::basic_record_ostream<char>&]’
/home/user/Documents/code/trunk-recorder/trunk-recorder/systems/p25_parser.cc:136:35:   required from here
/usr/include/boost/log/utility/formatting_ostream.hpp:859:19: error: no match for ‘operator<<’ (operand types are ‘boost::log::v2_mt_posix::basic_formatting_ostream<char>::ostream_type {aka std::basic_ostream<char>}’ and ‘std::__cxx11::basic_ostringstream<char>’)
     strm.stream() << value;
                    
/usr/include/boost/log/utility/formatting_ostream.hpp:859:19: note: candidate: operator<<(int, int) <built-in>
/usr/include/boost/log/utility/formatting_ostream.hpp:859:19: note:   no known conversion for argument 2 from ‘std::__cxx11::basic_ostringstream<char>’ to ‘int’
In file included from /usr/include/c++/6/iostream:39:0,
                 from /home/user/Documents/code/trunk-recorder/trunk-recorder/systems/parser.h:3,
                 from /home/user/Documents/code/trunk-recorder/trunk-recorder/systems/p25_parser.h:3,
                 from /home/user/Documents/code/trunk-recorder/trunk-recorder/systems/p25_parser.cc:1:
/usr/include/c++/6/ostream:108:7: note: candidate: std::basic_ostream<_CharT, _Traits>::__ostream_type& std::basic_ostream<_CharT, _Traits>::operator<<(std::basic_ostream<_CharT, _Traits>::__ostream_type& (*)(std::basic_ostream<_CharT, _Traits>::__ostream_type&)) [with _CharT = char; _Traits = std::char_traits<char>; std::basic_ostream<_CharT, _Traits>::__ostream_type = std::basic_ostream<char>]
       operator<<(__ostream_type& (*__pf)(__ostream_type&))
       ^~~~~~~~
/usr/include/c++/6/ostream:108:7: note:   no known conversion for argument 1 from ‘std::__cxx11::basic_ostringstream<char>’ to ‘std::basic_ostream<char>::__ostream_type& (*)(std::basic_ostream<char>::__ostream_type&) {aka std::basic_ostream<char>& (*)(std::basic_ostream<char>&)}’
/usr/include/c++/6/ostream:117:7: note: candidate: std::basic_ostream<_CharT, _Traits>::__ostream_type& std::basic_ostream<_CharT, _Traits>::operator<<(std::basic_ostream<_CharT, _Traits>::__ios_type& (*)(std::basic_ostream<_CharT, _Traits>::__ios_type&)) [with _CharT = char; _Traits = std::char_traits<char>; std::basic_ostream<_CharT, _Traits>::__ostream_type = std::basic_ostream<char>; std::basic_ostream<_CharT, _Traits>::__ios_type = std::basic_ios<char>]
       operator<<(__ios_type& (*__pf)(__ios_type&))
       ^~~~~~~~
/usr/include/c++/6/ostream:117:7: note:   no known conversion for argument 1 from ‘std::__cxx11::basic_ostringstream<char>’ to ‘std::basic_ostream<char>::__ios_type& (*)(std::basic_ostream<char>::__ios_type&) {aka std::basic_ios<char>& (*)(std::basic_ios<char>&)}’
/usr/include/c++/6/ostream:127:7: note: candidate: std::basic_ostream<_CharT, _Traits>::__ostream_type& std::basic_ostream<_CharT, _Traits>::operator<<(std::ios_base& (*)(std::ios_base&)) [with _CharT = char; _Traits = std::char_traits<char>; std::basic_ostream<_CharT, _Traits>::__ostream_type = std::basic_ostream<char>]
       operator<<(ios_base& (*__pf) (ios_base&))
       ^~~~~~~~
/usr/include/c++/6/ostream:127:7: note:   no known conversion for argument 1 from ‘std::__cxx11::basic_ostringstream<char>’ to ‘std::ios_base& (*)(std::ios_base&)’
/usr/include/c++/6/ostream:166:7: note: candidate: std::basic_ostream<_CharT, _Traits>::__ostream_type& std::basic_ostream<_CharT, _Traits>::operator<<(long int) [with _CharT = char; _Traits = std::char_traits<char>; std::basic_ostream<_CharT, _Traits>::__ostream_type = std::basic_ostream<char>]
       operator<<(long __n)
       ^~~~~~~~
/usr/include/c++/6/ostream:166:7: note:   no known conversion for argument 1 from ‘std::__cxx11::basic_ostringstream<char>’ to ‘long int’
[... cut ...]
/usr/include/boost/log/utility/formatting_ostream.hpp:859:19: note:   ‘std::__cxx11::basic_ostringstream<char>’ is not derived from ‘const std::__cxx11::basic_string<_CharT, _Traits, _Alloc>’
     strm.stream() << value;
                    
CMakeFiles/recorder.dir/build.make:230: recipe for target 'CMakeFiles/recorder.dir/trunk-recorder/systems/p25_parser.cc.o' failed
make[2]: *** [CMakeFiles/recorder.dir/trunk-recorder/systems/p25_parser.cc.o] Error 1
CMakeFiles/Makefile2:68: recipe for target 'CMakeFiles/recorder.dir/all' failed
make[1]: *** [CMakeFiles/recorder.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
```

robotastic, 2017-01-01:
Doh! That seems to have just made things worse. Would you be up for trying to run it using docker? I am working on an install that uses it.

thefinn93, 2017-02-09:
I also have this issue, on Debian Sid. It looks like [Ubuntu 16.04's gcc package](http://packages.ubuntu.com/xenial/gcc) is gcc 5, so I installed `gcc-5` and `g++-5`, then per [this StackOverflow post](http://askubuntu.com/a/26518) temporarily set my gcc and g++ versions to 5 and it gets further along, but then fails to eventually fails with:

```
CMakeFiles/recorder.dir/trunk-recorder/uploaders/call_uploader.cc.o: In function `boost::cpp_regex_traits<char>::transform_primary(char const*, char const*) const':
/usr/include/boost/regex/v4/cpp_regex_traits.hpp:966: undefined reference to `boost::re_detail_106200::cpp_regex_traits_implementation<char>::transform_primary(char const*, char const*) const'
CMakeFiles/recorder.dir/trunk-recorder/uploaders/call_uploader.cc.o: In function `boost::cpp_regex_traits<char>::transform(char const*, char const*) const':
/usr/include/boost/regex/v4/cpp_regex_traits.hpp:962: undefined reference to `boost::re_detail_106200::cpp_regex_traits_implementation<char>::transform(char const*, char const*) const'
collect2: error: ld returned 1 exit status
make[2]: *** [recorder] Error 1
CMakeFiles/recorder.dir/build.make:740: recipe for target 'recorder' failed
make[1]: *** [CMakeFiles/recorder.dir/all] Error 2
CMakeFiles/Makefile2:68: recipe for target 'CMakeFiles/recorder.dir/all' failed
make: *** [all] Error 2
Makefile:127: recipe for target 'all' failed
```

[full log here](https://gist.github.com/thefinn93/912ab0a48ba8ce7ac4d6e072a03c0e38). I assume this is an issue with the version of libboost that's installed, unfortunately the [gnuradio](https://packages.debian.org/sid/gnuradio) package depends on libboost1.62, and [the Ubuntu 16.04 package](http://packages.ubuntu.com/xenial/libboost-all-dev) is for 1.58.

thefinn93, 2017-02-09:
Update: I was able to make it build by removing the part of the uploader component that uses the regex:

```diff
 trunk-recorder/uploaders/call_uploader.cc | 4 ++--
 1 file changed, 2 insertions(+), 2 deletions(-)

diff --git a/trunk-recorder/uploaders/call_uploader.cc b/trunk-recorder/uploaders/call_uploader.cc
index 1785afd..ad95807 100644
--- a/trunk-recorder/uploaders/call_uploader.cc
+++ b/trunk-recorder/uploaders/call_uploader.cc
@@ -181,10 +181,10 @@ void send_call(Call *call, System *sys, Config config) {
   pthread_t    thread;
 
 
-  boost::regex  ex("(http|https)://([^/ :]+):?([^/ ]*)(/?[^ #?]*)\\x3f?([^ #]*)#?([^ ]*)");
+  // boost::regex  ex("(http|https)://([^/ :]+):?([^/ ]*)(/?[^ #?]*)\\x3f?([^ #]*)#?([^ ]*)");
   boost::cmatch what;
 
-  if (regex_match(config.upload_server.c_str(), what, ex))
+  if (false)
   {
     // from: http://www.zedwood.com/article/cpp-boost-url-regex
     call_info->upload_server = config.upload_server;
```

and it builds and appears to generally work, although I haven't got my system properly tuned yet...

thefinn93, 2017-03-11:
Why was this closed? Has it been fixed? I'm still seeing this on at least my Kali box, have not tested on on a debian box.

robotastic, 2017-03-11:
Sorry - I was cleaning up issues. Meant to label this one instead of closing. I will try to find a better way to do the RegEx

nkwood, 2017-03-12:
FWIW, building fails in the same manner on Ubuntu 16.10 which is on gcc/g++ 6.2 and libboost 1.61.0.2. I played around with moving down to gcc/g++ 5 and haven't had any success yet getting down to only the regex error like @thefinn93 did. My errors are coming from the boost logging library. My best guess right now is a [breaking change in the logging library](http://www.boost.org/doc/libs/1_63_0/libs/log/doc/html/log/changelog.html) between boost 1.58 (on Ubuntu 16.04) and boost 1.61 (on Ubuntu 16.10), but I haven't tracked it down yet. Debian sid is on 1.62 and jessie is on 1.55, so this probably explains the breaking on Kali linux as well.

vabiro, 2017-04-08:
Hi,
I am having the same issue with gcc version 6.3.1 20170306 on ArchLinux 4.4.39-1-ARCH

Thanks

robotastic, 2017-04-11:
This should be fixed with the change @nkwood made. Let me know if you still have issues. Closing for now, but we can reopen if it still doesn't work.

closed 2017-04-11 by robotastic