TrunkRecorder/trunk-recorder #84: Conventional P25 Recording

imagesafari, 2017-01-10:
I am absolutely LOVING the conventional recording. So cool.

We have a few conventional non-trunked P25 repeaters around here as well. Feature request :)

robotastic, 2017-01-10:
That's great to hear!

Cool! I think I could also add that added. I don't know there were many systems setup like that.

Treehouseman, 2017-01-16:
I'm gonna go ahead and take a stab at this, aiming for AM, FM, or P25 conventional recorders, gonna set the type by moving the system_modulation variable from a global one to a per system one.

Treehouseman, 2017-01-16:
If you look at the 'testing' branch in my fork I've duplicated the functionality of standard fm conventional recorders with the p25 recorders. I haven't tested it as I don't have any systems like that, and I don't know what op25 will do when used in this manner but it should at least output audio still. Just set the system type to "conventionalP25".

I started to attempt to make it all one function and having proper system modulation settings per system but I'll have to devise a way to rebuild the flowchart when switching modulation types and that's not something I know enough about yet.

robotastic, 2017-01-17:
Thanks @Treehouseman ! I started from your code and built from there. The P25 recorder need a squelch added in. I was able to test it by tuning to one of the channels from a trunked system. It looks like it works fine, so I baked it in to the main release. It would be a bit of work to move the modulation type to the system and change it dynamically. You would need to stop the block, disconnect the graph and rewire it. Definitely not impossible, though. Has it been an issue? I guess it could be when you have lots of systems on a single source. I can look into it a bit more.

@imagesafari if you are able to test this out, let me know if it works.

Treehouseman, 2017-01-17:
It's not been an issue, but I don't like the overall idea of having to make specific recorders for every type of system, having analog and digital is one thing, but having AM/FM/P25/SmartNet/etc can get complicated when you have to check so many areas, plus I have ideas on where to build on and it would be much easier if I can just specify a type for a recorder rather than setting it up from the start as things may change after I've started recorder.

Right now I'm interested in just adding AM demod because then I can capture CB and ATC alongside HAM repeaters and static FM channels. But after that I'm possibly going to look into adding GR-DSD back in and building support for DMR and NXDN, I'm actually a bit surprised you never tried to add it in when you still had GR-DSD because there's so many of those systems in DC.

robotastic, 2017-01-17:
@Treehouseman definitely agree about having to define lots of recorders at the start. I used to dynamically add and remove them, but one problem with that is that you have to stop the entire gnuradio flowgraph and it started surfacing really weird bugs in gnuradio. It could be interesting to look at it again. I would be cool to just generate a recorder when a new call comes in, attach it to the correct source and then delete it after the call is finished.

Agreed on GR-DSD. It is probably worth bundling it back in. I am not as familiar with how DMR trunking works... if it is trunked at all. Most of the DMR systems in DC are hotels & businesses, but I am sure that could still be interesting.

Treehouseman, 2017-01-17:
If it's going to be a stopping the whole entire flowgraph deal then it's really not worth it, it doesn't take too long to restart the radios anyways, but DMR has various levels of trunking, and if DSD+ is based on GR-DSD, then I have an aproximate idea on how it handles it. 

It's annoying and extremely basic for a "trunked" system, but basically it's similar to smartnet in that there's a "control" that will specify a conversation on a channel, but there is no math for a frequency, they must all be predefined, meaning you have to figure out the map for yourself. And to add to that, each channel can hold 2 streams, either being voice or data. DSD+ handles this by doing left and right channel. I've not messed much with DSD+'s built in DMR trunking features but I've gotten it to work before. 

Provided we can get the message data about channel numbers, we can trigger recorders like with smartnet. But DMR also does a /lot/ of just static frequencies. So first priority would be a conventional DMR decoder, then to work on trunking for the trunked versions to save CPU usage.

I'm still curious about using FFT math to save on CPU usage instead of relying on squelch, doing just 1 fft for a radio to monitor all it's squelch controled channels /has/ to be more cpu efficient. I'm not sure exactly how it would be done, but I know if we can get an FFT graph view and see with our eyes power levels and know when one is active, the same data can be used math wise. While even being more efficient because you don't have to look at the out of channel portions of that radio's spectrum.

As for NXDN, it's pretty much the same as DMR but less common and almost always trunked from what I've seen, the only annoyance is it's a different demodulation type, I think the same one as phase 1 or phase 2 on P25. And there's 2 or 3 data speeds, and I think only one of them supports 2 streams like DMR.

As for what's available on DMR networks, yeah, it's typically businesses, but some of them can be interesting, like a lot of colleges use it.

imagesafari, 2017-01-17:
Awesome. I will test this tonight.

imagesafari, 2017-01-17:
Ok, i'm working on testing this.

I noticed that the systems each start again with the TG at 1 for conventional stuff. Would it be possible to specify the TG # for each "channel" in a conventional system?

robotastic, 2017-01-17:
I don't think that would be too hard to add, I would have to check on what
kind of formatting is allowed in the config files.

It might have to be like this:

"channels": [{tg: 2500, freq: 850000000}]

Would that work?

imagesafari, 2017-01-17:
Something like that would be awesome.

imagesafari, 2017-01-17:
@Treehouseman "conventional" DMR recording would be awesome. I'd love to add some of the ham DMR repeaters in the area to it.

Treehouseman, 2017-01-17:
What I did in my current ncurses branch to fix the duplication for multiple conventional systems I incorporated to my system for separating the networks on the website, each system has it's files prefixed with the NAC value on the talkgroup, and for conventional systems it's 123X$TG where X is iterated as it counts through the conventional systems. So my current single conventional system's files are prefixed with 1230 before the talkgroup, it's a bit ugly, but it means they can all exist in the same folder for the website, though given the loadtimes it's developing, I'm going to have to change it again and make individual system folders.

Alternatively, a quick fix is define the integer outside that foreach loop, so it keeps the count through all the systems.

imagesafari, 2017-01-17:
That's interesting. We have 14 sources feeding our system, and what we did to solve this in the meantime was to modify call.cc to add a predefined "system" prefix to the TGs, and each one of those only has one "system" in the config.

Being able to explicitly set a TG would be awesome.

imagesafari, 2017-01-17:
FYI as a quick POC, this seems to work. I need to try it on something with a bit more CPU, since the conventional p25 recorder seems to take more CPU processing than 4 analog recorders.

Treehouseman, 2017-01-17:
Have you seen my fork? There's a lot of specifications I've made to define the way things are recorded. I don't have explicit TG's for conventional, but I do have it keeping them separate, each conventional system defined would be recorded in it's own folder and have a unique prefix, I've also added in CSV separation, each system will only check it's own CSV file for identified talkgroups, I had issues where smartnet analog groups were being recorded as digital because it looked there first and saw a group with the same ID.

If you want to see what changes I've made and see if that's what you're looking for let me know and I'll merge it real quick to include the conventional p25 recorders and update the readme to include all current config.json changes.

imagesafari, 2017-01-17:
@Treehouseman I haven't looked, but I will this evening. Thanks!

Treehouseman, 2017-01-17:
It's been updated and should be able to run. There's a possibility it'll segfault when it tries to load >10 radios as that was the original array size I had used, but I believe I got all of them changed, if not, if you can get a coredump and get me the line, I'll fix it if necessary. Make sure you look over the top portion of the readme where I've shown the needed config.json changes.

Just make sure that you get ncurses6 installed and have your $TERM value set to xterm-256color (run 'export TERM=xterm-256color') or it won't do full color. Also, given you're using 14 radios, you're gonna want to make your terminal text size /really/ small, if it can even display at all, The way the screen arranges the widnows it's gonna be crazy wide. But I just need to reconfigure it to fix that.

imagesafari, 2017-01-17:
I will check it out. I'm not running that many radios on any one instance, but we have about 14 different instances running in various locations uploading.

Treehouseman, 2017-01-17:
Ah, that makes more sense, though I will note that currently one of the instances I'm running has 5 radios on it, and as soon as we get AM and FM sorted, it's gonna have another 5 or so, then once there's DMR/NXDN, another 10 probably. We're running it on a server that has the overhead for it, provided we get enough USB hosts to handle the bandwidth.

jgarvas, 2017-01-17:
Luke you've outdone yourself with this ncurses stuff.  I've literally been imagining a unitrunker like view in ncurses instead of the typical stdout output.  Can I suggest a possible view that can be optionally toggled on/off?   If you run unitrunker you can look at the system channel utilization and what TG is landing on a given voice channel at a given time.   This is a far older view of it but note how the frequency is shown down one column, then the name or alpha-tag of the TG currently on that voice channel, hit counts, etc, then we could potentially add which RTL dongle its being sourced from?   Gives you a nice over view of how busy a given system is, how mixed-mode smartnet allocates channges, etc.

http://www.scpdny.com/UT.gif

Treehouseman, 2017-01-17:
Luke didn't make the ncurses interface :P 
I can't add a window exactly like the unitrunker one, but I can make it so that you can have a more detailed recent calls window that shows what the talkgroup descriptioin is.

Have it formattted like so:
$Time - $TG - $NAC - $Radio - $Desc.substr(0,25) - $Length

I don't have access to the users getting added or dropped right now, nor the source data, while possible it adds a ton of overhead, plus you have to remember that there's very limited screen space. I don't think I can even cleanly trust it to print out the whole description for most of the talkgroups I've pulled from radiorefrence, they'd start line wrapping and causing it to look ugly.

jgarvas, 2017-01-17:
Sorry I thought luke changed his name in github I didn't realize you forked it.   And yeah I just mean the channel view.  Half the time the users view is semi pointless.  Totally understand limited screen space.

Treehouseman, 2017-01-17:
Give me a few, I'm about halfway through implementing it.

Treehouseman, 2017-01-18:
A unitrunker like window now exists in groups 5, 6, 7, and 8, the readme shows how each is configured.

imagesafari, 2017-01-18:
Confirm that conventional-p25 works from the branch. Sounds awesome too. :)

robotastic, 2017-02-24:
Wahoo!!

closed 2017-02-24 by robotastic

kcwebby, 2020-03-26:
Rehashing this old issue.   Did these options for setting TG ID's on Conventional frequencies ever get documented?   In general, the Conventional frequency documentation (analog and P25 seem to be a bit scarce.   If I knew more about it, I could write up some sample config.json's and dissect them, but I don't.  (struggling with them now) @robotastic