TrunkRecorder/trunk-recorder #86: Troubleshooting - No recorded calls

zeryl, 2017-01-18:
Hello all.  I know that there isn't a support structure in place, so I apologize for asking via a github issue, but not sure where else to turn. 

Running Master (from last night), I setup a local system (may very well be slow on activity, due to where we are), and not getting any recordings.  I've pastebinn'd the log since running, to see if the errors shown are something I can fix, or what.

http://pastebin.com/r4dY17uN

Any suggestions?  I assume/home the "Call not found" is just because I've not got something setup right, but I just don't know.

Thank you in advance!

robotastic, 2017-01-18:
Not a problem! There is a link to Gitter in the Readme if you want to try
that, but this totally works.

Could you also post your config.sys?
It looks like you may have specified the modulation wrong, it should be
"qpsk"
It could also be that you need to specify the amount of tuning error. The
HackRF generally has some. It does look like it is tuning in the control
channel.

zeryl, 2017-01-18:
Config.json:

`{
    "sources": [{
        "center": 169087500.0,
        "rate": 10000000.0,
        "squelch": -50,
        "error": 0,
        "gain": 40,
        "ppm": 2,
        "antenna": "TX/RX",
        "digitalRecorders": 2,
        "driver": "osmosdr",
        "device": "hackrf",
        "modulation": "qpsk",
        "ifgain": "16",
        "bbgain": "42"
    }],
    "systems": [{
        "control_channels": [169662500],
        "type": "p25",
        "talkgroupsFile": "ChanList.csv"
    }]
}`

I do have some tuning error accounted for.  Like I mentioned, this may just be a dead system for me, and I may not be able to get much (rural area, but closest transmitter for the system).  Unfortunately, the next closest is just a bit out of reach, even for my antenna, to get a reliable decode on the control channel.  I believe I've even offset enough for the DC spike on the hackrf, but that could also be an issue, I guess.

robotastic, 2017-01-18:
Ah - so part of the problem could be around the gain. The HackRF actually
has 3 gains that could be set. The best thing to do is get a copy of gqrx
and tune the gain until you get a good receive. There is also an "lnaGain"
and a "bbGain"

I also may have implemented the PPM error wrong, so it may be a good idea
to use the "error" setting instead. For that you just specifiy the amount
of Hz that it is off by.

zeryl, 2017-01-18:
Gotcha.  I'll throw the box back on a monitor to get gqrx reporting good values, and check.  And understood on the ppm vs error, I'll check that out too.

zeryl, 2017-01-18:
Worked through it, and things are looking much better.  Using the ncruses interface, i'm seeing decodes, and getting some talk, thank you!

robotastic, 2017-01-27:
Cool - good to hear it worked!

closed 2017-01-27 by robotastic