TrunkRecorder/trunk-recorder #87: Aborted (Core Dumped) on new install.

jonathanendersby, 2017-01-23:
A brand new install as per the wiki guide for 16.04.
```
jonathan@tesla:~/radio/trunk-recorder$ cat /etc/issue
Ubuntu 16.04.1 LTS \n \l
```

The build looked good... I could redo it and copy/paste the output if that might help?

This is the output from running ./recorder
```
jonathan@tesla:~/radio/trunk-recorder$ ./recorder
linux; GNU C++ version 5.3.1 20151219; Boost_105800; UHD_003.009.002-0-unknown

[2017-01-23 08:03:27.525090] [0x00007f8158c4e900] [info]    System Type: smartnet
[2017-01-23 08:03:27.525166] [0x00007f8158c4e900] [info]    Control Channels:
[2017-01-23 08:03:27.525237] [0x00007f8158c4e900] [info]    8e+06
[2017-01-23 08:03:27.525253] [0x00007f8158c4e900] [info]
[2017-01-23 08:03:27.525265] [0x00007f8158c4e900] [info]    API Key:
[2017-01-23 08:03:27.525280] [0x00007f8158c4e900] [info]    Short Name: sys_1
[2017-01-23 08:03:27.525294] [0x00007f8158c4e900] [info]    Upload Script:
[2017-01-23 08:03:27.525307] [0x00007f8158c4e900] [info]    Talkgroups File: ChanList.csv
[2017-01-23 08:03:27.525338] [0x00007f8158c4e900] [info]    Capture Directory: /home/jonathan/radio/trunk-recorder
[2017-01-23 08:03:27.525354] [0x00007f8158c4e900] [info]    Config Directory: /home/jonathan/radio/trunk-recorder
[2017-01-23 08:03:27.525368] [0x00007f8158c4e900] [info]    Upload Server: encode-upload.sh
[2017-01-23 08:03:27.525379] [0x00007f8158c4e900] [info]    Default Mode: digital
[2017-01-23 08:03:27.525390] [0x00007f8158c4e900] [info]    Call Timeout (seconds): 3
[2017-01-23 08:03:27.525443] [0x00007f8158c4e900] [info]    Center: 8.54e+08
[2017-01-23 08:03:27.525464] [0x00007f8158c4e900] [info]    Rate: 8e+06
[2017-01-23 08:03:27.525481] [0x00007f8158c4e900] [info]    Error: 0
[2017-01-23 08:03:27.525495] [0x00007f8158c4e900] [info]    PPM Error: 0
[2017-01-23 08:03:27.525510] [0x00007f8158c4e900] [info]    Gain: 40
[2017-01-23 08:03:27.525521] [0x00007f8158c4e900] [info]    IF Gain: 0
[2017-01-23 08:03:27.525534] [0x00007f8158c4e900] [info]    BB Gain: 0
[2017-01-23 08:03:27.525547] [0x00007f8158c4e900] [info]    LNA Gain: 0
[2017-01-23 08:03:27.525559] [0x00007f8158c4e900] [info]    MIX Gain: 0
[2017-01-23 08:03:27.525577] [0x00007f8158c4e900] [info]    Squelch: -50
[2017-01-23 08:03:27.525594] [0x00007f8158c4e900] [info]    Idle Silence: false
[2017-01-23 08:03:27.525609] [0x00007f8158c4e900] [info]    Digital Recorders: 2
[2017-01-23 08:03:27.525622] [0x00007f8158c4e900] [info]    Debug Recorders: 0
[2017-01-23 08:03:27.525635] [0x00007f8158c4e900] [info]    Analog Recorders: 0
[2017-01-23 08:03:27.525647] [0x00007f8158c4e900] [info]    Driver: osmodr
[2017-01-23 08:03:27.525664] [0x00007f8158c4e900] [info]    Modulation: qpks
[2017-01-23 08:03:27.525681] [0x00007f8158c4e900] [info]    Max HZ: 8.58e+08
[2017-01-23 08:03:27.525695] [0x00007f8158c4e900] [info]    Min HZ: 8.5e+08
Using Volk machine: avx_64_mmx_orc
[2017-01-23 08:03:27.535852] [0x00007f8158c4e900] [error]   Size of LPF: 24243
[2017-01-23 08:03:27.546014] [0x00007f8158c4e900] [info]    Arb Rate: 0.498 Half band: 0.249 bw: 0.1992 tb: 0.0996
Project 25 IMBE Encoder/Decoder Fixed-Point implementation
Developed by Pavel Yazev E-mail: pyazev@gmail.com
Version 1.0 (c) Copyright 2009
This program comes with ABSOLUTELY NO WARRANTY.
This is free software, and you are welcome to redistribute it
under certain conditions; see the file ``LICENSE'' for details.
recorder: /usr/include/boost/smart_ptr/shared_ptr.hpp:648: typename boost::detail::sp_member_access<T>::type boost::shared_ptr<T>::operator->() const [with T = gr::basic_block; typename boost::detail::sp_member_access<T>::type = gr::basic_block*]: Assertion `px != 0' failed.
Aborted (core dumped)
jonathan@tesla:~/radio/trunk-recorder$
```

I'm using one rtlsdr dongle that I know works well (can use it with unitrunker on a windows machine)

This is my config, although I must admit I had to do some guessing re the "driver" section. I think the docs could be a bit more explicit there, particularly since a lof of noobs (like me) will come here with very little knowledge and a $8 dongle. 

```
jonathan@tesla:~/radio/trunk-recorder$ cat config.json
{
    "sources": [{
        "center": 854000000.0,
        "rate": 8000000.0,
        "squelch": -50,
        "error": 0,
        "gain": 40,
        "antenna": "TX/RX",
        "digitalRecorders": 2,
        "driver": "osmodr",
        "device": "",
        "modulation": "qpsk"
    }],
    "systems": [{
        "control_channels": [8000000.0],
        "type": "smartnet",
        "talkgroupsFile": "ChanList.csv"
    }]
}
```

dreinhold, 2017-01-23:
It's caused by the config.
Few things, you say you have an $8 dongle, but you have the rate set to 8mhz. Most likely you are using one that has a max around 3.2mhz I run mine at 2.04mhz [ "rate": 2048000 ].
Second issue is your control_channels you have it set to 800mhz which is 54mhz from center (you would need an SDR with a 108mhz bandwidth which I know you don't have. Your center frequency can not be more than 1/2 your rate from the control channel, With that center frequency using a 2.04mhz rate your control channel can not be less than 852.98mhz. 
If you post the radioreference link to the system you are trying to monitor we can help with the center and control channel settings.
Also last one, you don't have an error set. There is a good chance your dongle is not 100% spot on and you will need to set an error offset to compensate for that.

jonathanendersby, 2017-01-23:
Hi Dylan, Thanks for the quick reply.

I've updated my config, making my centre channel very close to my control channel (within the 2mhz) and changed my rate to 2048000, but I get the same problem.

There is no radioreference entry for the trunk I'm listening to, (I'm not in the US), but I do know itss control frequency and each of the channel frequencies very well. (I am unable to publish these)

My config is now:
```
{
    "sources": [{
        "center": 800000000.0,
        "rate": 2048000,
        "squelch": -50,
        "error": 0,
        "gain": 40,
        "antenna": "TX/RX",
        "digitalRecorders": 2,
        "driver": "osmodr",
        "device": "",
        "modulation": "qpsk"
    }],
    "systems": [{
        "control_channels": [800500000.0],
        "type": "smartnet",
        "talkgroupsFile": "ChanList.csv"
    }]
}
```

But I'm getting the same Abort as before. (I get the same thing if I use the actual centre and control frequencies.)

```
jonathan@tesla:~/radio/trunk-recorder$ ./recorder 
linux; GNU C++ version 5.3.1 20151219; Boost_105800; UHD_003.009.002-0-unknown

[2017-01-23 09:54:11.448572] [0x00007f009bd16900] [info]    System Type: smartnet
[2017-01-23 09:54:11.448680] [0x00007f009bd16900] [info]    Control Channels: 
[2017-01-23 09:54:11.448748] [0x00007f009bd16900] [info]    8.005e+08 
[2017-01-23 09:54:11.448765] [0x00007f009bd16900] [info]    
[2017-01-23 09:54:11.448779] [0x00007f009bd16900] [info]    API Key: 
[2017-01-23 09:54:11.448795] [0x00007f009bd16900] [info]    Short Name: sys_1
[2017-01-23 09:54:11.448809] [0x00007f009bd16900] [info]    Upload Script: 
[2017-01-23 09:54:11.448823] [0x00007f009bd16900] [info]    Talkgroups File: ChanList.csv
[2017-01-23 09:54:11.448870] [0x00007f009bd16900] [info]    Capture Directory: /home/jonathan/radio/trunk-recorder
[2017-01-23 09:54:11.448886] [0x00007f009bd16900] [info]    Config Directory: /home/jonathan/radio/trunk-recorder
[2017-01-23 09:54:11.448900] [0x00007f009bd16900] [info]    Upload Server: encode-upload.sh
[2017-01-23 09:54:11.448913] [0x00007f009bd16900] [info]    Default Mode: digital
[2017-01-23 09:54:11.448926] [0x00007f009bd16900] [info]    Call Timeout (seconds): 3
[2017-01-23 09:54:11.448978] [0x00007f009bd16900] [info]    Center: 8e+08
[2017-01-23 09:54:11.448999] [0x00007f009bd16900] [info]    Rate: 2.048e+06
[2017-01-23 09:54:11.449016] [0x00007f009bd16900] [info]    Error: 0
[2017-01-23 09:54:11.449031] [0x00007f009bd16900] [info]    PPM Error: 0
[2017-01-23 09:54:11.449047] [0x00007f009bd16900] [info]    Gain: 40
[2017-01-23 09:54:11.449058] [0x00007f009bd16900] [info]    IF Gain: 0
[2017-01-23 09:54:11.449068] [0x00007f009bd16900] [info]    BB Gain: 0
[2017-01-23 09:54:11.449080] [0x00007f009bd16900] [info]    LNA Gain: 0
[2017-01-23 09:54:11.449093] [0x00007f009bd16900] [info]    MIX Gain: 0
[2017-01-23 09:54:11.449111] [0x00007f009bd16900] [info]    Squelch: -50
[2017-01-23 09:54:11.449128] [0x00007f009bd16900] [info]    Idle Silence: false
[2017-01-23 09:54:11.449146] [0x00007f009bd16900] [info]    Digital Recorders: 2
[2017-01-23 09:54:11.449159] [0x00007f009bd16900] [info]    Debug Recorders: 0
[2017-01-23 09:54:11.449172] [0x00007f009bd16900] [info]    Analog Recorders: 0
[2017-01-23 09:54:11.449184] [0x00007f009bd16900] [info]    Driver: osmodr
[2017-01-23 09:54:11.449201] [0x00007f009bd16900] [info]    Modulation: qpks
[2017-01-23 09:54:11.449218] [0x00007f009bd16900] [info]    Max HZ: 8.01024e+08
[2017-01-23 09:54:11.449233] [0x00007f009bd16900] [info]    Min HZ: 7.98976e+08
Using Volk machine: avx_64_mmx_orc
[2017-01-23 09:54:11.452561] [0x00007f009bd16900] [error]   Size of LPF: 6207
[2017-01-23 09:54:11.464663] [0x00007f009bd16900] [info]    Arb Rate: 0.492188 Half band: 0.246094 bw: 0.196875 tb: 0.0984375
Project 25 IMBE Encoder/Decoder Fixed-Point implementation
Developed by Pavel Yazev E-mail: pyazev@gmail.com
Version 1.0 (c) Copyright 2009
This program comes with ABSOLUTELY NO WARRANTY.
This is free software, and you are welcome to redistribute it
under certain conditions; see the file ``LICENSE'' for details.
recorder: /usr/include/boost/smart_ptr/shared_ptr.hpp:648: typename boost::detail::sp_member_access<T>::type boost::shared_ptr<T>::operator->() const [with T = gr::basic_block; typename boost::detail::sp_member_access<T>::type = gr::basic_block*]: Assertion `px != 0' failed.
Aborted (core dumped)
```

I'm working under the assumption that if the program runs but can't find a valid trunk control on its control frequency, that it'll report something to that effect via stdout? (and not Abort?).

I feel like there's something else we're missing. Could you please share a complete config for an rtl-sdr that I can test with? (Obviously I realise it won't find a trunk etc, but hopefully it will not Abort?)

robotastic, 2017-01-23:
There are a couple other things that might be causing problems. I would turn the gain down a little. Have you looked at the signal using GQRX? http://gqrx.dk/ <http://gqrx.dk/>
That will help you find the amount of error needed and gain. Also, if this is a SmartNet system, the modulation should be “fsk4”

Also - since the system is down around 800MHz, you may need to mess with smartnet_parser.cc because you probably have a custom band plan. I have an example up above on what I had to do for WMATA’s system.

Still… the core dump is the real problem. It looks like the real problem is with something in the config file not parsing correctly. Maybe get rid of the device line?

My guess is that I have some weird bug and you are the first to find it. I can take a longer look at it tonight. Maybe try start with an RTL / WMata config from the examples folder and modding that.

porcej, 2017-01-23:
I think the issue here is the driver.  the driver should be "osmosdr" not 'osmodr'. Also, I believe you do need a device as well.

`{
    "sources": [{
        "center": 800000000.0,
        "rate": 2048000,
        "squelch": -50,
        "error": 0,
        "gain": 40,
        "antenna": "TX/RX",
        "digitalRecorders": 2,
        "driver": "osmosdr",
        "device": "",
        "modulation": "qpsk"
    }],
    "systems": [{
        "control_channels": [800500000.0],
        "type": "smartnet",
        "talkgroupsFile": "ChanList.csv"
    }]
}`

robotastic, 2017-01-23:
I think Joe found the bug! It is almost definitely the driver mispelling. I
will add an error check on that.

If you only have one device attached, the osmosdr driver will do a good job
of finding it, so you don't need to specify the "device".

jonathanendersby, 2017-01-24:
Ah! That was indeed it and I am incredibly embarrassed! Thank you all. (Perhaps we need to add a helper script that can check the config file for sanity and protect against fudging like that?... I could definitely write this in python if that would help?

robotastic, 2017-01-24:
Cool! I will go add better failure more, along with some helpful messages
on the config.json parsing. It definitely shouldn't core dump!

robotastic, 2017-03-10:
Added in checking to make sure the driver name is spelled correctly.

closed 2017-03-10 by robotastic