TrunkRecorder/trunk-recorder #88: Core Dump on poor reception

porcej, 2017-01-23:
When trying to receive a poor quality signal, an error condition arises that produces a seg fault. 

### Background 
My system is setup to receive Fairfax Co, Arlington, Co, and Alexandria City.  The system consists of 5 rtl-sdr dongles.  The dongle that covers the control channels uses a Yagi pointed in such a way to cover all three systems.  The other 4 dongles use unity-gain omnidirectional antennas.

I placed my system on a tower with good reception for all three radio systems.  While operating on the tower I was able to receive all three systems without error.  Last weekend (1/13) I brought the system down off the tower to facilitate, system updates, software upgrades (3rd party software), and to place the systems on beta.openmhz.com. 

### Issue
After taking the system down off the tower, I noticed that trunk-recorder would crash periodically.  Sometimes it would run for hours without a crash and sometimes it would run for a moment or two before crashing.  When the program crashed, its output was:

##### STDOUT
``` 
[2017-01-18 17:25:18.672555] [0x00007fd7c1ba1900] [info]    tsbk04	Unit to Unit Chan Update	Channel ID:   267	Freq: 852.675	Target ID:    8144	TDMA 0	Source ID: 8167
[2017-01-18 17:25:19.173424] [0x00007fd7c1ba1900] [info]    tsbk04	Unit to Unit Chan Update	Channel ID:   267	Freq: 852.675	Target ID:    8144	TDMA 0	Source ID: 8167
[2017-01-18 17:25:19.173480] [0x00007fd7c1ba1900] [info]    tsbk04	Unit to Unit Chan Update	Channel ID:   267	Freq: 852.675	Target ID:    8144	TDMA 0	Source ID: 8167
[2017-01-18 17:25:20.175089] [0x00007fd7c1ba1900] [info]    tsbk04	Unit to Unit Chan Update	Channel ID:   267	Freq: 852.675	Target ID:    8144	TDMA 0	Source ID: 8167
[2017-01-18 17:25:20.175172] [0x00007fd7c1ba1900] [info]    tsbk04	Unit to Unit Chan Update	Channel ID:   267	Freq: 852.675	Target ID:    8144	TDMA 0	Source ID: 8167
[2017-01-18 17:25:20.675732] [0x00007fd7c1ba1900] [info]    tsbk04	Unit to Unit Chan Update	Channel ID:   267	Freq: 852.675	Target ID:    8144	TDMA 0	Source ID: 8167
[2017-01-18 17:25:20.675796] [0x00007fd7c1ba1900] [info]    tsbk04	Unit to Unit Chan Update	Channel ID:   267	Freq: 852.675	Target ID:    8144	TDMA 0	Source ID: 8167
[2017-01-18 17:25:21.176542] [0x00007fd7c1ba1900] [info]    tsbk04	Unit to Unit Chan Update	Channel ID:   267	Freq: 852.675	Target ID:    8144	TDMA 0	Source ID: 8167
[2017-01-18 17:25:21.777398] [0x00007fd7c1ba1900] [info]    tsbk04	Unit to Unit Chan Update	Channel ID:   267	Freq: 852.675	Target ID:    8144	TDMA 0	Source ID: 8167
[2017-01-18 17:25:21.777463] [0x00007fd7c1ba1900] [info]    tsbk04	Unit to Unit Chan Update	Channel ID:   267	Freq: 852.675	Target ID:    8144	TDMA 0	Source ID: 8167
[2017-01-18 17:25:22.278276] [0x00007fd7c1ba1900] [info]    tsbk04	Unit to Unit Chan Update	Channel ID:   267	Freq: 852.675	Target ID:    8144	TDMA 0	Source ID: 8167
[2017-01-18 17:25:22.278350] [0x00007fd7c1ba1900] [info]    tsbk04	Unit to Unit Chan Update	Channel ID:   267	Freq: 852.675	Target ID:    8144	TDMA 0	Source ID: 8167
[2017-01-18 17:25:22.779096] [0x00007fd7c1ba1900] [info]    tsbk04	Unit to Unit Chan Update	Channel ID:   267	Freq: 852.675	Target ID:    8144	TDMA 0	Source ID: 8167
[2017-01-18 17:25:22.779192] [0x00007fd7c1ba1900] [info]    tsbk04	Unit to Unit Chan Update	Channel ID:   267	Freq: 852.675	Target ID:    8144	TDMA 0	Source ID: 8167
[2017-01-18 17:25:22.779304] [0x00007fd7c1ba1900] [info]    tsbk04	Unit to Unit Chan Update	Channel ID:   267	Freq: 852.675	Target ID:    8144	TDMA 0	Source ID: 8167
[2017-01-18 17:25:23.279870] [0x00007fd7c1ba1900] [info]    tsbk04	Unit to Unit Chan Update	Channel ID:   267	Freq: 852.675	Target ID:    8144	TDMA 0	Source ID: 8167
[2017-01-18 17:25:23.780636] [0x00007fd7c1ba1900] [info]    tsbk04	Unit to Unit Chan Update	Channel ID:   267	Freq: 852.675	Target ID:    8144	TDMA 0	Source ID: 8167
[2017-01-18 17:25:23.780732] [0x00007fd7c1ba1900] [info]    tsbk04	Unit to Unit Chan Update	Channel ID:   267	Freq: 852.675	Target ID:    8144	TDMA 0	Source ID: 8167
[2017-01-18 17:25:24.281388] [0x00007fd7c1ba1900] [info]    tsbk04	Unit to Unit Chan Update	Channel ID:   267	Freq: 852.675	Target ID:    8144	TDMA 0	Source ID: 8167
[2017-01-18 17:25:24.281466] [0x00007fd7c1ba1900] [
```
##### STDERR
```
Segmentation fault (core dumped)
```

### Troubleshooting Steps
After several days of troubleshooting, the problem was isolated to the Arlington system (removing the Arlington Co. system in config.json resolved the issues).  

After identifying Arlington as the source of the issue, I started looking at Arlington's system on GQRX using the yagi.  Everything appeared to be fine good SNR on all of Arlington's channels.   Next, I looked at the GQRX using one of the omnidirectional antennas.  Here I noticed that the signal was just barely above background noise.   I then replaced 3 of the omni antennas with yagis pointed at Arlington's tower and run trunk-recorder (4 yagi's and 1 omni).  I noticed that trunk-recorder would crash sometimes (but not every time), that a channel covered by the omni antenna was received.  This leads me to believe the issue is related to poor reception.

Any thoughts on this issue?

I have stopped monitoring Arlington's system until the system is put back on the tower (some time later this week.)

Thanks,
Joe

robotastic, 2017-01-23:
This is a very well documented bug report, thanks Joe!!

Would you be able to gnu debugger on the core dump? That will help point to
which function is causing it to crash.

to capture the core dump, run:
ulimit -c unlimited

that will capture what was in memory when it crashed.
You can then run the gdb command against it. I think it is:
gdb recorder corefile

Then type in the command `bt` and copy the output to here.
The `info threads` command might also help.

Treehouseman, 2017-01-23:
Do the unit to unit calls give a debug message for every update packet? I know normal calls are just for the start/end/retune, but that's some /very/ rapid messages for one conversation. I don't recall ever seeing any on my systems to compare it to sadly.

porcej, 2017-01-23:
@robotastic Running, waiting for a crash.

@Treehouseman Unit to unit (private calls) do appear to give numerous debug messages.  This can be seen on Alex and Fairfax's systems.  I was also able to test this by private calling another radio and confirming trunk-recorder's output.

robotastic, 2017-01-23:
Cool! I actually have never seen the private calling pop up. Does that get
recorded? I could easily see this causing some trouble. I will go walk
through the code branch and see if I can figure out what is happening.

Let me know if you get a crash.

porcej, 2017-01-23:
> Cool! I actually have never seen the private calling pop up. Does that get
recorded? I could easily see this causing some trouble. I will go walk
through the code branch and see if I can figure out what is happening.

Yes, the private calls are recorded.  They come up with a ?random? talkgroup.   I have added all the private call talkgroups I've heard to the talkgroup list with a priority of 200 as most of the private calls involve admin related activities (at least on the fire side of things).

Treehouseman, 2017-01-23:
The "TG" for the private calls is going to be the radio ID's, or at least it should be, same numbers recorded in the source list for normal calls.

Treehouseman, 2017-01-23:
Alternatively, if after the crash is fixed and you still don't want /any/ unit to unit calls, you can always try to comment out the unit to unit code in p25_trunking.cc.

porcej, 2017-01-23:
@Treehouseman `The "TG" for the private calls is going to be the radio ID's, or at least it should be, same numbers recorded in the source list for normal calls.`

This does not seem to be the case.  The talkgroups seem to be numerically sequential after all the "defined" talkgroups on the system.  For example, if system A has talk groups 1-200, the first private call channels will be 201, 202, 203....  From the example above, the source radio id is 8167, the target radio id is 8144 and the talkgroup is 267.

porcej, 2017-01-24:
Murphy strikes again!  No crashes tonight.  Will try again on Wednesday.

robotastic, 2017-01-24:
Murphy is the worst… I will take a look through the code and see if I can spot anything obvious.

robotastic, 2017-01-24:
I checked through the code, and I didn’t see anything obvious… it looks like Unit Calls should be treated like normal calls. The only difference could be that the updates are coming in a lot faster… but again, that shouldn’t matter. This section of the code is single threaded. On the debug message, the channel ID actually refers to the frequency. It looks like the system just rotates through the channels that are available. The talkgroup is actually set to the target id.

Let me know if you are able to capture a core file and we can probably pin point it.

porcej, 2017-01-24:
The issue appeared this morning.  I had remoted in from my phone so I was unable to perform the requested debugging.  I will try later, when I have access to a computer.

##### STDOUT & STDERR
```
[2017-01-24 09:05:47.224697] [0x00007f20ebfc0900] [info]        Retune - Elapsed: 14s   Since update: 0s        Talkgroup: 1    Old Freq: 8.55912e+08   New Freq: 8.56438e+08
[2017-01-24 09:05:47.726278] [0x00007f20ebfc0900] [info]         Update Retune - Total calls: 10        Talkgroup: 1    Old Freq: 8.56438e+08   New Freq: 8.55912e+08
[2017-01-24 09:05:47.726313] [0x00007f20ebfc0900] [info]        Retune - Elapsed: 14s   Since update: 0s        Talkgroup: 1    Old Freq: 8.56438e+08   New Freq: 8.55912e+08
[2017-01-24 09:05:48.227655] [0x00007f20ebfc0900] [info]         Update Retune - Total calls: 10        Talkgroup: 1    Old Freq: 8.55912e+08   New Freq: 8.56438e+08
[2017-01-24 09:05:48.227690] [0x00007f20ebfc0900] [info]        Retune - Elapsed: 15s   Since update: 0s        Talkgroup: 1    Old Freq: 8.55912e+08   New Freq: 8.56438e+08
[2017-01-24 09:05:48.228718] [0x00007f20ebfc0900] [error]        Call not found for Update Message, Starting one...  Talkgroup: 73      Freq: 8.53738e+08
[2017-01-24 09:05:51.835159] [0x00007f20ebfc0900] [info]        Retune - Elapsed: 18s   Since update: 3s        Talkgroup: 1    Old Freq: 8.56438e+08   New Freq: 8.60762e+08
[2017-01-24 09:05:51.835192] [0x00007f20ebfc0900] [info]                Stopping call, starting new call on new source
[2017-01-24 09:05:51.835209] [0x00007f20ebfc0900] [info]    Ending Recorded Call        TG: 1   Last Update: 0 Call Elapsed: 18
Segmentation fault (core dumped)
```

Treehouseman, 2017-01-24:
I'm an idiot! I just remembered that I had to fix a segfault in my fork, it had to do with grabbing the call length, that function you added for the wav block elapsed time on conventional recorders was also being used for normal calls. While that function itself was fine in usage, it would randomly segfault. It would happen when an update triggered or it scanned for inactive recorders, I'm in class now but I remember changing it to just the original elapsed value or something. Haven't had a segfault since.

Treehouseman, 2017-01-24:
Initially I didn't open an issue for it because I had assumed it was something I caused. In slack I posted this error message from GDB 
"Program terminated with signal SIGSEGV, Segmentation fault.
#0  Call::set_freq (this=this@entry=0x3ef3cc0, f=853962500) at /home/tripp/share/trtest/trunk-recorder/trunk-recorder/call.cc:190
190           position = get_recorder()->get_current_length();"

porcej, 2017-01-24:
##### STDOUT & STDERR 
```
[2017-01-24 10:18:19.511184] [0x00007f007aea7900] [info]    Starting Recorder on Src: rtl=20
[2017-01-24 10:18:19.511199] [0x00007f007aea7900] [info]    p25_recorder.cc: Starting Logger    [ 0 ] - freq[ 8.51712e+08]       talkgroup[ 5 ]
[2017-01-24 10:18:20.012258] [0x00007f007aea7900] [info]    Ending Recorded Call        TG: 63  Last Update: 4 Call Elapsed: 11
[2017-01-24 10:18:20.012371] [0x00007f007aea7900] [error]   p25_recorder.cc: Stopping Logger    [ 6 ] - freq[ 8.55912e+08]       talkgroup[ 63 ]
[2017-01-24 10:18:20.198743] [0x00007efedf74e700] [info]    SSL Exception: short read
[2017-01-24 10:18:20.198888] [0x00007efedf74e700] [info]    HTTPS Upload result: 1
[2017-01-24 10:18:21.514672] [0x00007f007aea7900] [info]    tbsk02      Moto Patch Grant        Channel ID:  1381       Freq: 859.638   sg    1059     TDMA 0   sa 3171339
[2017-01-24 10:18:21.514694] [0x00007f007aea7900] [info]    Starting Recorder on Src: rtl=28
[2017-01-24 10:18:21.514705] [0x00007f007aea7900] [info]    p25_recorder.cc: Starting Logger    [ 12 ] - freq[ 8.59638e+08]      talkgroup[ 1059 ]
[2017-01-24 10:18:33.731712] [0x00007f007aea7900] [info]    tsbk00      Chan Grant      Channel ID:  1189       Freq: 858.438   ga      27      TDMA 0 sa 3130229       Encrypt 0       Bandwidth: 12.5
[2017-01-24 10:18:33.731746] [0x00007f007aea7900] [info]    Starting Recorder on Src: rtl=27
[2017-01-24 10:18:33.731763] [0x00007f007aea7900] [info]    p25_recorder.cc: Starting Logger    [ 9 ] - freq[ 8.58438e+08]       talkgroup[ 27 ]
Segmentation fault (core dumped)
```
I'm getting No Stack. when attempting to run `bt` after `gdb recorder corefile`

Treehouseman, 2017-01-24:
What did gdb give you?

porcej, 2017-01-24:
```
porcej@kt3i:~/radio/trunk-recorder$ ulimit -c unlimited
porcej@kt3i:~/radio/trunk-recorder$ gdb recorder corefile
GNU gdb (Ubuntu 7.11.1-0ubuntu1~16.04) 7.11.1
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from recorder...done.
/home/porcej/radio/trunk-recorder/corefile: No such file or directory.
(gdb) bt
No stack.
(gdb) 
```

Treehouseman, 2017-01-24:
It says you didn't feed it a corefile, check it's called corefile and not
just core, that's what they were on my system, it'll be in the same folder
as recorder

porcej, 2017-01-24:
Thats it, thanks!  please see below

```
[New LWP 13549]
[New LWP 13490]
[New LWP 13582]
[New LWP 13492]
[New LWP 13552]
[New LWP 13519]
[New LWP 13563]
[New LWP 13575]
[New LWP 13569]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Core was generated by `./recorder'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x0000000000459a94 in Call::set_freq (this=this@entry=0x42cd940, f=854487500) at /home/porcej/radio/trunk-recorder/trunk-recorder/call.cc:189
189         freq_list[freq_count] = call_freq;
[Current thread is 1 (Thread 0x7f007aea7900 (LWP 13392))]
(gdb) bt
#0  0x0000000000459a94 in Call::set_freq (this=this@entry=0x42cd940, f=854487500) at /home/porcej/radio/trunk-recorder/trunk-recorder/call.cc:189
#1  0x000000000041f14c in update_recorder (message=..., sys=sys@entry=0x19c2620) at /home/porcej/radio/trunk-recorder/trunk-recorder/main.cc:663
#2  0x000000000041f6e8 in handle_message (messages=std::vector of length 1, capacity 1 = {...}, sys=sys@entry=0x19c2620)
    at /home/porcej/radio/trunk-recorder/trunk-recorder/main.cc:736
#3  0x000000000041fdff in monitor_messages () at /home/porcej/radio/trunk-recorder/trunk-recorder/main.cc:883
#4  0x0000000000417dac in main () at /home/porcej/radio/trunk-recorder/trunk-recorder/main.cc:1044
```

porcej, 2017-01-24:
Output from `info threads`:
```
 Id   Target Id         Frame 
* 1    Thread 0x7f007aea7900 (LWP 13392) 0x0000000000459a94 in Call::set_freq (this=this@entry=0x42cd940, f=854487500)
    at /home/porcej/radio/trunk-recorder/trunk-recorder/call.cc:189
  2    Thread 0x7f0061832700 (LWP 13403) pthread_cond_timedwait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:225
  3    Thread 0x7f00537fe700 (LWP 13407) pthread_cond_timedwait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:225
  4    Thread 0x7f0060830700 (LWP 13406) 0x00007f0076b48b5d in poll () at ../sysdeps/unix/syscall-template.S:84
  5    Thread 0x7f0032ffd700 (LWP 13415) pthread_cond_timedwait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:225
  6    Thread 0x7f002b7fe700 (LWP 13421) pthread_cond_timedwait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:225
  7    Thread 0x7eff27fdf700 (LWP 13515) pthread_cond_timedwait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:225
  8    Thread 0x7eff2bfe7700 (LWP 13507) pthread_cond_timedwait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:225
  9    Thread 0x7f0051ffb700 (LWP 13410) pthread_cond_timedwait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:225
  10   Thread 0x7f0063035700 (LWP 13400) pthread_cond_wait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_wait.S:185
  11   Thread 0x7eff217d2700 (LWP 13528) pthread_cond_timedwait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:225
  12   Thread 0x7eff3f7fe700 (LWP 13477) pthread_cond_timedwait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:225
  13   Thread 0x7eff4e7fc700 (LWP 13465) pthread_cond_timedwait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:225
  14   Thread 0x7eff03f97700 (LWP 13587) pthread_cond_timedwait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:225
  15   Thread 0x7effeffff700 (LWP 13427) pthread_cond_timedwait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:225
  16   Thread 0x7eff2d7ea700 (LWP 13504) pthread_cond_timedwait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:225
  17   Thread 0x7f00297fa700 (LWP 13425) pthread_cond_timedwait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:225
  18   Thread 0x7efefc788700 (LWP 13602) pthread_cond_timedwait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:225
  19   Thread 0x7eff367fc700 (LWP 13486) pthread_cond_timedwait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:225
  20   Thread 0x7eff3dffb700 (LWP 13480) pthread_cond_timedwait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:225
  21   Thread 0x7f0028ff9700 (LWP 13426) pthread_cond_timedwait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:225
    22   Thread 0x7f0062033700 (LWP 13402) pthread_cond_timedwait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:225
  23   Thread 0x7effeeffd700 (LWP 13429) pthread_cond_timedwait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:225
  24   Thread 0x7f006fbb2700 (LWP 13397) 0x00007f0076b48b5d in poll () at ../sysdeps/unix/syscall-template.S:84
  25   Thread 0x7effcffff700 (LWP 13434) pthread_cond_timedwait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:225
  26   Thread 0x7eff137b6700 (LWP 13556) pthread_cond_timedwait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:225
  27   Thread 0x7efefbf87700 (LWP 13603) pthread_cond_timedwait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:225
  28   Thread 0x7eff2dfeb700 (LWP 13503) pthread_cond_timedwait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:225
  29   Thread 0x7effef7fe700 (LWP 13428) pthread_cond_timedwait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:225
  30   Thread 0x7eff287e0700 (LWP 13514) pthread_cond_timedwait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:225
  31   Thread 0x7effedffb700 (LWP 13431) pthread_cond_timedwait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:225
  32   Thread 0x7eff47fff700 (LWP 13469) pthread_cond_timedwait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:225
  33   Thread 0x7efefaf85700 (LWP 13605) pthread_cond_timedwait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:225
  34   Thread 0x7eff377fe700 (LWP 13484) pthread_cond_timedwait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:225
  35   Thread 0x7eff357fa700 (LWP 13488) pthread_cond_timedwait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:225
  36   Thread 0x7effed7fa700 (LWP 13432) pthread_cond_timedwait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:225
  37   Thread 0x7eff477fe700 (LWP 13470) pthread_cond_timedwait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:225
[... cut ...]
  213  Thread 0x7eff25fdb700 (LWP 13519) pthread_cond_timedwait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:225
  214  Thread 0x7eff0ffaf700 (LWP 13563) pthread_cond_timedwait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:225
  215  Thread 0x7eff09fa3700 (LWP 13575) pthread_cond_timedwait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:225
  216  Thread 0x7eff0cfa9700 (LWP 13569) pthread_cond_timedwait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:225

```

Treehouseman, 2017-01-24:
I would rename the corefile to core2 or something and reset "ulimit -c unlimited" and let it run again to a crash, it's best to confirm this is the issue, it's possible it's just overloading various functions and it's just a waiting game for which one fails first.

porcej, 2017-01-24:
Ok, running again.

robotastic, 2017-01-24:
Interesting, based upon that it looks like there may have been a lot of
frequency changes during the call. If for some reason a call has more than
50 changes in the channel during the duration of a call, it could crash. I
will put in a check on the array bounds.

If it was a long data transmission that kept hopping frequencies, it could
trigger this. I can get a fix out tonight.
If you just wrap line 189 of call.cc with an if statement that checks if
freq_count > 49, that should fix it. You could also an else print out that
this happened.

robotastic, 2017-01-24:
@treehouseman - was this an error in your fork or the trunk-recorder?

Treehouseman, 2017-01-24:
Yes, after the initial code insertion, It would just randomly segfault always at line 190 in call.cc, even without having conventional recorders. My initial thoughts were that it was somehow slipping through the check for conventional calls and trying to get the length from the wav block so I disabled the check in main.cc and it seemed to fix it, then I did it properly and made normal calls only check via the old method and not the wav block, haven't had a segfault since. You should be able to see my changes to it on my commits.

porcej, 2017-01-24:
Second run seems to confirm original findings.

```
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Core was generated by `./recorder'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x0000000000459a94 in Call::set_freq (this=this@entry=0x34f5e90, f=853737500) at /home/porcej/radio/trunk-recorder/trunk-recorder/call.cc:189
189         freq_list[freq_count] = call_freq;
[Current thread is 1 (Thread 0x7f332335c900 (LWP 23826))]
```

robotastic, 2017-01-25:
@TreeHouseMan Ah! found the bug around get_current_length and conventional recording. When I call end_call and reset_call, I do not assign the new recorder to the recorder pointer in call. Fixed, that and also add some safety checks.

> On Jan 24, 2017, at 9:47 AM, Treehouseman <notifications@github.com> wrote:
> 
> get_recorder()->get_current_length();

robotastic, 2017-01-25:
Nevermind… recorder actually never gets unset. I did add in lots of safety checks though.

> On Jan 24, 2017, at 9:21 PM, Luke Berndt <lukekb@gmail.com> wrote:
> 
> @TreeHouseMan Ah! found the bug around get_current_length and conventional recording. When I call end_call and reset_call, I do not assign the new recorder to the recorder pointer in call. Fixed, that and also add some safety checks.
> 
> 
>> On Jan 24, 2017, at 9:47 AM, Treehouseman <notifications@github.com <mailto:notifications@github.com>> wrote:
>> 
>> get_recorder()->get_current_length();
>

Treehouseman, 2017-01-25:
That fix also applies to non conventional calls? because line 190 should have only been called via the conventional call check and it was also happening to **non** conventional setups

robotastic, 2017-01-25:
Well - I can’t figure out why the get_current_lenght() call would cause a crash, but I went with your fix. I also add in some protection that should helpfully catch the other crash. Everything is live in the master branch.

Treehouseman, 2017-01-25:
Argh, you made a mess of the naming structure again! At least I was gonna rework that section again anyways :P 

I still prefer at this point my naming scheme using NAC values rather than shortname, but that's really just personal preference due to always working with them that way, but that should be the only thing I have to change, I'm going to get rid of my nac prefix for normal files and just make it the temp file for uploading to my server. I really need to learn some more nodejs so I can modify it and make it better, While I like the new openmhz website, There's a lot more features I like about the old one, it was just nicer to deal with (though I do know the new one is in it's baby steps).

robotastic, 2017-01-25:
Ah! sorry about that. I wrapped all of the sprintfs with checks to make sure the filename+path wasn’t longer than the char array. definitely a corner case, but I wanted to make sure I was checking bounds on all arrays.

Thanks - More work is definitely needed, but it is a lot easier to use on a phone now.

robotastic, 2017-01-27:
I think this was fixed, I can reopen if it is not.

closed 2017-01-27 by robotastic

porcej, 2017-01-27:
Ah yes, sorry meant to close this earlier this week!  Thanks again for the fix!