TrunkRecorder/trunk-recorder #89: How to handle odd channel frequencies.

jonathanendersby, 2017-01-24:
I have a SmartNet system with a control channel near 417mhz and 11 analog channels distributed (with random offsets) between 417 and 419mhz. The entire system falls inside 2.4mhz and I have done the leg work to figure out each channel's frequency (using a couple of rtlsdrs, some patience and unitrunker). I have the system configured and working perfectly inside unitrunker.

Is there an easy way to configure this system in trunk recorder? I expected to be able to configure these channels in the "channels" entity under "system" in the config, but that's only for conventional systems.

Luke suggested that I might need to look in smartnet_trunking.cc, which I have, but it seems to be taking a mathematical approach to determining the various channel's frequencies, where I believe mine are randomly distributed. 

My output (which I'm assuming means that it's successfully decoding the trunk):

```   
[2017-01-24 06:52:41.583451] [0x00007f2f2d16d900] [error]   	 Call not found for Update Message, Starting one...  Talkgroup: 4560	Freq: 8.61838e+08
[2017-01-24 06:52:41.583548] [0x00007f2f2d16d900] [info]    	Recording not started because there was no source covering: 8.61838e+08 For TG: 4560
[2017-01-24 06:52:42.083925] [0x00007f2f2d16d900] [error]   	Control Channel Message Decode Rate: 1/sec, count:  3
[2017-01-24 06:52:45.186012] [0x00007f2f2d16d900] [error]   	Control Channel Message Decode Rate: 1.33333/sec, count:  4
[2017-01-24 06:52:45.286177] [0x00007f2f2d16d900] [info]    	Recording not started because there was no source covering: 8.61537e+08 For TG: 4880
[2017-01-24 06:52:48.188066] [0x00007f2f2d16d900] [error]   There is only one control channel defined
[2017-01-24 06:52:48.188114] [0x00007f2f2d16d900] [error]   	Control Channel Message Decode Rate: 0.666667/sec, count:  2
[2017-01-24 06:52:49.088765] [0x00007f2f2d16d900] [error]   	 Call not found for Update Message, Starting one...  Talkgroup: 28689	Freq: 8.51287e+08
[2017-01-24 06:52:49.088877] [0x00007f2f2d16d900] [info]    	Recording not started because there was no source covering: 8.51287e+08 For TG: 28689
[2017-01-24 06:52:51.190283] [0x00007f2f2d16d900] [error]   	Control Channel Message Decode Rate: 2/sec, count:  6
[2017-01-24 06:52:54.192292] [0x00007f2f2d16d900] [error]   	Control Channel Message Decode Rate: 2.66667/sec, count:  8
[2017-01-24 06:52:57.194277] [0x00007f2f2d16d900] [error]   	Control Channel Message Decode Rate: 1.66667/sec, count:  5
```
Is this something that Trunk Recorder already caters for?

Thanks in advance! 
Jonathan.

Treehouseman, 2017-01-24:
I'd just hardcode it, it's not too bad provided you know what exactly what the channel numbers are. The math is in either smartnet-trunking.cc or smartnet-parser.cc, make it a fixed switch case for the channels with a debug message for unknowns, currently it's jut a math thing from the channel number based on the control channel or something.

dreinhold, 2017-01-25:
From the output you are not getting a good decode from the control channel.
> Control Channel Message Decode Rate: 2.66667/sec, count:  8

With a smartnet system it sends 43 messages per second, if you are only getting 1 or 2 of them so you are missing a lot of traffic.
Mess with your error correction a bit, if you have unitrunker working see what PPM rate it is using and try the same one (pretty sure trunk-recorder wont take decimals so just round up or down).

Also trunk-recorder is only going to spit out the message 
> Control Channel Message Decode Rate: 2.66667/sec, count:  8

if its below 3 so you might want for testing to remove that if statement in main.cc [line 834](https://github.com/robotastic/trunk-recorder/blob/383a592bd12a67310773d28a75dbc93f5f3c6ee0/trunk-recorder/main.cc#L834-L836) so you can see when you get to 43/sec.

jonathanendersby, 2017-01-25:
Thanks Dylan, I've got a solid lock on the trunk now and am no longer receiving any Decode Rate warnings.

closed 2017-03-10 by robotastic