TrunkRecorder/trunk-recorder #90: What is source modulation?

jonathanendersby, 2017-01-25:
The documentation says that the source modulation is: 

> "the type of modulation that the system uses. The options are qpsk & fsk4. It is possible to have a mix of sources using FSK4 and QPSK demodulation."

Is this the modulation for the trunk control signal only?  or is it the modulation used on digital voice as well? (I only have analog voice to play with)

I appear to be able to set this value to "fsk4", "qpsk" or "BLAH" and still successfully decode the trunk control signal. (Admittedly I'm still trying to figure out how my smartnet fits together, which is why I'm trying to get a better understanding of all of the things, but it certainly appears to be decoding happily.) Why does it seem that this value doesn't matter?

There's also some ambiguity around the case sensitivity as the documentation switches between upper and lower case (eg "fsk4" vs "FSK4"). I'm happy to submit a PR fixing that if someone can confirm which is correct.

robotastic, 2017-01-25:
That is interesting. The modulation option also applies for the trunking control signal… well for P25. For SmartNet it is always fsk4. I am not that best at RF stuff. I think qpsk and fsk4 are similar enough that they might be able to decode each other.

Good call - it should always be lower case. I just fixed the readme. If you catch it somewhere else, please put in a pull request.

- Luke

robotastic, 2017-01-27:
I have fixed the docs so it should be more clear now.

closed 2017-01-27 by robotastic