TrunkRecorder/trunk-recorder #98: SmartNet transmissions cut off at the beginning

patrickfish, 2017-02-19:
I pulled the latest master branch tonight and restarted trunk-recorder and noticed almost all of my recordings were cut off by 2-5 sec. I reverted back to a975e01d662152e379a0e142624566053a6288dd (commit prior to https://github.com/robotastic/trunk-recorder/commit/d6613b940415c767bb58d4a18f118396c180be6d) and everything was back to normal. Has anyone else experienced this on SmartNet?

robotastic, 2017-02-19:
> I pulled the latest master branch tonight and restarted trunk-recorder and noticed almost all of my recordings were cut off by 2-5 sec. I reverted back to a975e01 <https://github.com/robotastic/trunk-recorder/commit/a975e01d662152e379a0e142624566053a6288dd> (commit prior to d6613b9 <https://github.com/robotastic/trunk-recorder/commit/d6613b940415c767bb58d4a18f118396c180be6d>) and everything was back to normal. Has anyone else experienced this on SmartNet?

Hmm… is that just for SmartNet or P25 too?
It was probably that change to P25 Recorder, with the low-pass filter change. I will try out some different settings.

robotastic, 2017-02-19:
What sampling rate are you using from the source? Is it RTL-SDR dongles, or something beefier, like a HackRF? The number of filter taps is related to delay, and with a higher sampling rate, I think there would be more taps. Can you look for the "Number of LPF taps: “ print out when you first start up?

patrickfish, 2017-02-19:
Using a HackRF with a sampling rate of 8000000.0
Number of LPF taps: 24243

I don't have any P25 systems to test with, only SmartNet with P25 voice.

robotastic, 2017-02-19:
Ah! Ok - I will test with my HackRF. I think it is from the low pass filter and the higher sample rate. My filters are too tight which adds delay.

robotastic, 2017-02-21:
Can you give the lpf_test branch a try? It seemed to work well with my hackrf

patrickfish, 2017-02-21:
Seems like something is off with the channel calculations...here's the full run output.

```
$ linux; GNU C++ version 5.3.1 20151219; Boost_105800; UHD_003.009.002-0-unknown

[2017-02-20 23:11:45.616100] [0x00007fe7cc39a900] [info]    System Type: smartnet
[2017-02-20 23:11:45.616142] [0x00007fe7cc39a900] [info]    Control Channels:
[2017-02-20 23:11:45.616190] [0x00007fe7cc39a900] [info]    8.53638e+08
[2017-02-20 23:11:45.616204] [0x00007fe7cc39a900] [info]
[2017-02-20 23:11:45.616212] [0x00007fe7cc39a900] [info]    API Key: 75559ce06bee6b7eb727e2df776463bb996b6472a3755cf0eb0e4192c06656c2
[2017-02-20 23:11:45.616222] [0x00007fe7cc39a900] [info]    Short Name: Davis
[2017-02-20 23:11:45.616229] [0x00007fe7cc39a900] [info]    Upload Script:
[2017-02-20 23:11:45.616238] [0x00007fe7cc39a900] [info]    Talkgroups File: bc3b.csv
[2017-02-20 23:11:45.616262] [0x00007fe7cc39a900] [info]    Capture Directory: /home/pfish/trunk-recorder
[2017-02-20 23:11:45.616271] [0x00007fe7cc39a900] [info]    Config Directory: /home/pfish/trunk-recorder
[2017-02-20 23:11:45.616279] [0x00007fe7cc39a900] [info]    Upload Server: https://api.openmhz.com
[2017-02-20 23:11:45.616286] [0x00007fe7cc39a900] [info]    Default Mode: digital
[2017-02-20 23:11:45.616292] [0x00007fe7cc39a900] [info]    Call Timeout (seconds): 3
[2017-02-20 23:11:45.616333] [0x00007fe7cc39a900] [info]    Center: 8.545e+08
[2017-02-20 23:11:45.616348] [0x00007fe7cc39a900] [info]    Rate: 8e+06
[2017-02-20 23:11:45.616359] [0x00007fe7cc39a900] [info]    Error: 0
[2017-02-20 23:11:45.616372] [0x00007fe7cc39a900] [info]    PPM Error: -5
[2017-02-20 23:11:45.616384] [0x00007fe7cc39a900] [info]    Gain: 0
[2017-02-20 23:11:45.616397] [0x00007fe7cc39a900] [info]    IF Gain: 24
[2017-02-20 23:11:45.616409] [0x00007fe7cc39a900] [info]    BB Gain: 24
[2017-02-20 23:11:45.616418] [0x00007fe7cc39a900] [info]    LNA Gain: 0
[2017-02-20 23:11:45.616425] [0x00007fe7cc39a900] [info]    MIX Gain: 0
[2017-02-20 23:11:45.616436] [0x00007fe7cc39a900] [info]    Squelch: -60
[2017-02-20 23:11:45.616448] [0x00007fe7cc39a900] [info]    Idle Silence: false
[2017-02-20 23:11:45.616459] [0x00007fe7cc39a900] [info]    Digital Recorders: 2
[2017-02-20 23:11:45.616467] [0x00007fe7cc39a900] [info]    Debug Recorders: 0
[2017-02-20 23:11:45.616476] [0x00007fe7cc39a900] [info]    Analog Recorders: 4
[2017-02-20 23:11:45.616485] [0x00007fe7cc39a900] [info]    Driver: osmosdr
[2017-02-20 23:11:45.616496] [0x00007fe7cc39a900] [info]    Modulation: fsk4
[2017-02-20 23:11:45.616505] [0x00007fe7cc39a900] [info]    Source Device not specified
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.9
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy redpitaya
Number of USB devices: 8
USB device 1d50:6089: 0000000000000000457863c8223e261f match
Using HackRF One with firmware 2015.07.2
[2017-02-20 23:11:46.318425] [0x00007fe7cc39a900] [info]    SOURCE TYPE OSMOSDR (osmosdr)
[2017-02-20 23:11:46.318447] [0x00007fe7cc39a900] [info]    Setting sample rate to: 8e+06
[2017-02-20 23:11:46.319648] [0x00007fe7cc39a900] [info]    Actual sample rate: 8e+06
[2017-02-20 23:11:46.319686] [0x00007fe7cc39a900] [info]    Tunning to 8.545e+08hz
[2017-02-20 23:11:46.320172] [0x00007fe7cc39a900] [info]    Gain Stage: RF supported values: 0 14
[2017-02-20 23:11:46.320230] [0x00007fe7cc39a900] [info]    Gain Stage: IF supported values: 0 8 16 24 32 40
[2017-02-20 23:11:46.320340] [0x00007fe7cc39a900] [info]    Gain Stage: BB supported values: 0 2 4 6 8 10 12 14 16 18 20 22 24 26 28 30 32 34 36 38 40 42 44 46 48 50 52 54 56 58 60 62
[2017-02-20 23:11:46.320365] [0x00007fe7cc39a900] [info]    Max HZ: 8.585e+08
[2017-02-20 23:11:46.320382] [0x00007fe7cc39a900] [info]    Min HZ: 8.505e+08
[2017-02-20 23:11:46.320645] [0x00007fe7cc39a900] [info]    IF Gain set to: 24
[2017-02-20 23:11:46.320885] [0x00007fe7cc39a900] [info]    BB Gain set to: 24
[2017-02-20 23:11:46.320919] [0x00007fe7cc39a900] [info]    Gain set to: 0
Using Volk machine: sse4_2_64_orc
O[2017-02-20 23:11:46.584967] [0x00007fe7cc39a900] [info]    Resampled Rate: 96385.5 Decimation: 83 System Rate: 48000 ARB Rate: 0.498
[2017-02-20 23:11:46.585196] [0x00007fe7cc39a900] [info]    P25 Recorder Taps - initial: 873 channel: 175 ARB: 1461 Total: 2509
Project 25 IMBE Encoder/Decoder Fixed-Point implementation
Developed by Pavel Yazev E-mail: pyazev@gmail.com
Version 1.0 (c) Copyright 2009
This program comes with ABSOLUTELY NO WARRANTY.
This is free software, and you are welcome to redistribute it
under certain conditions; see the file ``LICENSE'' for details.
[2017-02-20 23:11:46.595622] [0x00007fe7cc39a900] [info]    Resampled Rate: 96385.5 Decimation: 83 System Rate: 48000 ARB Rate: 0.498
[2017-02-20 23:11:46.595850] [0x00007fe7cc39a900] [info]    P25 Recorder Taps - initial: 873 channel: 175 ARB: 1461 Total: 2509
O[2017-02-20 23:11:46.605571] [0x00007fe7cc39a900] [info]    P25 Recorder Taps - initial: 873 channel: 131 ARB: 1461 Total: 2465
O[2017-02-20 23:11:46.613388] [0x00007fe7cc39a900] [info]    P25 Recorder Taps - initial: 873 channel: 131 ARB: 1461 Total: 2465
[2017-02-20 23:11:46.620002] [0x00007fe7cc39a900] [info]    P25 Recorder Taps - initial: 873 channel: 131 ARB: 1461 Total: 2465
[2017-02-20 23:11:46.626441] [0x00007fe7cc39a900] [info]    P25 Recorder Taps - initial: 873 channel: 131 ARB: 1461 Total: 2465
[2017-02-20 23:11:46.627226] [0x00007fe7cc39a900] [info]    Loading Talkgroups...

[... cut ...]
[2017-02-20 23:11:46.627427] [0x00007fe7cc39a900] [info]    Control Channel: 8.53638e+08
[2017-02-20 23:11:46.627462] [0x00007fe7cc39a900] [info]    SmartNet Trunking - SysId: 0
[2017-02-20 23:11:46.627475] [0x00007fe7cc39a900] [info]    Control channel: 8.53638e+08
O[2017-02-20 23:11:46.633378] [0x00007fe7cc39a900] [info]    P25 Recorder Taps - initial: 873 channel: 175 ARB: 1947 Total: 2995
O[2017-02-20 23:14:15.994188] [0x00007fe7cc39a900] [info]    	Recording not started because there was no source covering: 4.92638e+08 For TG: 44240
```

robotastic, 2017-02-21:
Sorry about that - I had left the code I use for my local smartnet sys in. I just fixed it, the channels should be correct now.

patrickfish, 2017-02-22:
So far so good...will keep this branch running for a bit and let you know if I have any issues.

patrickfish, 2017-02-24:
Yup, looks like we're back to normal. Thanks for checking into this! I see it's been merged into master now.

robotastic, 2017-02-24:
Wahoo!!!

closed 2017-02-24 by robotastic