TrunkRecorder/trunk-recorder #100: Add Support for Digital Coded Squelch

kkotowick, 2017-03-02:
I'm using this software to record some conventional channels. The radio network uses digital coded squelch (DCS) so that receivers don't hear a bunch of non-verbal traffic that is transmitted over the channel (morse code for the station ID, etc.). Unfortunately, using your software, it captures all this non-verbal transmission as well.

What are the chances of getting a feature added that allows us to put a receive squelch code in the config file (usually 3-digit octal), so that it only records audio when the DCS is present?

Thanks!

robotastic, 2017-03-02:
I looked into this a little while ago and I think it might be possible. I found a project that already implemented most of the code:
https://bitbucket.org/holiman/gnuradio-dcs
I will add this on to the list of features to add.

kkotowick, 2017-03-03:
Awesome, thanks! One thing to keep in mind: it will have to be able to detect the presence of the DCS even if it appears in the middle of an ongoing transmission (i.e. can't just check once when a new transmission starts, then ignore the rest of the transmission if it doesn't have it).

Dygear, 2017-10-15:
This along with tone squelch is the would make this an almost competle system for me. Recording the DCS / TSQ values would be very handy. I actually have a frequency that is used by two different departments that are active in the same area, but the 2nd department uses a different tone. So it would be cool if we could setup the system like this ...

```
"systems": [{
  "type":     "conventional",
  "channels": [151070000: [{"DCS": 071}], 154115000: [{"TSQ": 100.0], 154570000: [{"TSQ": 151.4}]]
}, {
  "type":     "conventional",
  "channels": [460175000: [{"DCS": [315, 243]}], 461175000: [{"DCS": 516}], 467725000: [{"DCS": 043}]]
}]
```

And the tones that were detected and present within the waveform, it would be perfect if this information also exposed to the outlets (JSON, API, WebSocket).

kkotowick, 2018-03-13:
@robotastic You had mentioned you had looked into this a while back. Any luck with implementing it? It would be an amazing enhancement for conventional systems, so it wouldn't capture all of the unwanted traffic.

imagesafari, 2018-06-16:
+1 for tone squelch

imagesafari, 2019-03-29:
Bump for DCS / CTCSS squelch! :)

Cole0209, 2019-03-29:
Agreed this would be great

EricTendian, 2020-04-29:
While not DCS, it looks like GNURadio does have built in support for CTCSS squelch, so I gave a shot at adding it: https://github.com/robotastic/trunk-recorder/compare/master...CrimeIsDown:ctcss-squelch

So far it's been working pretty well, after spending a while trying to get the correct level value needed by the block. It's configured via the `talkgroupsFile` for the system, like so:

```csv
1,151.4,A,ChiPD Zone10,Police Dispatch Zone 10,Law Dispatch,Police,1
2,110.9,A,ChiPD Zone 3,Police Dispatch Zone 3,Law Dispatch,Police,1
```

The PL tones are put into the second column of the CSV, then the parser detects the analog mode and sets the tone value. I'm not sure if using the CSV file is the best place to put it, I was thinking about the syntax mentioned above but changing the `channels` value from a simple array to key-value pairs might be backwards-incompatible.

So, a question to all: what would be your ideal way to specify the CTCSS tone for a frequency?

robotastic, 2020-05-01:
Interesting! The Talkgroups CSV could use a good redefinition - there are some fields there that are not being used, so not the end of the world if it needs to change. I can also update OpenMHz and finally make them align so it is the same file for both.

K2IE, 2021-02-11:
Any news/progress on this?  CTCSS & DCS decoding and squelch would be quite useful for the conventional systems that I monitor.

jliedtke6014, 2021-02-12:
Same here. Trying to make Trunk and Rdio-Scanner my goto for everything.  CTCSS is my last piece of the puzzle.

photounit, 2021-02-18:
Same with me.... TCS would be a huge improvement...

imagesafari, 2021-05-16:
Bumping again, ctcss/dcs would be hugely useful for me

leee, 2021-05-16:
A PL/CTCSS squelch/gate already exists in gnuradio, and there's at least one DPL/DCS module (I think already linked here) out of tree.

But as was already covered in this thread, providing PL/DPL support is a lot more than just adding those blocks and conditionally connecting them within `analog_recorder`. There's the matter of config formatting and parsing:
- perhaps as part of the frequency array,
- or perhaps as part `talkgroupsFile`s as specified earlier,
- perhaps with frequency information too, so instead of keying "talkgroups" by index, they're keyed by the file.

Additionally, there's the matter of how to introduce this change as a non-breaking feature, or to bump config format version and treat it as breaking, etc.

Therefore, this issue is effectively blocked (or must be worked in lockstep) on just that: a robust discussion on the future of config handling; as well as the possibility of tooling this in with the new proposed plugins system from last year. Along with the work for adding support for this in, of course.

The last four comments in this issue have been bumps only, with at least another four in the rest of the issue also bumping for this. I don't think there's lack of awareness or a misunderstanding that having PL/DPL support would be a wonderful enhancement to have. Please be mindful of how bumping adds noise to issue threads.

smyers119, 2022-02-17:
Is there a way I could start a bounty for this future?

aaknitt, 2022-02-18:
Would post-processing the WAV files to detect CTCSS and DCS content and add it to the JSON data be acceptable, or is gating what's getting recorded in real time needed?

ZeroChaos-, 2022-02-18:
If the wav is "post processed" at the concatenation stage then both could pretty easily be done.  Config options would be default all (like now) or specified code(s) only (and discard the rest during concatenation)

imagesafari, 2022-02-18:
For me, actual CTCSS and/or DCS squelch on the recorder itself would be
key, so that it can run without a squelch threshold.

robotastic, 2022-02-18:
Nice - yea, it would be great if someone is able to pull something together. I would definitely welcome a solid PR here. I would need help testing it out because I don't have any systems near me that use it.

aaknitt, 2022-02-18:
For those that missed it, there is a fork with CTCSS working in the comments above.  I think moving the config for the CTCSS from the talkgroups.csv file to the config.json file as @Dygear suggests probably makes sense.  So CTCSS is doable, just needs some work to tidy it up and get a PR in.  DCS may be tougher.  The bitbucket repository linked above for DCS in gnuradio is no longer active.  I've emailed the author to see what the status is.  I did a DCS decoder a long time ago (in LabView of all things?!) but I don't remember any of it, I'd have to do a lot of memory refreshing.  And I'm not yet competent in gnuradio.  Its feature set seems to largely consist of raising my blood pressure at the moment, but I'm trying :-)

aaknitt, 2022-02-19:
DSC block for gnuradio is now here:

https://github.com/holiman/gr-dcsinfo

I believe this just reports the detected code, so more work would need to be done to use it to gate audio.

robotastic, 2022-02-19:
Question - Could we add the tone info as a field in the "talkgroup" file? I was thinking of creating a custom format that would just be for conventional channels. That would keep the info out of the config file.

ZeroChaos-, 2022-02-19:
I would still prefer an option for detecting the tone and not gating on it.  Another common scanner feature is to gate on any tone but not none.

cherizon, 2022-02-19:
There are a few issues in the area I am in where it is 90% conventional analog (with some p25 sprinkled in).  In a lot of cases we are stuck with around 240 T-Band channels which are closely split between cities and towns in a region.  It's not odd to be in areas where these overlap - making the need for tone detect like this to work properly while being recorded.  For example - I might want agency A on frequency A.  Agency B shares Frequency A.   Agency B starts talking - however - let's say they are heard at -100dbm.  Agency A is at -75dbm shows up mid transmission - I'm more concerned about the voice traffic for Agency A than I am for B.  If the detection / gate isn't done right and at the time of recording (I'm thinking of how a radio uses COR w/PL/DPL) it had mixed audio - then the feature is not going to work right.   Hopefully I explained this right - as a consumer of the application I would only care about Agency A in this case - and would not care about B (and more so because the signal might not be as good where I am).

taclane, 2022-02-19:
Couldn't we just start a new call if the recorder detects a change in PL/DPL?   

Over in https://github.com/robotastic/trunk-recorder/pull/630, I'm trying to imagine a future where conventional channels can be organized into "virtual talkgroups" for the purpose of meta-data and handling.  Instead of strictly gating the channel, I'm kinda partial to letting the recorder record everything, and then using some analogous method to the trunked side to figure out if the call should be ignored.

Like in the example above, a theoretical `conventional.csv` like this could tell TR to only record the call if it catches a certain tone.
```
TG,  FREQ,      TONE,     ALPHA,         Description,            Tag,    Group,  Record? 
300, 462275000, 94.8 PL,  Town A Police, Town A Police Dispatch, Police, Town A, True
325, 462275000, 151.4 PL, Town B DPW,    Town B Trash Dispatch,  DPW,    Town B, False
350, 462275000, CSQ,      Town A,        Town A No Tones,        -,      Town A, False
```

Dygear, 2022-02-20:
Taclane is probably the most correct answer here. Flexibility being the goal where as I might want to ignore getting audio without a PL tone attached to it, it's entirely valid that other people would want that.

robotastic, 2022-02-20:
@taclane - hmm... I really like this approach, but I will have to see if something like this could work. The way things are currently setup is quite a bit different. There is a basic hierarchy of System -> Call -> Recorder -> Wav sink. Each Call is dedicated for specific Talkgroup. This works fine now a Conventional system where the Call / Recorder stay on a single freq, and the TG is just that "Channel".

For approaches, it might be possible to have the recorder "reset" the call it is under.  This would change the Talkgroup for the call and also start a new .wav file.

Question on Tones - are they unique for each frequency, or would you want a Tone.csv file? Would the 94.8 PL tone only be Town A Police on 462275000 or would it also be the same police on 463000000?

aaknitt, 2022-02-20:
@robotastic is there anything in the current architecture that would break if two or more conventional Calls/Recorders are configured for the same frequency?  If that works, then someone that wanted to record multiple tones on the same frequency could set up a different Call/Recorder/TG for each.  So each row in @taclane's CSV would spool up a separate recorder chain.  Not the most efficient, but may work without a big tearup of the existing architecture.  Each chain only cares about its tone, so no resetting or talkgroup changing required.  Basically replace the existing squelch block with the CTCSS squelch block if a tone is specified.  DCS would require more work, but CTCSS should be an easy first step (there's already a fork that does it).  

I'm not sure I fully understand your question, but it's common to use the same tone across multiple frequencies.  In some places they use the same PL tone on all public safety frequencies in the county, for example.  

I'm seeing a few different use cases in these comments:

1. Use CTCSS/DCS to avoid having to set traditional squelch parameter.  This requires gating prior to the WAV file sink, it can't be done in post-processing.  Users wanting to use it in this manner will know the tone frequency ahead of time.  This use case is well-supported by the CTCSS block that's already in gnuradio.  The out-of-tree DCS block linked to above doesn't do this but it could be used to do so with some additional blocks around it.  
2. Tone search.  Record everything on a frequency but tag the JSON with detected tones.  The gnuradio CTCSS block does not support this, it requires that the tone be known ahead of time.  It's a squelch block, not a search block.  The out-of-tree DCS block may support this, but I'm not sure.
3. Gate on any tone.  This is a combination of 1 and 2.  If any tone is detected, record it and tag it with the tone. 
4. Gate on no tone.  This is the opposite of 3.  Only record of no tone is detected.

I think focusing on scenario 1 makes sense as a first step as I think it's the one most people are looking for.  2 could be done entirely in post-processing with an uploadScript.  3 and 4 get a lot more complicated under the hood with the gnuradio block arrangement.  Good to keep them in mind when deciding how to do the config/csv stuff but may make sense to get the simple stuff that will cover most users implemented first.

leee, 2022-02-21:
I haven't been active lately (wow, last comment before the recent activity) although I did offer the following thoughts elsewhere, but just wanted to jump in here:

I also agree that 1 (which is effectively behaving like any other receiver) is the right move - at its core `t-r` is a recording platform. Even though it can be applied as a discovery/sigint tool, that's not the focus.

However, to support post-processing on squelch-gated recordings, the 300 Hz HPF could be within an if and disabled as desired on a per-recorder/channel basis (e.g. a tone option such "DIS" in place of "CSQ" or something to that effect). Users interested in discovery can perform detection on that (and filter out the low end in post if they'd like).

It also might be possible to save some cycles by tossing out the power squelch on recorders with tone enabled squelch. Saves some users without RF competency or a compromised RX chain the trouble of figuring out gains and squelches.

---

I don't think expanding the scope of `config.json` is a good direction. It's been growing more complex especially for conventional configs like some sort of config hydra, and there's already the potential growing pains of:

- mixed-mode channels
- channel numbering that doesn't follow straight incrementing

on top of multi-tone conventional channels as discussed here, like community repeaters (ugh), multi-site repeaters, alternate tone backup systems, co-channel systems where receivers are right between multiple licensed system contours (the best receivers are up in the air, but not too tall! ) etc.

CSVs already exist, I absolutely believe their format isn't perfect, but it'll be better than stuffing everything into json. So might as well start moving conventional system configs over and introduce the breaking change as a new major semver (hey, we survived moving versions 2, 3, and 4, so we all have some practice now).

ZeroChaos-, 2022-02-21:
I agree the csv is a good idea, but does that maintain consistency?  Like, where is NAC set for P25?  I am all for doing what makes the most sense, but keeping it consistent really helps it make sense.

robotastic, 2022-02-22:
That is a great point @aaknitt - You could have a multiple conventional recorders on a single frequency - one for each of the distinct Tones. 

Mapping the tones to the channel freq could be a little work since the config.json could have 3 channels, but the channel.csv might only have 2 tones specified. You could just exit things I suppose... but it can make for a bad experience when you have the config for a single "object" split across 2 different files. Adding a new tone, would mean making sure you get it right in both places. It is probably best to have it all defined in one file... another array of objects in the config.json makes me shudder a little.  Having all of the conventional channel/recorder definition be in a .csv file does seem cleaner. There could be backwards support for the current approach. @ZeroChaos- you make a good point, though that this would break consistency with the current approach.  I would be open to other approaches, if there is a better way to go about this.

How

aaknitt, 2022-02-22:
Yeah I agree that trying to bounce back & forth to sync up the JSON and CSV files is ugly.  Just pick one and go with it.  As much as I don't like having to manually edit JSON, this does seem like "config" type data...it's used to configure the signal chain, not just provide metadata like talkgroups.csv.  So it seems like that's the more "correct" place to put it from a SW standpoint based on the current architecture. However, the user experience for JSON config files sucks so...I'm not sure what the right answer is, that's why you're in charge :-)

robotastic, 2022-02-26:
OK - here is my current plan. Continue to support the simple approach for describing conventional channels in the config.json. Add support for a channels.csv if you want to have a more complex setup. 

TG,  FREQ,      TONE,     ALPHA,         Description,            Tag,    Group, Mode
300, 462275000, 94.8 PL,  Town A Police, Town A Police Dispatch, Police, Town A, A
325, 462275000, 151.4 PL, Town B DPW,    Town B Trash Dispatch,  DPW,    Town B, A
350, 462275000, CSQ,      Town A,        Town A No Tones,        -,      Town A, A

At startup, TR would go through the channel.csv and create a Call & Recorder for entry in the table. I would create a new Analog Recorder that has the CTCSS block instead of the Squelch block. There would be no support for Tones on a P25 Conventional system. If the Tone cell is left empty, traditional squelch will be used instead. 

Would this be useful? Are there things I should do differently? I know this probably won't cover all of the weird corner cases, but trying to come up with a maintainable 80% solution.

robotastic, 2022-03-05:
OK - things are moving along. I have the `conventional` branch pretty much complete for reading in a .csv file for describing conventional channels. I still have to do a bunch of testing. There is no support for tone based squelch... but step one is to read in what tone to watch for.

smyers119, 2022-03-08:
Everything posted so far, sounds great.  I like where your taking this.

kb2ear, 2022-05-28:
poke

ZeroChaos-, 2023-03-30:
Not sure if it helps, but sdrangel supports this and the code may give us hints or clues.  I'm not suggesting stealing it as it, but maybe if there is a lib or something we can use https://github.com/f4exb/sdrangel/tree/master/plugins/channelrx/demodnfm

slowbalt, 2024-10-11:
bump for CTCSS

ZeroChaos-, 2024-10-11:
Ctcss works in 5.x

robotastic, 2024-12-15:
Keeping this alive for DCS... really need to make it happen

iphands, 2025-02-14:
> Ctcss works in 5.x

Are the docs just out of date then?
https://trunkrecorder.com/docs/CONFIGURE#channelfile

Anyone have any working examples?

ZeroChaos-, 2025-02-14:
The docs in git are the only ones kept up to date https://github.com/robotastic/trunk-recorder/blob/master/docs/CONFIGURE.md#channelfile

iphands, 2025-02-15:
CTCSS works so well! THIS PROJECT IS AMAZING!!! :D