TrunkRecorder/trunk-recorder #99: Add an example config for conventional non-trunked system

kkotowick, 2017-03-02:
There are a bunch of good sample config files in the /examples directory, but they're all for trunked systems. Could you please add an example config for a conventional (non-trunked) system, including an example ChanList.csv for a conventional system?

The specific issue I'm confused about is how "virtual talkgroup numbers" are assigned to conventional channels, and how those are linked to a ChanList.csv file. Do priorities matter for conventional channels?

robotastic, 2017-03-02:
Great idea - I can work on this on the weekend. For ChanList, the first channel freq you list in the System section has a talkgroup number of 1, the next 2 and so on. or atleast that was how it was supposed to work.

Priorities don't have an effect because there is a recorder dedicated to each channel.

robotastic, 2017-03-10:
Added in an example for conventional channels

closed 2017-03-10 by robotastic