TrunkRecorder/trunk-recorder #102: Add tracking of times for sources

kkotowick, 2017-03-04:
In the json file for recorded files on trunked systems, there's an array of sources for transmissions in the conversation. It has the radio IDs for each transmission in that conversation. Would it be possible to add an array for the times associated with each source? For example, say that source 2435 started at 5400ms into the recording?

Treehouseman, 2017-03-04:
That's actually already tracked, I guess it's just not getting recorded into the json file. It's used when uploading to openmhz.com, you can see the source times there.

kkotowick, 2017-03-04:
Yeah, I figured it must be. Right now the json file looks something like this:

```
{
    "freq": 8.53788e+08,
    "start_time": 1488668468,
    "stop_time": 1488668482,
    "emergency": 0,
    "talkgroup": 1223,
    "srcList": [ 22315, 22344, 22315, 22344 ],
    "freqList": [ { "freq": 8.53788e+08, "time": 1488668468, "pos": 0} ]
}
```

Instead, maybe you can turn each element in the "srcList" into it's own json object with source and time, like this:

```
{
    "freq": 8.53788e+08,
    "start_time": 1488668468,
    "stop_time": 1488668482,
    "emergency": 0,
    "talkgroup": 1223,
    "srcList": [ {"src": 22315, "time":1488668468}, {"src": 22344 , "time":1488670100}],
    "freqList": [ { "freq": 8.53788e+08, "time": 1488668468, "pos": 0} ]
}
```

Treehouseman, 2017-03-05:
Look in trunk-recorder/uploaders/call_uploader.cc, the json formatting for the detailed src_list should be in there, for now you could just copy it into call.cc.

kkotowick, 2017-03-05:
Yeah, that's pretty much what I ended up doing. I changed the code in call.cc from this:

```
for (int i = 0; i < src_count; i++) {
      if (i != 0) {
        myfile << ", " <<  src_list[i].source;
      } else {
        myfile << src_list[i].source;
      }
}
```

to this:

```
for (int i = 0; i < src_count; i++) {
      if (i != 0) {
        myfile << ", ";
      }
      myfile << "{\"src\": " << src_list[i].source << ", \"time\": " << src_list[i].time << "}";
}
```

Would be great if that change could be added to the repo though.

robotastic, 2017-03-05:
I can definitely add something like this in. I will slip it into the next release.

kkotowick, 2017-03-05:
Awesome, thanks. While you're at it, what are the chances of changing it to log times in milliseconds, instead of seconds? Unfortunately, seconds are too large of a unit to accurately split the audio files into individual transmissions.

robotastic, 2017-03-10:
Added to the latest release. The position within the file where each source starts is in milliseconds. It looks like things get complicated when trying to get milliseconds for time of day.

closed 2017-03-10 by robotastic