TrunkRecorder/trunk-recorder #103: Conventional Channel Overlap

kkotowick, 2017-03-06:
I have a conventional digital system in my config, which has a single channel that it records. The config looks something like this:

```
"sources": [
                {
                        "center": 453300000,
                        "rate": 1000000,
                        "squelch": -40,
                        "error": 700,
                        "gain": 28,
                        "modulation": "QPSK",
                        "analogRecorders": 1,
                        "analogLevels": 8,
                        "driver": "osmosdr",
                        "device": "rtl=0"
                },
        ],
        "systems": [
                {
                        "shortName": "Conventional",
                        "channels": [453100000],
                        "type": "conventional",
                        "talkgroupsFile": "../Conventional.csv"
                }
        ],
        "captureDir": "../recordings"
}
```

The channel in the config is a 12.5kHz narrow-band channel. There are other 12.5kHz narrow-band channels nearby on each side of it. The problem is that with this config, trunk-recorder is recording not only the channel I've entered into the config, but other "nearby" channels as well (and is thinking that they are part of the one I put in the config). Does the software record in wide-band or something? Is there any way to make it restricted to just the 12.5kHz I'm interested in?

robotastic, 2017-03-07:
Weird! I wonder if it could be because the channel filters are not small enough. You could try making the filter values on this line smaller. Maybe 4000 and 1000?
https://github.com/robotastic/trunk-recorder/blob/master/trunk-recorder/recorders/analog_recorder.cc#L39 <https://github.com/robotastic/trunk-recorder/blob/master/trunk-recorder/recorders/analog_recorder.cc#L39>

I wonder if the amount of error is also off slightly? maybe it is straddling the two channels?

Let me know if that fixes it. I could be filter wrong.

kkotowick, 2017-03-07:
I tried tightening the filter to 4000 and 1000, no luck there. I also triple-checked the error, no problems there. I've tried tightening the filter more, to 2000 and 1000, I'll keep messing with it and see if I can find values that work.

There are some other issues with conventional recording, that may or may not be related. When an audio file is recorded, I notice that the .json file for the recording almost never has anything in the "freqs" field. Occasionally it will though. So it seems that there is a bug with sometimes putting a value in that field, and sometimes not.

Also, it is kind of strange but when a conventional recording is completed and a .json file is made for it, the "startTime" field is way earlier than when the recording actually starts. It looks like the software immediately creates a new file/recording as soon as the old one is done, even if the squelch isn't open, and marks that as the "startTime". So if there's 100 seconds of silence on the channel and then 10 seconds of transmission, the "startTime" field would be something like 1488913750 and the "endTime" field would be 1488913860. The difference between them SHOULD only be 10 seconds because there was only 10 seconds of transmission, but it's actually 110 seconds because it's including the 100 seconds of silence since the previous transmission ended.

Are you open to accepting pull requests? I have a bunch of small fixes/changes I've been making locally that you might appreciate, so I could fork this repo, make changes there, then make pull requests if you like.

kkotowick, 2017-03-07:
OK, so I don't think the `channel_lpf_taps` is the issue. When I reduced the filter size down to 500 and 250, the audio was all distorted (still able to make out the words though), but it was still capturing both channels. I'm thinking the issue has to do with what's going into the filter, not the filter itself.

robotastic, 2017-03-10:
Pull requests are always welcome!
Right now, it is probably best to go off of the Develop branch. There are a lot of changes there that are not in Master.
I am totally confused what it could be. Are you able to take a picture of it in GQRX or some other waterfall, just to get a sense of how tight it is? Actually, can you try tuning in the channels using GQRX? It uses gnuradio too, and has a similar filter structure. Are you able to record just one channel using it?

For the start times, the start of the recording is determined by the squelch, and there isn't a great way of pulling that info. I suppose it could be possible to poll it every second to look for a change and then mark that down as the start. Or, alternatively, subtracted the lenght of the recording from the stop time.

EricTendian, 2019-10-07:
(I know this is an old issue, but...)

I was running into this same problem where the call duration based on `stop_time - start_time` did not match the actual audio file duration. This led to wrong call lengths being shown in OpenMHz. I think I solved it by implementing the suggestion in the last sentence of the above comment, durations are more accurate now. Here's the fix: https://github.com/CrimeIsDown/trunk-recorder/commit/626dbcd105207747dcd584d15707c1cb9abbfddb

I'm not sure what side effects this may have, but I can make a PR if this fix is wanted.

leee, 2021-05-15:
I can't speak to co-channel or adjacent-channel bleed coming over BAMA, but I hope previous work on channel deviation filtering and RF/audio processing (and hopefully future work as well, including on PL/DPL squelch) will make this better.

This issue also brought up issues with:
- conventional recording `freqList` being frequently empty
- conventional recording time duration/length and `start_time` being off

I can't speak to the latter two issues, but I assume the last one was fixed in the batch of latest work - @robotastic ?

Perhaps the freqList issue and continued work on channel filtering should be opened up as separate new issues, and this one closed out.

robotastic, 2021-05-16:
Agreed - freq_list is being handled in #471

closed 2021-05-16 by robotastic