TrunkRecorder/trunk-recorder #104: Error: Call: more than 50 freq

UnderwaterHockey, 2017-03-11:
I just updated, and I'm still getting a lot of these errors.  
[image]

UnderwaterHockey, 2017-03-12:
new screenshot
[image]

UnderwaterHockey, 2017-03-12:
New error
[image]

UnderwaterHockey, 2017-03-13:
Config

geeojr, 2017-03-20:
I have this same issue and a potential cause.

In my case it appears to be caused when I'm monitoring multiple linked trunked systems that are carrying traffic for the same talkgroup at the same time. It seems to ping-pong between both towers/frequencies and leaves quite a bit of noise in the recording. If I remove one or more of the systems from the configuration file the issue goes away since it only sees one instance of the call. I'm happy that the system doesn't record the call twice; but it probably just needs to disregard the second duplicate transmission on the second system in this case. 

Let me know if you need more details or maybe a reference to the system on radioreference, etc.

UnderwaterHockey, 2017-03-20:
Try the development branch.  He made a change there that distinguished the two systems, it seems to be working much better for me.

robotastic, 2017-04-11:
I think this should be fixed. Closing, but re-open if it is still an issue.

closed 2017-04-11 by robotastic