TrunkRecorder/trunk-recorder #106: Logging Update(s)

kcwebby, 2017-03-14:
I think it would be nice to separate out some of the logging features.

I think the control channel traffic monitoring / assignment should record to a configured file:
I think it should have a unified format that starts with date/time, then RTL #, then message

For example
1/1/1900 0000 SYS# RTL#  Channel Grant TG:XXXX FREQ:####, ENCR:01 TDMA:-1
1/1/1900 0001 SYS# RTL5-15 Recording Started TG:XXXX Priority:XXX  (PID)
1/1/1900 0002 SYS# RTL#  Channel Re-Tune TG:XXXX FREQ:####, ENCR:01 TDMA:-1
1/1/1900 0002 SYS# RTL5-15 Recording Passoff to RTL6-3
1/1/1900 0002 SYS# RTL6-3 Recording Handoff from RTL-15 TG:XXXX FREQ:####, ENCR:01 TDMA:-1
1/1/1900 0003 SYS# RTL6-3 Nothing Recorded for 3 sec.. Ending Monitoring Process (PID)

Something Similar with a consistant, parseable format.
Perhaps even something with some kind of a delimiter, so it could be parsed into fields (and irrelevant fields left blank for consistancy)

I would imagine that doing it this way, you could configure each system to have its own log, which would further the seperation and ease diagnostics if something was not working right.

kcwebby, 2017-03-14:
Add into that the capture of the "uploadscript" logging to a seperate file (without configuring in the upload script)

NarkyBark, 2017-03-14:
I second this

Treehouseman, 2017-03-17:
if you're talking about "uploadscript" in terms of "encodeUpload.sh" please note that this is a bash script, if you're getting logging info from it then there's not much you can do, that stuff just goes straight to stdout and stderr, the file logging here is done through boost, which is a middleman between recorder and stdout/stderr, using the uploadserver option and openmhz you have just boost messages. Upload script should only be producing a like saying it's running the script, if that, any further upload related messages from the use of uploadscript can't easily be contained. Your best bet is to edit your script file to eliminate error messages there, because that's outside recorder's reach.

kcwebby, 2017-04-16:
@Treehouseman   I am not sure how the code goes, but is it not possible to wrap the "exec() function of the uploadScript in such a manner that stdout/stderr are captured?   Maybe not.  Most of my scripts output  dignostic data if enabled, but even still, they are in the overall 2>&1 >logfile that I am using when executing recorder.   Your branch even does a much better job of capturing this, however I think you turned off uploadScript.

Also   Add to my original note  some kind of ability to rotate logfiles on a configurable schedule

vabiro, 2017-04-21:
I would like to second this request. 
I have a lot of encrypted traffic and much of the notifications in the console are about those things not being recorded. But I really need to see channels that aren't assigned because there isn't a radio covering the channel. Segregating that information would be very helpful to determine how to set-up the RTL-SDR and how many more I need.
Cheers,

robotastic, 2023-02-19:
Closing this for now because there have been a number of logging improvements. Let start a fresh thread if there are new ideas.

closed 2023-02-19 by robotastic