TrunkRecorder/trunk-recorder #107: Signal Strength

kcwebby, 2017-03-14:
Is it possible to record the signal strength or raw DB Level of incoming signals in the .json.

Perhaps even when a periodic status is recorded, you could record some measurement of the signal strength and current frequency of the control channel as well, by system?

Perhaps this could be further enhanced down the line to vote in multi-site systems to pick the strongest/best signal to record if the same TG presents on multiple sites at the same time.

Dygear, 2017-10-15:
Yes! This was exactly what I was about to ask for! Getting the dBFS level and exposing this to the various outlets would be amazing!

It allows for:
  1. No longer require GQRX for finding signal strength. All we have to do now is check the json file for the dBFS on the signals that breakthrough the squelch but are still actually background nose and set the squelch level to that value.
  2. Muli-site systems, where only the strongest signal is kept and rebroadcast. This does require a central server where the files are uploaded to for web sockets or the api interface for OpenMHz or even a master / slave relationship within other trunk-recorder instances. They could then take dBFS into account, but it's very easy to implement after this values are exposed us.
  3. With 3 or more instances in known location it allows for rough triangulation of units.

robotastic, 2017-10-16:
Unfortunately I haven't found a simple way to measure signal strength of a channel in GnuRadio. If someone comes up with this, I would be happy to add it in.

rosecitytransit, 2017-11-13:
Not the same, but what I've done remove the `if` statement around `BOOST_LOG_TRIVIAL(error) << "[" << sys->get_short_name() << "]\t Control Channel Message Decode Rate: " <<  msgs_decoded_per_second << "/sec, count:  " << sys->message_count;` in main,cc so it always gets logged and then change 867 so it only reports every e.g. 60 seconds.

rosecitytransit, 2017-12-10:
Note that #148 addresses my last comment

ellenpoe, 2019-06-08:
Not quite signal strength, but related. @robotastic would you be willing to merge a PR that listens to channel grant messages in smartnet systems and uses them to measure the delay between the input signal and the output signal via cross-correlation? That might be sufficient for reasonably accurate TDOA-based localization, assuming knowledge of the repeater sites. Who knows, though. I'd like to know if you're open to this before spending too much time on it since I can see it taking a lot of time and frustration to get right.

robotastic, 2019-06-08:
@ellenpoe That would be pretty interesting. If it is possible to get reliable results it could be pretty interesting. It might make most sense to have TR generate a log message and then a separate program do the calculations on distance.

ellenpoe, 2019-06-08:
I agree that letting another program handle the triangulation makes a lot of sense. Do you think it might make sense to extend the recorder interface and implementations to allow monitoring of an input frequency in addition to the repeater's output frequency? Then I can just extend the gnuradio blocks to conditionally calculate and store the delays in a metadata file. The nice thing about that is that it might make it easier to use the TR's uploader code to push all data to a central server for processing and number crunching. Config for all this can be plumbed through from the main config file.

I'm going to hack something together first to experimentally verify the accuracy. A friend agreed to host a recording station, and two receivers is enough to draw a hyperbola on the map. A lot of calls to dispatch in my local PD's system include an address, so I'll just listen until I find some addresses and see how far off from my hyperbola they are. If it's decent I'll probably get another two receivers to narrow it down to a single point and send you the PR.

robotastic, 2019-06-25:
Hi @ellenpoe I wanted to check and see if you had any luck? Adding a new type of System monitor on the Inbound freq makes sense to me. I would need to build a new parser since the format is a little different, but that shouldn't be impossible. I guess I should check and see if I can even get anything on the inbound freq. I am guessing only nearby units would show up. Since that is not on Radio Reference, is the best place to check the FCC filings for the system?

ellenpoe, 2019-06-25:
No luck yet. I think that the way I proposed above won't work, but I'm going to keep exploring. I built up a gnuradio graph for cross-correlation in the frequency domain and learned a lot, so I think the next time I take a crack at this problem I'll have a better idea of how to do it. As for the input frequencies, I'm not sure. I imagine that the FCC filings would have the list of frequencies but I don't know ULS very well so I don't know where to find that stuff. I've been experimenting on IQ data captured from a local 2m ham-band repeater since that's a fixed 600khz offset and it's all on RadioReference. All of my open-source contributions are on hold until my life calms down a bit. I'm really hoping that in about 2 months I'll be able to work on this and some of my other projects some more. 😄

robotastic, 2023-02-19:
Closing for now - I would love a way to measure signal strength if anyone comes up with it

closed 2023-02-19 by robotastic