TrunkRecorder/trunk-recorder #115: Compile error on fresh install with `make` command

bctrainers, 2017-04-10:
Fairly sure i have all the required modules and packages installed already. However, am getting the following error on compile.

```
[ 61%] Building CXX object op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/p25p2_vf.cc.o
[ 63%] Building CXX object op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/imbe_decoder.cc.o
[ 64%] Building CXX object op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/software_imbe_decoder.cc.o
[ 66%] Building C object op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/ambe.c.o
[ 67%] Building C object op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/mbelib.c.o
[ 69%] Linking CXX shared library libgnuradio-op25_repeater.so
[ 69%] Built target gnuradio-op25_repeater
[ 70%] Building CXX object CMakeFiles/recorder.dir/trunk-recorder/main.cc.o
/root/radio/trunk-recorder/trunk-recorder/main.cc: In function ‘int main(int, char**)’:
/root/radio/trunk-recorder/trunk-recorder/main.cc:1069:29: error: expected ‘;’ before ‘__attribute__’
   BOOST_STATIC_ASSERT(true) __attribute__((unused));
                             ^~~~~~~~~~~~~
/root/radio/trunk-recorder/trunk-recorder/main.cc:1069:29: error: expected primary-expression before ‘__attribute__’
CMakeFiles/recorder.dir/build.make:62: recipe for target 'CMakeFiles/recorder.dir/trunk-recorder/main.cc.o' failed
make[2]: *** [CMakeFiles/recorder.dir/trunk-recorder/main.cc.o] Error 1
CMakeFiles/Makefile2:68: recipe for target 'CMakeFiles/recorder.dir/all' failed
make[1]: *** [CMakeFiles/recorder.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
```

This is with the latest code from the git repo, master branch (56673bf).

robotastic, 2017-04-10:
That is weird - What OS are you using? It has been a while since I have do a compile from scratch.

bctrainers, 2017-04-10:
Hi, hopefully this should suffice?

```
root@Scanner:~/radio/trunk-recorder# make --version
GNU Make 4.1
Built for x86_64-pc-linux-gnu

root@Scanner:~/radio/trunk-recorder# cmake --version
cmake version 3.5.2

root@Scanner:~/radio/trunk-recorder# uname -a
Linux Scanner 4.8.0-46-generic #49-Ubuntu SMP Fri Mar 31 13:57:14 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

root@Scanner:~/radio/trunk-recorder#  lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 16.10
Release:        16.10
Codename:       yakkety
```

robotastic, 2017-04-10:
Ah - some people have had trouble with 16.10 - I haven’t had a chance to test it out yet. 16.04 definitely works. I will see if I can figure it out this week.

bctrainers, 2017-04-10:
I've attached a trace+debug log of both cmake and make commands. Hopefully this is of use.

[cmake and make.zip](https://github.com/robotastic/trunk-recorder/files/908726/cmake.and.make.zip)

vabiro, 2017-04-10:
Hi,

I'm having the same issue:

`# make
[ 35%] Built target imbe_vocoder
[ 69%] Built target gnuradio-op25_repeater
[ 70%] Building CXX object CMakeFiles/recorder.dir/trunk-recorder/main.cc.o
/root/radio/trunk-recorder/trunk-recorder/main.cc: In function ‘int main(int, char**)’:
/root/radio/trunk-recorder/trunk-recorder/main.cc:1069:29: error: expected ‘;’ before ‘__attribute__’
   BOOST_STATIC_ASSERT(true) __attribute__((unused));
                             ^~~~~~~~~~~~~
/root/radio/trunk-recorder/trunk-recorder/main.cc:1069:29: error: expected primary-expression before ‘__attribute__’
make[2]: *** [CMakeFiles/recorder.dir/build.make:63: CMakeFiles/recorder.dir/trunk-recorder/main.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:69: CMakeFiles/recorder.dir/all] Error 2
make: *** [Makefile:128: all] Error 2
`
Here is the info on my machine:

`#  make --version
GNU Make 4.2.1
Built for x86_64-pc-linux-gnu
Copyright (C) 1988-2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
# cmake --version
cmake version 3.7.2

CMake suite maintained and supported by Kitware (kitware.com/cmake).
# uname -a
Linux mythtv 4.4.39-1-ARCH #1 SMP PREEMPT Thu Dec 15 23:25:58 UTC 2016 x86_64 GNU/Linux`

robotastic, 2017-04-10:
> html

I will try to get this fixed up. In the interim, try the Docker method. It
should run the 16.04 based container/image.

bctrainers, 2017-04-10:
Hi robotastic, 

As an update to my issue, i went ahead and reinstalled ubuntu, however with the 16.04.2 LTS iso this morning. The compile went without issue and are in the process of getting things setup. 

As it stands, it would appear the code is indeed correct - but something with the compiler on 16.10 is making it throw an error due to unknown reasons.

robotastic, 2017-04-11:
This should be fixed with the change @nkwood made. Let me know if you still have issues. Closing for now, but we can reopen if it still doesn't work.

closed 2017-04-11 by robotastic