TrunkRecorder/trunk-recorder #119: Excluding P25 TG

vabiro, 2017-04-12:
Hi,

I am monitoring a system with more than 200 Talkgroups, and I am only interested in a few, but trunk-recorder is recording all of them. 

I did come across issue [77](https://github.com/robotastic/trunk-recorder/issues/77) which mentioned that changing the priority to 100 in the ChanList.csv should exclude those TG, but this doesn't seem to be working for me. I have even tried a priority of 500 with no luck.

Any suggestions?

Thanks

robotastic, 2017-04-12:
That is weird - When you first start does it saw it read in 200 Talkgroups? I am wondering if it is having some trouble reading in the file.

Also - under the System in config.json, there is a setting called: recordUnknown 
IF you set it to false, it will only record talkgroups listed in the file. with this you can trim the talkgroup file to only the groups you want to record.

UnderwaterHockey, 2017-04-12:
Two things to check in your talkgroup file that gave me trouble when setting up my system.  
First, Excel/Libre can "helpfully" mangle your HEX column.  HEX values that contain an "e" get automatically converted into scientific notation, which confuses things.  When I open the .csv I set the HEX column to "text".
Second, if you got database info from RR, occasionally entries will have quote marks, commas, or other things that can act like a delimeter when trunk-recorder processes it.

vabiro, 2017-04-12:
The recordUnknown did it
Thanks,

kcwebby, 2017-04-12:
I think there is an outstanding issue wherein the Priority tag is not
working properly, even over 100...  Is that still active Luke?

mrtrev, 2017-04-12:
On the Smartnet system I'm monitoring, talk groups are excluded when I set priority to 999.

closed 2017-04-21 by vabiro