TrunkRecorder/trunk-recorder #121: Documentation Suggestion: What can I monitor?

ShakataGaNai, 2017-04-16:
This project is very interesting to me and I happen to have a spare HackRF One that has no use. However the one piece of documentation that seems to be missing on the wiki is: Where can I find information on what I can record?

From my old scanner days I know I can check [RadioReference](http://www.radioreference.com/apps/db/?sid=60) for services in the San Francisco area. But others might not. Also even in my case, I'm not exactly sure what I should be looking for and how to translate that into the software configuration.

Just a suggestion.

robotastic, 2017-04-16:
Good call - There is a bit up on the Wiki ( https://github.com/robotastic/trunk-recorder/wiki/Configuring-a-System ), but the Readme doesn't really point there. Radio Reference is probably still the best point to start.

I am going to work on improving the Docs to make things easier to get started.

zglozman, 2017-04-20:
Its relly quite hard to figure out. I looked up on radio reference and i simply dont detenct anything that looks remotly like a control channel anywhere close to this. Am I missing somehting big ?
\

robotastic, 2017-04-21:
@zglozman  can you post a link of the system you are trying to record? I have seen the alternate control channels being used instead of the main one.

nkwood, 2017-04-21:
The "Retrieve Systemwide Reports" at the bottom of [this Radio Reference page](http://www.radioreference.com/apps/db/) are handy. I believe these require you to be logged in, but you are able to search their entire database by system type (e.g. "Project 25 trunked systems" or "Project 25 Phase II and X2-TDMA Systems"). You can then sort the report by city, county, state, etc. I stumbled on a system that someone needs to put up on Openmhz: https://www.radioreference.com/apps/db/?sid=7253

closed 2020-08-04 by robotastic