TrunkRecorder/trunk-recorder #128: Excessively fast playback for conventional signals

lkjww, 2017-06-28:
Hi, I have got everything setup and configured, but on two separate machines (x86 and pi) while using a rtl-sdr to decode analog signals, the audio is excessively "fast" Oddly it seems to remain the same pitch, but it is playing a a much to fast rate... Talking sounds very rapid and impossible to understand. Not sure what could be the cause. Have tried several different bandwidth speeds and everything seems to be of no value.

chudgoo, 2017-07-12:
I'm seeing the exact same issue on my install on a Pi 3.
I don't see anything in the logs to indicate why.  I am seeing a lot of "OOOOOOOOO" though... What does that indicate?

robotastic, 2017-07-13:
Weird - This is off of the master branch? Could you post your config file? I am recording analog and it is working fine. 

For the 'OOOO's that just means the Pi 3 is getting overloaded. It can't handle too much at once, way just a single recorder.

dreinhold, 2017-07-14:
Well the overruns (OOOOOO's) I guess could cause the wav to sound sped up. If you are only capturing some of the data in the wav. When you play that back, it would sound fast

closed 2020-08-04 by robotastic