TrunkRecorder/trunk-recorder #129: ENHANCEMENT - EDACS

ScanCaster, 2017-07-09:
Adding EDACS to this would allow all the current trunking protocols to be supported. 

YES, I AM AWARE EDACS is EOL and Harris is trying to KILL IT for P25. BUT...

[EDACS.c.txt](https://github.com/robotastic/trunk-recorder/files/1134121/EDACS.c.txt)

There are FOUR (4) systems in my area alone, 2 might get migrated to P25 in 2018, if they get $$. The other ones are questionable as to when they get to P25.

Linux needs EDACS... Right now there is NO WAY to do EDACS on Linux.

The EDACS protocol was "outted" in 1997. The main updates are a 4800 baud for 900MHz, and ESK which is nothing more than an XOR with a value. The other is ProVoice which DSD can do, and even now, finally there are scanners offering it, but those units are so tied to one point of failure for operating data..

Attached is basic data on EDACS that led to ETrunker.. There ARE persons who can aid with this, IF you can get them to share! Maybe this will break open the gate.....

Getting just basic EDACS would be a boon...

Dygear, 2017-10-15:
I also have an EDACS system in my area, actually the department I'm working for right now is using it. It would be awesome to have that available here.

ozjimbob, 2018-01-12:
Yes I would love to see this - just because a system is EOL doesn't mean a bunch of people aren't still out there using it.  I would love to see this enhancement.

MaxwellDPS, 2018-04-19:
I agree - Bump

photounit, 2018-08-05:
I agree - Bump

MaxwellDPS, 2018-12-03:
I really thing this needs to be added. after that there would be little to no reason to use unitrunker and that would be nice

MaxwellDPS, 2019-06-11:
As for Edacs this should provide most of the needed code and logic ;) 
https://github.com/cIay/gr-edacs
https://github.com/cIay/gr-edacs/blob/master/EDACS.txt

robotastic, 2023-02-19:
Closing because I dont have any systems near me to build against. If someone submits a PR though, I will gladly take it.

closed 2023-02-19 by robotastic

Dygear, 2023-02-20:
Yeah, the one closest to me moved to P25. I think it's dead Jim.